RfidResearchGroup/proxmark3 #1793: HF antenna is UNUSABLE

duckie37, 2022-10-05:
Hello. So I broke out my Proxmark 3 v4.01, and updated and I am unable to detect any HF RFID tags.  I have tried two antennas on this and reflashed it. I haven't done HF before so I couldn't tell you if it was working out of the box plus I purchased it long ago. 

When I run data tune  it says the HF Antenna us UNUSABLE.  LF is fine. 
Only other thing that maybe I thought was strange is when I do an HF tune I get this. Isn't it supposed to be 32 mV or is this too high? I

 [usb] pm3 --> hf tune
[=] Measuring HF antenna, click pm3 button or press Enter to exit
[=] 64 mV /  0 V /  0 Vmax
_______________________________________________________________________________________

usb] pm3 --> data tune
[=] ---------- Reminder ------------------------
[=] `hw tune` doesn't actively tune your antennas,
[=] it's only informative.
[=] Measuring antenna characteristics, please wait...
 🕛   9
[=] ---------- LF Antenna ----------
[+] LF antenna: 37.05 V - 125.00 kHz
[+] LF antenna: 29.09 V - 134.83 kHz 
[+] LF optimal: 37.05 V - 125.00 kHz
[+] Approx. Q factor (*): 5.6 by frequency bandwidth measurement
[+] Approx. Q factor (*): 6.5 by peak voltage measurement
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[!] ⚠️  HF antenna is UNUSABLE
****
(*) Q factor must be measured without tag on the antenna

[+] Displaying LF tuning graph. Divisor 88 (blue) is 134.83 kHz, 95 (red) is 125.00 kHz

[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 40904
[#]   Available memory........ 37060
[#] Tracing
[#]   tracing ................ 0
[#]   traceLen ............... 68
[#] Current FPGA image
[#]   mode.................... HF image 2s30vq100 2022-03-23 17:21:16
[#] Flash memory
[#]   Baudrate................ 24 MHz
[#]   Init.................... OK
[#]   Memory size............. 2 mbits / 256 kb
[#]   Unique ID............... 0xD567A882A79D8825
[#] Smart card module (ISO 7816)
[#]   version................. v3.10
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0
[#]
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A |
[#]     long leading reference |N/A | N/A | N/A | N/A | N/A | N/A | N/A |
[#]               leading zero |N/A | N/A | N/A | N/A | N/A | N/A | N/A |
[#]    1 of 4 coding reference |N/A | N/A | N/A | N/A | N/A | N/A | N/A |
[#]
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 283136
[#]   Transfer Speed PM3 -> Client... 566272 bytes/s
[#] Various
[#]   Max stack usage......... 4088 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... 1
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 31015 Hz
[#] Installed StandAlone Mode
[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[#] Flash memory dictionary loaded
[#]
_______________________________________________________________________________________
[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 40904
[#]   Available memory........ 37060
[#] Tracing
[#]   tracing ................ 0
[#]   traceLen ............... 68
[#] Current FPGA image
[#]   mode.................... HF image 2s30vq100 2022-03-23 17:21:16
[#] Flash memory
[#]   Baudrate................ 24 MHz
[#]   Init.................... OK
[#]   Memory size............. 2 mbits / 256 kb
[#]   Unique ID............... 0xD567A882A79D8825
[#] Smart card module (ISO 7816)
[#]   version................. v3.10
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0
[#]
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A |
[#]     long leading reference |N/A | N/A | N/A | N/A | N/A | N/A | N/A |
[#]               leading zero |N/A | N/A | N/A | N/A | N/A | N/A | N/A |
[#]    1 of 4 coding reference |N/A | N/A | N/A | N/A | N/A | N/A | N/A |
[#]
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 281600
[#]   Transfer Speed PM3 -> Client... 563200 bytes/s
[#] Various
[#]   Max stack usage......... 4088 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... 1
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 31015 Hz
[#] Installed StandAlone Mode
[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[#] Flash memory dictionary loaded

iceman1001, 2022-10-07:
Check the screws?  flash latest source,   run same client and firmware from same build?

side note,
follow the guides on this repo to setup your rdv4 properly

duckie37, 2022-10-07:
Still a problem. 
Ok, I'm sure I followed the guides on setup and thought I had the latest but I went ahead and did it again. 
Sorry, I think I forgot to post the firmware and software versions on my first post. 
Tried 3 antennas all bolted down well. 
I am starting to think it's an HW issue or there would be more posts with this release. Is there a way to test the antenna with a volt meter while running tune or other commands (e.g. Voltage drop or Ohmz while it's running a test?)

 [ Proxmark3 RFID instrument ]

    MCU....... AT91SAM7S512 Rev A
    Memory.... 512 Kb ( 65% used )

    Client.... Iceman/master/v4.14831-976-g8b4701efe 2022-10-07 10:36:55
    Bootrom... Iceman/master/v4.14831-976-g8b4701efe 2022-10-07 10:35:53
    OS........ Iceman/master/v4.14831-976-g8b4701efe 2022-10-07 10:36:37
    Target.... RDV4

iceman1001, 2022-12-05:
Ping!
Does this still occur for you?

melniks-bsi, 2022-12-14:
Hey sorry. I haven't gotten back to it but I am convinced since I replaced the antenna with 2 others I had that I have bad hardware because nothing makes sense after re-flashing and updating software a few times.  We can close this out unless there is more info you would like.

duckie37, 2022-12-14:
Hey sorry. I haven't got back to this but after re-flashing a few times and trying a few other  antennas from another friends kit I am convinced that I have a hardware problem somewhere.  I will close this out unless there is more info you would like to see.

closed 2022-12-14 by duckie37