RfidResearchGroup/proxmark3 #1795: Not able to "make acsessrights"

remsforian, 2022-10-15:
I am attempting to install the iceman fork of proxmark 3 on fedora linux. I had used it with no problem on Pop_OS! (based on Ubuntu) (well, less major problems). I went through all of the steps (including disabling modem manager), but came across an error when running the command "make accessrights" I get the below output `[solmasters@fedora proxmark3]$ make accessrights

===================================================================
Version info:      Iceman/master/v4.14831-976-g8b4701efe
Platform name:     Proxmark3 RDV4
PLATFORM:          PM3RDV4
PLATFORM_FPGA:     xc2s30
PLATFORM_SIZE:     512
Platform extras:   No extra selected
Included options:  SMARTCARD FLASH -DRDV4 LF HITAG EM4x50 EM4x70 ZX8211 GENERAL_HF ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   LF_SAMYRUN
===================================================================
Usage: adduser [options] LOGIN
       adduser -D
       adduser -D [options]

Options:
      --badnames                do not check for bad names
  -b, --base-dir BASE_DIR       base directory for the home directory of the
                                new account
      --btrfs-subvolume-home    use BTRFS subvolume for home directory
  -c, --comment COMMENT         GECOS field of the new account
  -d, --home-dir HOME_DIR       home directory of the new account
  -D, --defaults                print or change default useradd configuration
  -e, --expiredate EXPIRE_DATE  expiration date of the new account
  -f, --inactive INACTIVE       password inactivity period of the new account
  -g, --gid GROUP               name or ID of the primary group of the new
                                account
  -G, --groups GROUPS           list of supplementary groups of the new
                                account
  -h, --help                    display this help message and exit
  -k, --skel SKEL_DIR           use this alternative skeleton directory
  -K, --key KEY=VALUE           override /etc/login.defs defaults
  -l, --no-log-init             do not add the user to the lastlog and
                                faillog databases
  -m, --create-home             create the user's home directory
  -M, --no-create-home          do not create the user's home directory
  -N, --no-user-group           do not create a group with the same name as
                                the user
  -o, --non-unique              allow to create users with duplicate
                                (non-unique) UID
  -p, --password PASSWORD       encrypted password of the new account
  -r, --system                  create a system account
  -R, --root CHROOT_DIR         directory to chroot into
  -P, --prefix PREFIX_DIR       prefix directory where are located the /etc/* files
  -s, --shell SHELL             login shell of the new account
  -u, --uid UID                 user ID of the new account
  -U, --user-group              create a group with the same name as the user
  -Z, --selinux-user SEUSER     use a specific SEUSER for the SELinux user mapping

make: *** [Makefile:271: accessrights] Error 2`

I also tried to run "pm3" which gives me the following output:
`[=] Session log /home/solmasters/.proxmark3/logs/log_20221015.txt
[=] Using UART port /dev/ttyACM0

[!] ⚠️  ERROR: invalid serial port /dev/ttyACM0

[?] Try the shell script `./pm3 --list` to get a list of possible serial ports
`
and the test command did not give an output. I also tried rebooting my system after install.

iceman1001, 2022-10-15:
If you look in the Makefile line 271,  like the error messages says,   

`	$(Q)sudo adduser $(USER) dialout ` 

Seems like your OS doesn't support the adduser command.

remsforian, 2022-10-15:
> 

Huh, that's odd. The command add userseems to be working in the command line. Is there perhaps a place in the make file where I could enter my username to circumvent the issue?

iceman1001, 2022-10-16:
Ok,  then there is an bug in the Makefile when `adduser` is called on Fedora 35. 

how do you call `adduser` from command line?

remsforian, 2022-10-16:
When I use adduser in the command line I just say `adduser username`. I also just went into the make file and replaced `adduser` with `useradd` and got the same error, although running `make` with no modifications seems to work.
I also should note that I am running Fedora 36, not 35, but I don't know how much of a difference that makes

iceman1001, 2022-10-17:
running `make` w/o changes work?   I am confused,  we are talking about `make accessrights` ..
they do different things.

remsforian, 2022-10-18:
Yes. I am aware they do two different things, but I thought I would try running just `make` to see if there was a similar error there. There was not.

iceman1001, 2022-12-05:
ping!
Any progress found?

mooey5775, 2022-12-10:
Running Fedora 37 here - adding a user to a group is done the same way as in Arch (`sudo usermod -aG {GROUP} {USERNAME}`). I can write a quick PR to fix this in the Makefile.

I would actually continue to argue that maybe `usermod` should be used in the Makefile for all distros, as it should work to add a user to a group even on non-Arch/Fedora distros, and would get rid of the distro check.

iceman1001, 2022-12-10:
The PR is nice,   you can test running against all the different docker setups to verify a different solution,

PR #1840 (2022-12-10): Fix `make accessrights` for Fedora
Fixes #1795

Fedora doesn't allow the use of `adduser` to add users to groups. Instead, `usermod -aG` should be used, like in Arch, but with the dialout group instead of uucp.

As a brief discussion item before merge, should non-Arch distros just use `usermod` instead of `adduser`? This should work in general, and would get rid of the extra conditional. However, doing the testing for this might be difficult.
closed 2022-12-10 by iceman1001