RfidResearchGroup/proxmark3 #1796: faile to flash fullimage.elf > 256K

4ft35t, 2022-10-15:
My device has 512K flash memory, but can't flash big than 256K.

## hw version
```bash
 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 47% used )
```

## > 256K flash error
compile with `PLATFORM=PM3GENERIC`, fullimage.elf size is 296K
```bash
 ls -alh armsrc/obj/fullimage.elf
-rwxr-xr-x 1 kali kali 296K Oct 16 03:48 armsrc/obj/fullimage.elf
```
flash error
```bash
kali@SF5P-BOOK:~/iceman/proxmark3$ ./pm3-flash-all
[=] Session log /home/kali/.proxmark3/logs/log_20221015.txt
[+] loaded from JSON file /home/kali/.proxmark3/preferences.json
[+] About to use the following files:
[+]    /home/kali/iceman/proxmark3/client/../bootrom/obj/bootrom.elf
[+]    /home/kali/iceman/proxmark3/client/../armsrc/obj/fullimage.elf
[+] Loading ELF file /home/kali/iceman/proxmark3/client/../bootrom/obj/bootrom.elf
[+] ELF file version Iceman/master/v4.14831-979-ga0e275acc 2022-10-16 03:47:21 180bffbf5

[+] Loading ELF file /home/kali/iceman/proxmark3/client/../armsrc/obj/fullimage.elf
[+] ELF file version Iceman/master/v4.14831-979-ga0e275acc 2022-10-16 03:47:57 180bffbf5

[+] Waiting for Proxmark3 to appear on /dev/ttyS7
 🕑  59 found
[+] Entering bootloader...
[+] (Press and release the button only to abort)
[+] Waiting for Proxmark3 to appear on /dev/ttyS7
 🕒  48 found
[=] Available memory on this board: 512K bytes

[=] Permitted flash range: 0x00100000-0x00180000
[+] Loading usable ELF segments:
[+]    0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
[+]    1: V 0x00200000 P 0x00100200 (0x00000d1c->0x00000d1c) [R X] @0x298

[+] Loading usable ELF segments:
[+]    1: V 0x00102000 P 0x00102000 (0x00047dac->0x00047dac) [R X] @0xb8
[+]    2: V 0x00200000 P 0x00149dac (0x00001b04->0x00001b04) [R X] @0x47e68
[=] Note: Extending previous segment from 0x47dac to 0x498b0 bytes

[+] Flashing...
[+] Writing segments for file: /home/kali/iceman/proxmark3/client/../bootrom/obj/bootrom.elf
[+]  0x00100000..0x001001ff [0x200 / 1 blocks]
. ok
[+]  0x00100200..0x00100f1b [0xd1c / 7 blocks]
....... ok

[+] Writing segments for file: /home/kali/iceman/proxmark3/client/../armsrc/obj/fullimage.elf
[+]  0x00102000..0x0014b8af [0x498b0 / 589 blocks]
...................................................................
        @@@  @@@@@@@ @@@@@@@@ @@@@@@@@@@   @@@@@@  @@@  @@@
        @@! !@@      @@!      @@! @@! @@! @@!  @@@ @@!@!@@@
        !!@ !@!      @!!!:!   @!! !!@ @!@ @!@!@!@! @!@@!!@!
        !!: :!!      !!:      !!:     !!: !!:  !!! !!:  !!!
        :    :: :: : : :: :::  :      :    :   : : ::    :
        .    .. .. . . .. ...  .      .    .   . . ..    .
...........................................................[!!] 🚨 Error: Unexpected reply 0x00fe NACK (expected ACK)
       Lock Error

       Lock Bits:      0xffff
[!!] 🚨 Error writing block 496 of 589
[!] ⚠️  The flashing procedure failed, follow the suggested steps!
[+] All done

[=] Have a nice day!
```

## < 256K flash OK
fellow https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Use_of_Proxmark/4_Advanced-compilation-parameters.md#256kb-versions, reduce image size

Makefile.platform
```bash
PLATFORM=PM3GENERIC
PLATFORM_SIZE=256
STANDALONE=
SKIP_HITAG=1
SKIP_FELICA=1
SKIP_HFPLOT=1
SKIP_LF=1
```
fullimage.elf size is 238K
```bash
ls -alh armsrc/obj/fullimage.elf
-rwxr-xr-x 1 kali kali 238K Oct 16 04:05 armsrc/obj/fullimage.elf
```
flash ok
```

[+] Flashing...
[+] Writing segments for file: /home/kali/iceman/proxmark3/client/../bootrom/obj/bootrom.elf
[+]  0x00100000..0x001001ff [0x200 / 1 blocks]
. ok
[+]  0x00100200..0x00100f1b [0xd1c / 7 blocks]
....... ok

[+] Writing segments for file: /home/kali/iceman/proxmark3/client/../armsrc/obj/fullimage.elf
[+]  0x00102000..0x0013d38e [0x3b38f / 474 blocks]
...................................................................
        @@@  @@@@@@@ @@@@@@@@ @@@@@@@@@@   @@@@@@  @@@  @@@
        @@! !@@      @@!      @@! @@! @@! @@!  @@@ @@!@!@@@
        !!@ !@!      @!!!:!   @!! !!@ @!@ @!@!@!@! @!@@!!@!
        !!: :!!      !!:      !!:     !!: !!:  !!! !!:  !!!
        :    :: :: : : :: :::  :      :    :   : : ::    :
        .    .. .. . . .. ...  .      .    .   . . ..    .
..................................... ok

[+] All done

[=] Have a nice day!
```

hf works fine, no lf
```bash
  [ Proxmark3 RFID instrument ]

    MCU....... AT91SAM7S512 Rev B
    Memory.... 512 Kb ( 48% used )

    Client.... Iceman/master/v4.14831-979-ga0e275acc 2022-10-16 04:05:40
    Bootrom... Iceman/master/v4.14831-979-ga0e275acc 2022-10-16 04:04:36
    OS........ Iceman/master/v4.14831-979-ga0e275acc 2022-10-16 04:05:10
    Target.... PM3 GENERIC
```

## with v4.9237, fullimage.elf size 243K, both hf and lf works.

iceman1001, 2022-10-16:
sounds like a hardware issue with your device.

4ft35t, 2022-10-16:
> sounds like a hardware issue with your device.

maybe，just wonder is anyone else has the same issue。

iceman1001, 2022-10-16:
you can try flashing some more but in my experience its your hardware that seem to be faulty

4ft35t, 2022-10-17:
my  workaround is compile two fullimage.elf, `lf only` and `hf only`, to use the latest feature. LOL

closed 2022-10-17 by 4ft35t

joelsernamoreno, 2023-01-23:
Hello!

I have taken out of the closet my two proxmark easy (I currently use proxmark3 rdv4). I have configured Makefile.platform for PLATFORM=PM3GENERIC and I get the same problem on my two proxmark easy

This did not happen before, I have always been able to flash my proxmark easy (512K) with this firmware :)

iceman1001, 2023-01-24:
@joelsernamoreno  What does your output look like when you flash?

joelsernamoreno, 2023-01-24:
Hi @iceman1001 

My steps:

1.- The bootrom flashing is OK:

[image]

2.- Full image error:

[image]

3.- I can edit Makefile.platform like this:

[image]

4.- With the Makefile.platform edited, no errors when flashing fullimage:

[image]

I hope this can help others. I use RVD4, but today I was bored and tried to update my old proxmark and got this error

Cheers!

iceman1001, 2023-01-24:
Still looks like your ARM chip onboard flash memory is corrupt.

joelsernamoreno, 2023-02-06:
Just a note, the next release does not give problems here:

[proxmark3-v4.9237](https://github.com/RfidResearchGroup/proxmark3/releases/tag/v4.9237)