RfidResearchGroup/proxmark3 #1799: Different results with "LEGIC prime MIM 1024 card" for Windows Laptop vs MacBook Pro (Apple M1 Pro)

tmeyerinfo, 2022-10-22:
I am currently testing my new "PM3 easy" on both my Windows 10 and MacBook Pro (Apple M1 Pro) laptop, running the latest Iceman firmware (v4.14831).

On my Windows 10 laptop I get good reads on my "LEGIC prime MIM 1024 card", on my MacBook Pro the PM3 doesn't recognize the card at all.

I tried the original cable that came with the "PM3 easy" + an USB-C dongle and also tried a brand new "USB Micro to USB-C" cable.

For other cards like the "Mifare Classic" I get good reads on both OS.

Where similar problems reported before?

iceman1001, 2022-10-22:
two different OS,...  two different client?   

the recommendation is to use the same client & fw from the same build.

You can see when you start the client if the build is the same

tmeyerinfo, 2022-10-23:
Thank you it worked!

I reflashed the image on my MacBook and the client was now able to read the card.

My assumption was that the firmware is always the same and just exposes how to speak to the pm3 and the client on every OS just implements these commands in a language suitable to the OS.

closed 2022-10-23 by tmeyerinfo

iceman1001, 2022-10-23:
As you notice that is not the case.  It comes in pairs,  client and firmware.