RfidResearchGroup/proxmark3 #1801: Error [!!] Failed to communicate with card

Mannilie, 2022-10-25:
**Description**
I'm not sure if it's possibly something I'm doing wrong or it's a dud card but I'm having a really weird issue when trying to write a dump from a HID iClass card. 

**To Reproduce**
I'm trying to run the following command `hf iclass dump -k <hid key>`

and it's producing the following:

`[=] Card has at least 2 application areas. AA1 limit 18 (0x12) AA2 limit 31 (0x1F)
.
[!!] failed to communicate with card`

Everywhere I look about this error is very inconclusive. I've tried all kinds of things like the following:

- Repositioning the card on various edges/angles of the Proxmark3.
- Using different USB cables.
- Trying alternate outputs of the Proxmark3.
- Trying different / alternate known keys.

Still unsure what is causing this issue. Maybe someone can shine some light on where I'm going wrong here?

**Expected behavior**
From what I saw with other cards, it should be able to dump the data into a file.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Proxmark3 is using Iceman Firmware.

HW Version:
` [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  Iceman/master/v4.14831-1008-ga1633f932 2022-10-25 11:09:07 8b5c14153
  compiled with............. MinGW-w64 10.3.0
  platform.................. Windows (64b) / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... absent
  Python script support..... present
  Lua SWIG support.......... present
  Python SWIG support....... present

 [ PROXMARK3 ]
  firmware.................. PM3 GENERIC

 [ ARM ]
  bootrom: Iceman/master/v4.14831-1008-ga1633f932 2022-10-25 11:08:34 8b5c14153
       os: Iceman/master/v4.14831-1008-ga1633f932 2022-10-25 11:08:44 8b5c14153
  compiled with GCC 10.1.0

 [ FPGA ]
  LF image 2s30vq100 2022-03-23 17:21:05
  HF image 2s30vq100 2022-03-23 17:21:16
  HF FeliCa image 2s30vq100 2022-03-23 17:21:27
  HF 15 image 2s30vq100 2022-03-23 17:21:38

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 59% used )`

HW Status:
`[#] Memory
[#]   BigBuf_size............. 42780
[#]   Available memory........ 37400
[#] Tracing
[#]   tracing ................ 0
[#]   traceLen ............... 402
[#] Current FPGA image
[#]   mode.................... HF 15 image 2s30vq100 2022-03-23 17:21:38
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0
[#]
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
[#]     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
[#]               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A |
[#]    1 of 4 coding reference |  31 |  20 |  18 |  34 |  15 |  50 |  66 |
[#]
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 272896
[#]   Transfer Speed PM3 -> Client... 545792 bytes/s
[#] Various
[#]   Max stack usage......... 4120 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... 38
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 32964 Hz
[#] Installed StandAlone Mode
[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[#]`

bettse, 2022-10-25:
have you tried `hf iclass chk -f iclass_default_keys.dic --elite` ?

Mannilie, 2022-10-25:
Hey @bettse,

Thanks for your prompt response. 
Unfortunately, it didn't provide me with any debug keys. It's almost like it's failing to read the key again.

[image]

iceman1001, 2022-10-25:
try run it w/o the elite param   `hf iclass chk -f iclass_default_keys.dic`

iceman1001, 2022-10-25:
Anyway,  I suggest you join the discord and ask questions in #iclass channel.    I don't think there is an issue with the source code

Mannilie, 2022-10-25:
> try run it w/o the elite param `hf iclass chk -f iclass_default_keys.dic`

Yeah it produces the same issue. Not to worry, I'll ask in Discord! Thanks for your prompt replies! :)

closed 2022-10-25 by Mannilie