RfidResearchGroup/proxmark3 #1804: hf mf staticnested command return unknown error

Paliddo86, 2022-10-29:
**Describe the bug**
I try to grab the keys for sector 0 and 1 on mifare 1k with all other key seated on default 0xFFFFFFFFFFFF.
The info command found a static nonce but when i run the command for staticnested card the result is unknown error. 

This is result of the executed command, i use block 8 for input with key A

[usb] pm3 --> hf mf staticnested --1k -k ffffffffffff --blk 8
[#] 1 static nonce 01200145
[+] Testing known keys. Sector count 16
[=] Chunk 0,6s | found 0/32 keys (43)
[+] Time to check 42 known keys: 1 seconds

[+] enter static nested key recovery
[!!] 🚨 unknown error.

** This is the trace of the command  list **
[usb] pm3 --> hf mf list
[=] downloading tracelog data from device
[+] Recorded activity (trace len = 800 bytes)
[=] start = start of start frame end = end of frame. src = source of transfer
[=] ISO14443A - all times are in carrier periods (1/13.56MHz)

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
                0 |       4768 | Rdr |50  00  57  cd                                                           |  ok | HALT
     142336 |     143328 | Rdr |52(7)                                                                    |     | WUPA
     144452 |     146820 | Tag |04  00                                                                   |     | 
     149376 |     151840 | Rdr |93  20                                                                   |     | ANTICOLL
     152900 |     158788 | Tag |8d  2d  6f  67  a8                                                       |     | 
     161408 |     171872 | Rdr |93  70  8d  2d  6f  67  a8  f4  65                                       |  ok | SELECT_UID
     172996 |     176516 | Tag |08  b6  dd                                                               |     | 
     178688 |     183456 | Rdr |60  08  bd  f7                                                           |  ok | AUTH-A(8)
     185284 |     190020 | Tag |01  20  01  45                                                           |     | AUTH: nt 
     199296 |     208672 | Rdr |dd  47! 17  32! 9a  87! fc  ec                                           |     | AUTH: nr ar (enc)
     209732 |     214404 | Tag |7f  54  ae  24                                                           |     | AUTH: at (enc)
     222848 |     227616 | Rdr |16! 0c  09! 26                                                           |     | 
            |            |  *  |                                              key FFFFFFFFFFFF prng WEAK |     |
            |            |  *  |60  00  F5  7B                                                           |  ok | AUTH-A(0)
     229444 |     234180 | Tag |b5  36! 13! e4!                                                          |     | AUTH: nt (enc)
     237824 |     242592 | Rdr |d9! 64! 2f! 34                                                           |     | 
     380160 |     381152 | Rdr |52(7)                                                                    |     | WUPA
     382276 |     384644 | Tag |04  00                                                                   |     | 
     387200 |     389664 | Rdr |93  20                                                                   |     | ANTICOLL
     390724 |     396612 | Tag |8d  2d  6f  67  a8                                                       |     | 
     399232 |     409696 | Rdr |93  70  8d  2d  6f  67  a8  f4  65                                       |  ok | SELECT_UID
     410820 |     414340 | Tag |08  b6  dd                                                               |     | 
     416512 |     421280 | Rdr |60  08  bd  f7                                                           |  ok | AUTH-A(8)
     423108 |     427844 | Tag |01  20  01  45                                                           |     | AUTH: nt 
     437120 |     446496 | Rdr |25! d0! 11! 73  4b! d0  26  fb                                           |     | AUTH: nr ar (enc)
     447556 |     452228 | Tag |35! 92! d7  23!                                                          |     | AUTH: at (enc)
     458496 |     463264 | Rdr |ce  ab! 46! 6f!                                                          |     | 
            |            |  *  |                                              key FFFFFFFFFFFF prng WEAK |     |
            |            |  *  |60  08  BD  F7                                                           |  ok | AUTH-A(8)
     465092 |     469828 | Tag |81! 70  a1! ea                                                           |     | AUTH: nt (enc)
     479104 |     488416 | Rdr |a5! 14! be! 16  11  05  b1! 3c                                           |     | AUTH: nr ar (enc)
     489540 |     494212 | Tag |2d  a5! 90  60                                                           |     | AUTH: at (enc)
     500096 |     504800 | Rdr |b2! 31  ec  6c                                                           |     | 
            |            |  *  |                                               last used key FFFFFFFFFFFF|     |
            |            |  *  |60  00  F5  7B                                                           |  ok | AUTH-A(0)
     506692 |     511364 | Tag |79  23  c2  2f!                                                          |     | AUTH: nt (enc)
     515072 |     519840 | Rdr |86  b7  4d! 77!                                                          |     | 
     657408 |     658400 | Rdr |52(7)                                                                    |     | WUPA
     659524 |     661892 | Tag |04  00                                                                   |     | 
     664448 |     666912 | Rdr |93  20                                                                   |     | ANTICOLL
     667972 |     673860 | Tag |8d  2d  6f  67  a8                                                       |     | 
     676480 |     686944 | Rdr |93  70  8d  2d  6f  67  a8  f4  65                                       |  ok | SELECT_UID
     688068 |     691588 | Tag |08  b6  dd                                                               |     | 
     693760 |     698528 | Rdr |60  08  bd  f7                                                           |  ok | AUTH-A(8)
     700356 |     705092 | Tag |01  20  01  45                                                           |     | AUTH: nt 
     714368 |     723744 | Rdr |cd! 76  0b! 4c  9b  be! 3a  4e!                                          |     | AUTH: nr ar (enc)
[... 3 lines cut ...]
            |            |  *  |60  00  F5  7B                                                           |  ok | AUTH-A(0)
     744516 |     749252 | Tag |b5  36! 13! e4!                                                          |     | AUTH: nt (enc)
     752896 |     757664 | Rdr |b6  ed! 40  4f                                                           |     | 
     895232 |     896224 | Rdr |52(7)                                                                    |     | WUPA
     897348 |     899716 | Tag |04  00                                                                   |     | 
     902272 |     904736 | Rdr |93  20                                                                   |     | ANTICOLL
     905796 |     911684 | Tag |8d  2d  6f  67  a8                                                       |     | 
     914304 |     924768 | Rdr |93  70  8d  2d  6f  67  a8  f4  65                                       |  ok | SELECT_UID
     925892 |     929412 | Tag |08  b6  dd                                                               |     | 
     931584 |     936352 | Rdr |60  08  bd  f7                                                           |  ok | AUTH-A(8)
     938180 |     942916 | Tag |01  20  01  45                                                           |     | AUTH: nt 
     952192 |     961504 | Rdr |1d  67  0b! 20! 0b! ab! 03  4a!                                          |     | AUTH: nr ar (enc)
     962628 |     967364 | Tag |fa  2b! 88! 1e!                                                          |     | AUTH: at (enc)
     973568 |     978272 | Rdr |0d  6b  b9  d2                                                           |     | 
[... cut ...]
    1015168 |    1019872 | Rdr |cc  9a! 82  d3!                                                          |     | 
            |            |  *  |                                               last used key FFFFFFFFFFFF|     |
            |            |  *  |60  00  F5  7B                                                           |  ok | AUTH-A(0)
    1021764 |    1026436 | Tag |79  23  c2  2f!                                                          |     | AUTH: nt (enc)

If you need other data write here and i get you all needed info for this key

iceman1001, 2022-10-29:
I suspect its more user issue than a software issue.
Please use the discord to ask your questions why it fails.

Paliddo86, 2022-10-30:
Thanks for reply the last versione of software client fix this issue

iceman1001, 2022-10-30:
good,  closing

closed 2022-10-30 by iceman1001