RfidResearchGroup/proxmark3 #1808: data diff: read out of bounds if file length is not a multiple of output width

DidierA, 2022-11-07:
**Describe the bug**
When using `data diff` with files whose length is not a multiple of 16 or the width specified with -w, the last line of diff will contain bogus data, reading the buffer after its end.

**To Reproduce**
Steps to reproduce the behavior:

For instance, using a dump of mifare ultralight whose length is 136 (0x88) bytes and the default width of 16, and asking the diff between the file and itself:
```
[usb] pm3 --> data diff -a hf-mfu-0467E4AAC85580-dump.bin -b hf-mfu-0467E4AAC85580-dump.bin
[=] Searching hf-mfu-0467E4AAC85580-dump.bin
[=] Found hf-mfu-0467E4AAC85580-dump.bin
[+] loaded 136 bytes from binary file hf-mfu-0467E4AAC85580-dump.bin
[=] Searching hf-mfu-0467E4AAC85580-dump.bin
[=] Found hf-mfu-0467E4AAC85580-dump.bin
[+] loaded 136 bytes from binary file hf-mfu-0467E4AAC85580-dump.bin
[#] data len:  136   A 136  B 136
[=] 
[=] ----+------------------------------------------------------------------+------------------------------------------------------------------
[=]  #  | a                                                                | b
[=] ----+------------------------------------------------------------------+------------------------------------------------------------------
[=] 000 | 00 04 03 01 01 00 0B 03 00 00 00 13 0B FD 2C DF ..............,. | 00 04 03 01 01 00 0B 03 00 00 00 13 0B FD 2C DF ..............,.
```
everything ok so far ... I cut to the last line
```
[=] 070 | 00 00 00 00 00 00 00 00 00 00 00 FF 00 05 00 00 ................ | 00 00 00 00 00 00 00 00 00 00 00 FF 00 05 00 00 ................
[=] 080 | 00 00 00 00 00 00 00 00 21 00 00 00 00 00 00 00 ........!....... | 00 00 00 00 00 00 00 00 31 00 00 00 00 00 00 00 ........Q.......
[=] ----+------------------------------------------------------------------+------------------------------------------------------------------
```
At offset 0x88 (past the end of file) if finds a diff between the two same files an ddisplays data that does not exist.

**Expected behavior**

```
[=] 070 | 00 00 00 00 00 00 00 00 00 00 00 FF 00 05 00 00 ................ | 00 00 00 00 00 00 00 00 00 00 00 FF 00 05 00 00 ................
[=] 080 | 00 00 00 00 00 00 00 00                         ........         | 00 00 00 00 00 00 00 00                         ........        
[=] ----+------------------------------------------------------------------+------------------------------------------------------------------
```

**Additional context**
Add any other context about the problem here.

DidierA, 2022-11-07:
Attached a patch that partially fixes the problem (only works for 2 files of the same size and with no diff in the last line of output)
[cmddata-patch.txt](https://github.com/RfidResearchGroup/proxmark3/files/9955234/cmddata-patch.txt)

iceman1001, 2022-11-08:
yeah,   I assumed more or less that you would use same style files...   and not compare files that differ too much.
And I saw you made a fix.  Nice!

closed 2022-11-08 by iceman1001

DidierA, 2022-11-08:
I was scratching my head trying to understand why I had strange diffs between 2 dumps of the same ultralight tag, until I realized it was due to this bug 😅

iceman1001, 2022-11-08:
you shouldnt have `-w 16`  for MFU files,    or rather, you should compare similar files with each other.
But I am glad you fixed this case.

Now the width should be automatically set to 4 for MFU files and 8 for Iclass files ;)

PR #1809 (2022-11-08): Fix read oob in data diff
Fixes #1808