RfidResearchGroup/proxmark3 #1812: pm3 script hangs when launch without arguments, no proxmark pluged in, and bluetooth has been used before.

DidierA, 2022-11-15:
**Describe the bug**
When launching `pm3` without arguments, and blueshark was used earlier, the script finds the blueshark's bluetooth address in bluetooth parameters and tries to connect to it. If it's off or not in range, the connection is left hanging, never fails, and neither `Ctrl-C` nor `kill -9` can stop the `proxmark3` process.

**To Reproduce**
Steps to reproduce the behavior:
1. `client/pm3 -p bt:00:21:08:35:25:E8`

**Expected behavior**
A timeout should occur, and the client switch to offline mode.

**Desktop (please complete the following information):**
 - OS: linux
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
 ```
[offline] pm3 --> hw version 

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  Iceman/master/v4.15864-41-g6c163fa26-dirty-unclean 2022-11-15 11:50:23 c6f8b6756
  compiled with............. GCC 11.3.0
  platform.................. Linux / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... present
  Python script support..... present
  Lua SWIG support.......... present
  Python SWIG support....... present
```
**Additional context**
`strace` shows that the process is hung in a `connect()` call:
```
write(1, "[\33[33m=\33[0m] Using UART port \33[3"..., 59[=] Using UART port bt:00:21:08:35:25:E8
) = 59
write(3, "[=] Using UART port bt:00:21:08:"..., 41) = 41
alarm(5)                                = 0
socket(AF_BLUETOOTH, SOCK_STREAM, BTPROTO_RFCOMM) = 4
connect(4, {sa_family=AF_BLUETOOTH, rc_bdaddr=e8:25:35:08:21:00, rc_channel=1}, 10
```
The `connect()` call is at line 199 of `client/src/uart/uart_posix.c`, I guess an alarm() call and signal handler should allow for a timeout and Ctrl-C  handling.

DidierA, 2022-11-15:
Nevermind, I cannot reproduce this now, the connect() call times out... I guess my bluetooth stack was in a bad mood before.

closed 2022-11-15 by DidierA

iceman1001, 2022-11-16:
well,   if an alarm call / signal handler can make things gentler,  then by all means add it