RfidResearchGroup/proxmark3 #1822: script hf_mf_ultimatecard report "partial read of configuration, use -k or change cfg0 block"

et-ness, 2022-11-26:
**Describe the bug**
I have an Ultime Magic Card, I tried to use the script hf_mf_ultimatecard for read configuration and give me the error:

```
[usb] pm3 --> script run hf_mf_ultimatecard -c 
[+] executing lua /mnt/data/progetti/pm3/iceman/client/luascripts/hf_mf_ultimatecard.lua
[+] args '-c'

ERROR: 	partial read of configuration, use -k or change cfg0 block	

[+] finished hf_mf_ultimatecard
```
My ultimate card return 32 byte from configuration

```
[usb] pm3 --> hf 14a raw -s -c -t 1000 CF00000000C6
[+] 00 00 00 00 00 00 02 09 09 78 00 91 02 BD AC 19 13 10 11 12 13 14 15 16 04 00 08 02 6B 00 4F 6B [ F4 41 ]
```

I tried to apply this patch

```
diff --git a/client/luascripts/hf_mf_ultimatecard.lua b/client/luascripts/hf_mf_ultimatecard.lua
index b9011c36f..13710a4f1 100644
--- a/client/luascripts/hf_mf_ultimatecard.lua
+++ b/client/luascripts/hf_mf_ultimatecard.lua
@@ -186,7 +186,7 @@ local function read_config()
     atqaf = atqa1..' '..atqa2
     cardtype, cardprotocol, gtustr, atsstr = 'unknown', 'unknown', 'unknown', 'unknown'
     if magicconfig == nil then lib14a.disconnect(); return nil, "can't read configuration, "..err_lock end
-    if #magicconfig ~= 64 then lib14a.disconnect(); return nil, "partial read of configuration, "..err_lock end
+    if #magicconfig ~= 68 then lib14a.disconnect(); return nil, "partial read of configuration, "..err_lock end
     if gtumode == '00' then gtustr = 'Pre-write/Shadow Mode'
     elseif gtumode == '01' then gtustr = 'Restore Mode'
     elseif gtumode == '02' then gtustr = 'Disabled'
```

and works

```
[usb] pm3 --> script run hf_mf_ultimatecard -c
[+] executing lua /mnt/data/progetti/pm3/iceman/client/luascripts/hf_mf_ultimatecard.lua
[+] args '-c'

--> #magicconfig 68	expected 64
========================================================================================	
			Ultimate Magic Card Configuration	
========================================================================================	
 - Raw Config      	00000000000002090978009102BDAC191310111213141516040008026B00	
 - Card Protocol    	MIFARE Classic Protocol	
 - Ultralight Mode   	Disabled	
 - ULM Backdoor Key 	00000000	
 - GTU Mode     	Disabled	
 - Card Type     	MIFARE 1k S50 4-byte UID	
 - UID           	04223344	
 - ATQA          	00 04	
 - SAK          	08	

[+] finished hf_mf_ultimatecard
```

In my card there are these 2 bytes 6B 00 in last position but I don't know what do they mean. I never executed change commands, only a "script run hf_mf_ultimatecard -m 02"

```
[usb] pm3 --> hf search
 🕕  Searching for ISO14443-A tag...          
[+]  UID: 04 22 33 44 
[+] ATQA: 00 04
[+]  SAK: 08 [2]
[+] Possible types:
[+]    MIFARE Classic 1K
[=] -------------------------- ATS --------------------------
[+] ATS: 09 78 00 91 02 BD AC 19 13 [ 7C 00 ]
[=]      09...............  TL    length is 9 bytes
[=]         78............  T0    TA1 is present, TB1 is present, TC1 is present, FSCI is 8 (FSC = 256)
[=]            00.........  TA1   different divisors are supported, DR: [], DS: []
[=]               91......  TB1   SFGI = 1 (SFGT = 8192/fc), FWI = 9 (FWT = 2097152/fc)
[=]                  02...  TC1   NAD is NOT supported, CID is supported

[=] -------------------- Historical bytes --------------------
[+]   BDAC1913

[+] Prng detection: weak
[#] Auth error
[?] Hint: try `hf mf` commands

[+] Valid ISO 14443-A tag found

 🕓  Searching for ISO14443-B tag...          
[+]  UID    : 04 22 33 44 
[+]  ATQB   : 1C 00 00 11 77 81 85 
[+]  CHIPID : 00
[+]       App Data: 1C 00 00 11 
[+]       Protocol: 77 81 85 
[+]       Bit Rate: 212 kbit/s PICC -> PCD supported
[+]       Bit Rate: 424 kbit/s PICC -> PCD supported
[+]       Bit Rate: 847 kbit/s PICC -> PCD supported
[+]       Bit Rate: 212 kbit/s PICC <- PCD supported
[+]       Bit Rate: 424 kbit/s PICC <- PCD supported
[+]       Bit Rate: 847 kbit/s PICC <- PCD supported
[+] Max Frame Size: 256 bytes
[+]  Protocol Type: Protocol is compliant with ISO/IEC 14443-4
[+] Frame Wait Integer: 8 - 8192 ETUs | 77312 us
[+]  App Data Code: Application is Standard
[+]  Frame Options: NAD is not supported
[+]  Frame Options: CID is supported
[+] Tag :
[+]   Max Buf Length: 1 (MBLI) 
[+]   CID : 0

[+] 14443-3b tag found:

[+] unknown tag type answered to a 0x000b3f80 command ans:
```

hf search report "unknown tag type answered to a 0x000b3f80 command ans:" what does it mean?

Factory test, don't returns 6666, but this 
```
[usb] pm3 --> hf 14a raw -s -c -t 1000 CF00000000CC
[+] 00 00 00 03 AB [ E7 31 ]
```

**Expected behavior**
Read configuration without issue

**Desktop (please complete the following information):**
 - OS: linux
 - hw version
```
 [usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  Iceman/master/v4.15864-78-gc88c3bc4f 2022-11-26 08:25:47 2644e31b5
  compiled with............. GCC 10.2.1 20210110
  platform.................. Linux / x86_64
  Readline support.......... present
  QT GUI support............ absent
  native BT support......... absent
  Python script support..... present
  Lua SWIG support.......... present
  Python SWIG support....... present

 [ PROXMARK3 ]
  firmware.................. PM3 GENERIC

 [ ARM ]
  bootrom: Iceman/master/v4.15864-31-gbde4e8d75 2022-11-12 09:13:54 2afcb5732
       os: Iceman/master/v4.15864-78-gc88c3bc4f 2022-11-26 08:25:58 2644e31b5
  compiled with GCC 11.2.1 20220111

 [ FPGA ] 
  LF image 2s30vq100 2022-03-23 17:21:05
  HF image 2s30vq100 2022-03-23 17:21:16
  HF FeliCa image 2s30vq100 2022-03-23 17:21:27
  HF 15 image 2s30vq100 2022-03-23 17:21:38

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 59% used )
```
 - hw status
```
 [usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 42724
[#]   Available memory........ 40416
[#] Tracing
[#]   tracing ................ 0
[#]   traceLen ............... 189
[#] Current FPGA image
[#]   mode.................... HF image 2s30vq100 2022-03-23 17:21:16
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] 
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  31 |  20 |  18 |  50 |  15 | N/A | N/A | 
[#]     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A | 
[#]               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A | 
[#]    1 of 4 coding reference |  31 |  20 |  18 |  34 |  15 |  50 |  66 | 
[#] 
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 274432
[#]   Transfer Speed PM3 -> Client... 548864 bytes/s
[#] Various
[#]   Max stack usage......... 4088 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... 84
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 29948 Hz
[#] Installed StandAlone Mode
[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[#] 
```
 - data tune
```
 [=] ---------- LF Antenna ----------
[+] LF antenna: 27.48 V - 125.00 kHz
[+] LF antenna: 20.25 V - 134.83 kHz
[+] LF optimal: 27.48 V - 125.00 kHz
[+] Approx. Q factor (*): 6.9 by frequency bandwidth measurement
[+] Approx. Q factor (*): 8.0 by peak voltage measurement
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[+] HF antenna: 16.87 V - 13.56 MHz
[+] Approx. Q factor (*): 4.9 by peak voltage measurement
[+] HF antenna is OK

(*) Q factor must be measured without tag on the antenna

[+] Displaying LF tuning graph. Divisor 88 (blue) is 134.83 kHz, 95 (red) is 125.00 kHz.
```

DidierA, 2022-11-28:
Hello,
interesting, where did you get this card?

et-ness, 2022-11-30:
> Hello, interesting, where did you get this card?

lab401

DidierA, 2022-12-01:
Funny, I got two from them 2 months ago and they behave as described in the notes (30 bytes returned)
lab401 also links here for documentation...

et-ness, 2022-12-01:
it was out of stock for a time.
does your version emulate a tag 14b?

DidierA, 2022-12-02:
Yes

iceman1001, 2022-12-04:
make a PR with your patch?

DidierA, 2022-12-05:
Please test PR #1827 . 
it contains your patch to hf_mfu_ultimatecard, and `hf 14a info` should detect your card as magic gen4.

commit 2f4167ee (2022-12-05): Consider different behavior of some Gen4 cards. Should fix #1822
commit 989b830a (2022-12-05): Merge pull request #1827 from DidierA/magic_32

Consider different behavior of some Gen4 cards. Should fix #1822
closed 2022-12-05 by iceman1001