RfidResearchGroup/proxmark3 #1837: card is not vulnerable to Darkside attack

xbebhxx3, 2022-12-09:
**Describe the bug**
When I use `hf mf darkside` to obtain the cuid card password, it returns

    [-] card is not vulnerable to Darkside attack (its random number generator seems to be based on the wellknown
    [-] generating polynomial with 16 effective bits only, but shows unexpected behaviour

**To Reproduce**
1. Put a fully encrypted cuid card on proxmark3
2. run `hf mf darkside`

**Expected behavior**
obtain the cuid card password

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: windows11
 - hw version
[image]

 - hw status
[image]

 - data tune
 
[image]

xbebhxx3, 2022-12-09:
- hf search
[image]

iceman1001, 2022-12-09:
In what why is there a bug? 
Use the forum or discord server for user related questions