RfidResearchGroup/proxmark3 #1843: Compilation error on openSuse Leap 15.3

nitraiolo, 2022-12-17:
Compilation was working till a couple of weeks ago (sorry I didn't save the git pull log).

Now I get a this compilation error:

src/cmdscript.c: In function ‘CmdScriptRun’:
src/cmdscript.c:406:9: error: unknown type name ‘PyConfig’
         PyConfig py_conf;
         ^~~~~~~~
src/cmdscript.c:407:9: error: implicit declaration of function ‘PyConfig_InitIsolatedConfig’ [-Werror=implicit-function-declaration]
         PyConfig_InitIsolatedConfig(&py_conf);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
src/cmdscript.c:407:9: error: nested extern declaration of ‘PyConfig_InitIsolatedConfig’ [-Werror=nested-externs]
src/cmdscript.c:411:16: error: request for member ‘use_environment’ in something not a structure or union
         py_conf.use_environment = 1;
                ^
src/cmdscript.c:412:16: error: request for member ‘user_site_directory’ in something not a structure or union
         py_conf.user_site_directory = 1;
                ^
src/cmdscript.c:419:9: error: implicit declaration of function ‘PyConfig_SetBytesArgv’ [-Werror=implicit-function-declaration]
         PyConfig_SetBytesArgv(&py_conf, argc + 1, argv);
         ^~~~~~~~~~~~~~~~~~~~~
src/cmdscript.c:419:9: error: nested extern declaration of ‘PyConfig_SetBytesArgv’ [-Werror=nested-externs]
src/cmdscript.c:421:9: error: implicit declaration of function ‘PyConfig_SetBytesString’; did you mean ‘PyMapping_SetItemString’? [-Werror=implicit-function-declaration]
         PyConfig_SetBytesString(&py_conf, &py_conf.home, getenv("PYTHONHOME"));
         ^~~~~~~~~~~~~~~~~~~~~~~
         PyMapping_SetItemString
src/cmdscript.c:421:9: error: nested extern declaration of ‘PyConfig_SetBytesString’ [-Werror=nested-externs]
src/cmdscript.c:421:51: error: request for member ‘home’ in something not a structure or union
         PyConfig_SetBytesString(&py_conf, &py_conf.home, getenv("PYTHONHOME"));
                                                   ^
src/cmdscript.c:423:9: error: implicit declaration of function ‘Py_InitializeFromConfig’; did you mean ‘Py_InitializeEx’? [-Werror=implicit-function-declaration]
         Py_InitializeFromConfig(&py_conf);
         ^~~~~~~~~~~~~~~~~~~~~~~
         Py_InitializeEx
src/cmdscript.c:423:9: error: nested extern declaration of ‘Py_InitializeFromConfig’ [-Werror=nested-externs]
src/cmdscript.c:426:9: error: implicit declaration of function ‘PyConfig_Clear’; did you mean ‘PyErr_Clear’? [-Werror=implicit-function-declaration]
         PyConfig_Clear(&py_conf);
         ^~~~~~~~~~~~~~
         PyErr_Clear
src/cmdscript.c:426:9: error: nested extern declaration of ‘PyConfig_Clear’ [-Werror=nested-externs]
cc1: all warnings being treated as errors
make[1]: *** [Makefile:920: obj/cmdscript.o] Error 1
make: *** [Makefile:158: client/all] Error 2

In the old versions, "client/src/cmdscript.d" was referencing "/usr/include/python3.6m/pyconfig.h" and I've verified that it's still there.

This is my environment:

===================================================================
Version info:      Iceman/master/v4.15864-144-ge049a7c5a
Platform name:     Proxmark3 generic target
PLATFORM:          PM3GENERIC
PLATFORM_FPGA:     xc2s30
PLATFORM_SIZE:     512
Platform extras:   No extra selected
Included options:  LF HITAG EM4x50 EM4x70 ZX8211 GENERAL_HF ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   LF_SAMYRUN
===================================================================
[*] MAKE client/all
===================================================================
Version info:      Iceman/master/v4.15864-144-ge049a7c5a
Client platform:   Linux
GUI support:       QT5 found, enabled (Qt version 5.12.7 in /usr/lib64)
native BT support: Bluez found, enabled
Jansson library:   system library not found, using local library
Lua library:       system library not found, using local library
Python3 library:   Python3 v3.6 found, enabled
Readline library:  enabled
Whereami library:  system library not found, using local library
Lua SWIG:          wrapper found
Python SWIG:       wrapper found
compiler version:  cc (SUSE Linux) 7.5.0
===================================================================

Thanks!

iceman1001, 2022-12-17:
You have an old Python installation,  try a more modern one?

nitraiolo, 2022-12-17:
I've done all updates available. I prefer to use the system one instead installing another one side by side. I'm trying to find what triggers this error.

iceman1001, 2022-12-17:
there is a change in the python3 libs..   We adapted it and it works on 3.10 on onwards

jmichelp, 2022-12-17:
Well, according to Python's website, python3.6 is end-of-life and not supported anymore(https://www.python.org/downloads/release/python-3615/):
```
The release you are looking at is Python 3.6.15, the final security bugfix release for the legacy 3.6 series which has now reached end-of-life and is no longer supported.```

nitraiolo, 2022-12-17:
I understand it.
Only as info also the new openSuSE Leap 15.4 uses 3.6. Even the update to the latest version will not solve problem for OpenSuSE users.

nitraiolo, 2022-12-18:
This last commit is intended to help maintenance. This way the "default" when condition is not met (#else part) is always the new code.

iceman1001, 2022-12-18:
they should really update their python version.  
Anyway the repo compiles nicely now with the changes.  

Thanks!

closed 2022-12-18 by iceman1001

nitraiolo, 2022-12-18:
I agree.
Python release on openSuSE Leap has merged to the SLES release so they maintain that version as LTS for Enterprise users.
I hope they will change it in the future 15.5 release.

Thanks to you!