RfidResearchGroup/proxmark3 #1848: Build fail on termux (mbedtls, unused variable)

alexcher-im, 2022-12-25:
I'm trying to build the client (`make client`) on termux (aarch64 device), and getting compilation error:
```
bignum.c:1444:29: error: variable 't' set but not used [-Werror,-Wunused-but-set-variable]
    mbedtls_mpi_uint c = 0, t = 0;
                            ^
1 error generated.
```
Manually removing variable and it's write-usage fixes the problem (`common/mbedtls/bignum.c` lines `1444` and `1490`)

Javimoya03, 2023-01-09:
Hello, did you find a way to fix this?

iceman1001, 2023-01-14:
Would this work?
```
    mbedtls_mpi_uint c = 0, t = 0;
    (void)t;
```

Javimoya03, 2023-01-15:
but where do i write that?

Javimoya03, 2023-01-15:
Nothing, its says:

bignum.c:1444:36: error: expected expression
    mbedtls_mpi_uint c = 0, t = 0 (void)t;
                                   ^
1 error generated.
make[2]: *** [../../Makefile.host:95: ../../client/obj/mbedtls/bignum.o] Error 1
make[1]: *** [Makefile:867: obj/libmbedtls.a] Error 2
make: *** [Makefile:158: client/all] Error 2

iceman1001, 2023-01-15:
the suggested fix has been pushed to the repo,  so you pull latest and compile

Javimoya03, 2023-01-17:
> the suggested fix has been pushed to the repo,  so you pull latest and compile

Thanks bro!!

closed 2023-01-22 by iceman1001