RfidResearchGroup/proxmark3 #1849: No GUI in Build

NordicSailer, 2022-12-26:
Hi,

I am running the build on Ubuntu. Everything works brilliant, except the GUI.

When I run HW tune, it does not produce the graph and says "No GUI in this build."

It looks like qt-dev has been deprecated, but I did an install of the libutils-qt-dev.

Any suggestion on what to do to get the GUI to work?

Thank you.

iceman1001, 2022-12-26:
You followed the guide and it didnt work?

NordicSailer, 2022-12-27:
I did....I even uninstalled and reinstalled it twice. My Google-foo works pretty good, but it could not find a solution for this.
I really think it has to do with the deprecated QT library.
The program works great, other than the GUI build.
Didn't know if this was a common occurance?
NOTE: OS -> Ubuntu 22.04.1 LTS

iceman1001, 2022-12-28:
Odd, I have no issues what so ever with the GUI build on my Ubuntu build machines. 
Since you followed the guide its hard to tell what you did wrong.

iceman1001, 2023-01-14:
Progress?

NordicSailer, 2023-01-16:
Not yet. Had been preoccupied with other things.

Will let you know when I try it again.

Thank you for the follow up.

closed 2023-01-29 by iceman1001