RfidResearchGroup/proxmark3 #1853: Doc: the description of GTU/GDM-IC is incorrect.

taichunmin, 2023-01-01:
**Describe the bug**
I think the description of `GDM-IC/GTU` cards in the [Magic Cards Notes](https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/magic_cards_notes.md#multi) document is incorrect because I recently purchased several `GDM-IC/GTU` cards on taobao and found that Proxmark3 does not support writing to these cards at all. After some research, I discovered that these types of cards are actually designed to avoid being written to by card readers, so I think the description in the document is wrong.

**Expected behavior**
Maybe be can edit the document?

**Screenshots**
[image]

**Additional context**
* [滚动码复制卡GTU/GUID卡复制教程](https://nfcreader.ourword.press/2020/01/08/%E6%BB%9A%E5%8A%A8%E7%A0%81%E5%A4%8D%E5%88%B6%E5%8D%A1gtu-guid%E5%8D%A1%E5%A4%8D%E5%88%B6%E6%95%99%E7%A8%8B/)

wh201906, 2023-01-01:
> Proxmark3 does not support writing to these cards at all.

Maybe the tags you purchased don't use the default password?

taichunmin, 2023-01-01:
I think the GTU is not equal to Gen4 card.

[image]

[image]

wh201906, 2023-01-01:
I guess It's hard to know what exactly `GDM-IC/GTU` card is without sniffing the communication between these tags and the special readers designed for `GDM-IC/GTU` card.
As you said, `GDM-IC/GTU` card might not be equal to Gen4 card. But It's also possible that they are just Gen4 card with different passwords, as [the document](https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/magic_cards_notes.md#identify-15) says
> The card will be identified only if the password is the default one.

iceman1001, 2023-01-01:
The claims for names and versions that is used in marketing or product description on those sites doesn't match what we are trying to do in the magic note document where we are trying to have a naming convention and identification.   
In order to see how to use the tag given the different possibilities of libnfc/pm3/smart phone.

The only way to forward is to ask the seller about it.  If they are selling their _unique_ software it comes down to sniffing and reversing it in order to extract what commands and how they are used.

taichunmin, 2023-01-01:
I still trying to buy a reader for these cards.

iceman1001, 2023-01-01:
For those who want to look at the software,  I downloaded it from site and put a copy here
Its both android APK and their application.

http://www.icesql.se/download/files/gtu40_apk_and_app.rar

iceman1001, 2023-01-08:
I will close this issue and when someone finds out more,  it can be reopened.

closed 2023-01-08 by iceman1001

taichunmin, 2023-01-15:
Here's the GDM trace of write:

```
[=] Session log C:\Users\taich/.proxmark3/logs/log_20230115.txt
[+] loaded from JSON file C:\Users\taich/.proxmark3/preferences.json
[+] execute command from commandline: trace list -u -t mf

[=] Using UART port COM4
[=] Communicating with PM3 over USB-CDC
[usb|script] pm3 --> trace list -u -t mf
[=] downloading tracelog data from device
[+] Recorded activity (trace len = 7901 bytes)
[=] start = start of start frame end = end of frame. src = source of transfer
[=] ISO14443A - all times are in microseconds

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
        0.0 |       77.9 | Rdr |26(7)                                                                    |     | REQA
      165.5 |      340.1 | Tag |04  00                                                                   |     | 
      442.5 |      624.2 | Rdr |93  20                                                                   |     | ANTICOLL
      711.8 |     1146.0 | Tag |de  77  52  ac  57                                                       |     | 
     1243.7 |     2015.3 | Rdr |93  70  de  77  52  ac  57  49  5c                                       |  ok | SELECT_UID
     2107.7 |     2367.3 | Tag |08  b6  dd                                                               |     | 
     2469.6 |     2821.2 | Rdr |50  00  57  cd                                                           |  ok | HALT
     3876.1 |     3949.3 | Rdr |40(7)                                                                    |     | MAGIC WUPC1
    17933.9 |    18007.1 | Rdr |40(7)                                                                    |     | MAGIC WUPC1
    34368.1 |    34446.0 | Rdr |26(7)                                                                    |     | REQA
    34533.6 |    34708.3 | Tag |04  00                                                                   |     | 
    34810.6 |    34992.3 | Rdr |93  20                                                                   |     | ANTICOLL
    35079.9 |    35514.2 | Tag |de  77  52  ac  57                                                       |     | 
    35611.8 |    36383.5 | Rdr |93  70  de  77  52  ac  57  49  5c                                       |  ok | SELECT_UID
    36475.8 |    36735.4 | Tag |08  b6  dd                                                               |     | 
    36837.8 |    37189.4 | Rdr |50  00  57  cd                                                           |  ok | HALT
    38234.8 |    38308.0 | Rdr |40(7)                                                                    |     | MAGIC WUPC1
    52292.6 |    52365.8 | Rdr |40(7)                                                                    |     | MAGIC WUPC1
    68737.5 |    68815.3 | Rdr |26(7)                                                                    |     | REQA
    68902.9 |    69077.6 | Tag |04  00                                                                   |     | 
    69179.9 |    69361.7 | Rdr |93  20                                                                   |     | ANTICOLL
    69449.3 |    69883.5 | Tag |de  77  52  ac  57                                                       |     | 
    69981.1 |    70752.8 | Rdr |93  70  de  77  52  ac  57  49  5c                                       |  ok | SELECT_UID
    70846.3 |    71105.9 | Tag |08  b6  dd                                                               |     | 
    71207.1 |    71558.7 | Rdr |80  00  6c  92                                                           |  ok | 
    71704.1 |    72053.4 | Tag |a9  5a  83  fd                                                           |     | 
    72152.2 |    72843.7 | Rdr |47  ed  9e  b7! d2! 21! c0! 42!                                          | !crc| 
    72931.3 |    73280.5 | Tag |f0! 9b  2b! 90!                                                          |     | 
    73378.2 |    73729.8 | Rdr |b8! 50! 0c  47!                                                          | !crc| 
    73817.4 |    73859.9 | Tag |0a(3)                                                                    |     | 
    73962.2 |    75498.5 | Rdr |06! f4  71! f2  84  f3! 00  b5! 7c  b1! 11! 55  48! e0  08  c5  8a  16!  | !crc| 
    78800.3 |    78842.8 | Tag |0b(3)                                                                    |     | 
    78945.1 |    79292.0 | Rdr |50! 45! 2e  fc                                                           | !crc| HALT
    79441.0 |    79785.5 | Tag |f4! c4  50! 0f                                                           |     | 
    79889.1 |    80575.8 | Rdr |d5  91  23  d2! 95! c7  f2  6f                                           | !crc| 
    80668.1 |    81017.4 | Tag |4f  33  f9! 0f!                                                          |     | 
    81115.0 |    81461.9 | Rdr |e8  38! bf! 6b!                                                          | !crc| 
    81554.3 |    81596.8 | Tag |0d(3)                                                                    |     | 
    81699.1 |    83240.1 | Rdr |32! 12! cd  1b! 8d! 23  af  36! 37  b5  bb! ed  9c! 71  c8! ca! f8! a6!  | !crc| 
    86537.2 |    86584.4 | Tag |04(4)                                                                    |     | 
[... cut ...]
   561399.4 |   562940.4 | Rdr |67  cc! 30! c0  ab! 9d  88  8b  6f! a3! d2  01! 69  df  a2! 4f! f2! 8c   | !crc| 
   566238.6 |   566285.8 | Tag |03(4)                                                                    |     | 
   566382.3 |   566733.9 | Rdr |f9  c1! e9  d0                                                           | !crc| 
   566832.2 |   568366.1 | Tag |55! ae  ab! b4! 7b  8a  e7! 85  33  08! ba  59! fd  49  7f! 46  21! 4a!  | !crc| 
   576501.5 |   576574.6 | Rdr |40(7)                                                                    |     | MAGIC WUPC1

```

taichunmin, 2023-01-15:
I guess the GDM card is similar to CUID card. But replace real auth block command to `0x80`.

taichunmin, 2023-01-15:
Here is another trace of write to GDM card:

```
[=] Session log C:\Users\taich/.proxmark3/logs/log_20230108.txt
[+] loaded from JSON file C:\Users\taich/.proxmark3/preferences.json
[+] execute command from commandline: trace list -u -t mf

[=] Using UART port COM4
[=] Communicating with PM3 over USB-CDC
[usb|script] pm3 --> trace list -u -t mf
[=] downloading tracelog data from device
[+] Recorded activity (trace len = 9727 bytes)
[=] start = start of start frame end = end of frame. src = source of transfer
[=] ISO14443A - all times are in microseconds

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
        0.0 |      174.6 | Tag |04  00                                                                   |     | 
   499424.2 |   499598.8 | Tag |04  00                                                                   |     | 
   998874.3 |   999049.0 | Tag |04  00                                                                   |     | 
  1498324.5 |  1498499.1 | Tag |04  00                                                                   |     | 
  1997582.0 |  1997655.2 | Rdr |52(7)                                                                    |     | WUPA
  1997748.7 |  1997923.3 | Tag |04  00                                                                   |     | 
  2497194.1 |  2497368.7 | Tag |04  00                                                                   |     | 
  2996632.4 |  2996807.1 | Tag |04  00                                                                   |     | 
  3495908.8 |  3495982.0 | Rdr |52(7)                                                                    |     | WUPA
  3496074.3 |  3496249.0 | Tag |04  00                                                                   |     | 
  3995505.6 |  3995680.2 | Tag |04  00                                                                   |     | 
  4494782.0 |  4494855.2 | Rdr |52(7)                                                                    |     | WUPA
  4494947.5 |  4495122.1 | Tag |04  00                                                                   |     | 
  4994231.3 |  4994304.4 | Rdr |52(7)                                                                    |     | WUPA
  4994397.6 |  4994572.3 | Tag |04  00                                                                   |     | 
  5493686.1 |  5493759.3 | Rdr |52(7)                                                                    |     | WUPA
  5493852.5 |  5494027.1 | Tag |04  00                                                                   |     | 
  5993133.9 |  5993207.1 | Rdr |52(7)                                                                    |     | WUPA
  5993297.9 |  5993472.6 | Tag |04  00                                                                   |     | 
  6492575.8 |  6492649.0 | Rdr |52(7)                                                                    |     | WUPA
  6492742.2 |  6492916.8 | Tag |04  00                                                                   |     | 
  6992009.4 |  6992082.6 | Rdr |52(7)                                                                    |     | WUPA
  6992174.6 |  6992349.3 | Tag |04  00                                                                   |     | 
  7491449.0 |  7491522.1 | Rdr |52(7)                                                                    |     | WUPA
  7491615.3 |  7491790.0 | Tag |04  00                                                                   |     | 
  7990910.9 |  7990984.1 | Rdr |52(7)                                                                    |     | WUPA
  7991074.9 |  7991249.6 | Tag |04  00                                                                   |     | 
 10488180.5 | 10488253.7 | Rdr |52(7)                                                                    |     | WUPA
 10503479.6 | 10503557.5 | Rdr |26(7)                                                                    |     | REQA
 10987623.6 | 10987696.8 | Rdr |52(7)                                                                    |     | WUPA
 11002922.7 | 11003000.6 | Rdr |26(7)                                                                    |     | REQA
 11487085.5 | 11487158.7 | Rdr |52(7)                                                                    |     | WUPA
 11487250.7 | 11487425.4 | Tag |04  00                                                                   |     | 
 11487509.1 | 11487587.0 | Rdr |26(7)                                                                    |     | REQA
 11986555.8 | 11986628.9 | Rdr |52(7)                                                                    |     | WUPA
 11986719.8 | 11986894.4 | Tag |04  00                                                                   |     | 
 12486014.2 | 12486087.3 | Rdr |52(7)                                                                    |     | WUPA
 12486179.4 | 12486354.0 | Tag |04  00                                                                   |     | 
 12985453.7 | 12985526.8 | Rdr |52(7)                                                                    |     | WUPA
 12985620.1 | 12985794.7 | Tag |04  00                                                                   |     | 
[... cut ...]
 27869470.2 | 27869644.8 | Tag |04  00                                                                   |     | 
 28368729.2 | 28368802.4 | Rdr |52(7)                                                                    |     | WUPA
 28368896.8 | 28369071.4 | Tag |04  00                                                                   |     | 
 28868169.9 | 28868243.1 | Rdr |52(7)                                                                    |     | WUPA
 28868337.5 | 28868512.1 | Tag |04  00                                                                   |     | 
```

wh201906, 2023-01-15:
Would you like to post the trace logs in the Discord server? I guess people there would love to check these files.

iceman1001, 2023-01-15:
Wow,  great sniffs.
I love to see the trace files.  

``
... sniff...
trace save -f hf_mf_gmd...
```

The findings with 0x80  is good one.   Very annoying to now have to handle all cases of AUTH with another command in the project.

taichunmin, 2023-01-15:
[hf_mf_gmd_copykey.zip](https://github.com/RfidResearchGroup/proxmark3/files/10419488/hf_mf_gmd_copykey.zip)