RfidResearchGroup/proxmark3 #1854: Incorrect simulation of Mifare 1K card

szjiajin, 2023-01-03:
**Describe the bug**
According to this [data sheet](https://www.nxp.com/docs/en/data-sheet/MF1S50YYX_V1.pdf), in chapter 7.8.3 at page 13, at the end of table 8 the foot note says:

> If key B may be read in the corresponding Sector Trailer it cannot serve for authentication (see grey marked lines in Table
7). As a consequences, if the reader authenticates any block of a sector which uses such access conditions for the Sector
Trailer and using key B, the card will refuse any subsequent memory access after authentication.

It means that if the access bit (C1|C2|C3) for sector trailer is 0b000 or 0b001 or 0b010, the response should be an NACK.

But when I simulate a mifare card, I set the access bit of sector trailer to be 001 and perform authenticate with key B followed by a read operation, it seems to return 16 byte value with some of the bytes masked of if not accessible.
I checked the source code in [proxmark3/armsrc/mifaresim.c](https://github.com/RfidResearchGroup/proxmark3/blob/master/armsrc/mifaresim.c) from line 887 to line 983. It does not seem to implement this feature.

**To Reproduce**
Steps to reproduce the behavior:
1. Obtain a normal mifare 1K card, change the access bit of one of the sector to 0xFF078069 (the transport configuration)
2. Authenticate with key B for one block in that sector and try to read the block, after the authentication is completed, the card will return an error code (0x01 or 0x05, I am not sure) for the ReadBlock command.
3. Reproduce it in the simulated card in pm3, it returns the value of the data block (or the masked value of the sector trailor as keyA and access bit is not accesssible), but not an error code.

**Expected behavior**
As mentioned above, it should return an error code instead of 16 byte value.

**Screenshots**
I think my explanation should be enough, but I will upload my screenshot of sniffing if necessary.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - inside proxmark3 client run the following commands and paste the output here.
 I am not with my workbench now, will submit the version later, but this should still be the issue in the latest firmware.
 - hw version
 - hw status
 - data tune

**Additional context**
I also checked repository of official proxmark but their repository stopped updating 2 years ago, so I decided to report the issue here.

iceman1001, 2023-01-03:
I am sure there are many more cases of the Proxmark3 simulation isn't 100% according to the datasheet.

Feel free to contribute!

iceman1001, 2023-01-08:
When fiddling with the proxmark3 simulation code its important to consider timings. 
Every now and then we have to choose between functional complete vs available clock ticks to execute code.  The simulation is very time critical.

closed 2023-10-13 by iceman1001