RfidResearchGroup/proxmark3 #1861: hf_mfu_amiibo_restore succeed but invalid for the switch

saltbo, 2023-01-08:
**To Reproduce**
`script run hf_mfu_amiibo_restore -f myfile -k password`

**Expected behavior**
Restore a valid amiibo card

**Extra**

Hello @gtalusan, thanks for your scripts about amiibo.

I use the script `hf_mfu_amiibo_sim.lua` is very great, It can sim an amiibo card to use. 

When I use the script `hf_mfu_amiibo_restore.lua`, It restores success, but I don't know why it can't be recognized by Switch.

I reviewed your script code with the post articles https://farewell-ladmin.com/backing-up-your-amiibo-with-a-proxmark3/ and https://forum.dangerousthings.com/t/cloning-amiibo-onto-new-cards/11585. I have not found any problem...

With all due respect, is there no problem for you to successfully clone it using this script?

gtalusan, 2023-01-08:
Both work great for me and my main test console has been the Nintendo Switch, in particular Breath of the Wild.

I have tested briefly with a Nintendo Wii U using Yoshi's Woolly World and both the simulator and restored NTAG215 worked as expected.

If you're using the "pregen" Amiibos, you'll need to first encrypt them using a null UID.

saltbo, 2023-01-09:
> If you're using the "pregen" Amiibos, you'll need to first encrypt them using a null UID.

How to encrypt using a null UID?  I using the script https://github.com/RfidResearchGroup/proxmark3/blob/master/client/pyscripts/amiibo_change_uid.py to change the  UID

gtalusan, 2023-01-09:
Sorry, I was thinking about something different when I said null UID.

The proxmark3 scripts will only write/restore a binary as if it were read from an existing Amiibo.  This means the data is assumed to be encrypted when estoring to a card.

So for a pregen Amiibo, your first step will be to encrypt it before using the script to restore it to an NTAG215.

Use a tool such as `amiitool` to encrypt and sign your pregen Amiibo since they're decrypted.

saltbo, 2023-01-09:
Would you happen to know pyamiibo?  The [amiibo_change_uid.py](https://github.com/RfidResearchGroup/proxmark3/blob/master/client/pyscripts/amiibo_change_uid.py) base on pyamiibo.

I thought it was the same as the `amiitool`. Does the `amiibo_change_uid.py` not contain encrypt logic? 
I thought it was already encrypted because it works fine for the sim script.

gtalusan, 2023-01-09:
They serve the same functionality.  You may use PyAmiibo as well.

If your Amiibo data is decrypted then it must be encrypted before writing it to a tag.

The UID and Amiibo identification blocks are NOT encrypted.  If you are using software that only checks those pieces then it'll succeed.  However, software that checks game data blocks will assume it's encrypted -- so if you have not encrypted the data properly, then the game may not work correctly with that tag.

If you dump a real Amiibo then the game data will be encrypted.  This would be the perfect input into the proxmark3 scripts.

saltbo, 2023-01-09:
[image]

I use the amiitool build a bin file, it's same as the pyamiibo gen

gtalusan, 2023-01-09:
Right, I have no idea if your originating data is correct to begin with.

Next step would be to restore it to the card using the new card's PWD.

saltbo, 2023-01-09:
My Amiibo data is encrypted. And it's dumped from a real Amiibo. 

My NTAG215 can not modify the UID, so I think I only need to update the UID of bin file.

I run `pm3 -c "script run amiibo_change_uid 0491DF01464503 wolf-dump-noh.bin wolf-dump-pm3.bin key_retail.bin"` to get a new bin file, 

Then run "script run hf_mfu_amiibo_restore -f wolf-dump-pm3.bin"

It's right?

gtalusan, 2023-01-09:
The restore script will run amiibo_change_uid for you.

You only need to provide the new card's PWD and path to the bin file.

saltbo, 2023-01-09:
> If you dump a real Amiibo then the game data will be encrypted. This would be the perfect input into the proxmark3 scripts.

I try restore the dump data, happened errors：
```bash
[usb] pm3 --> script run hf_mfu_amiibo_restore -f hf-mfu-045A397ADF6180-dump.bin -k FFFFFFFF
[+] executing lua /usr/local/Cellar/proxmark3/4.15864/bin/../share/proxmark3/luascripts/hf_mfu_amiibo_restore.lua
[+] args '-f hf-mfu-045A397ADF6180-dump.bin -k FFFFFFFF'
----------------------------------------
----------------------------------------
Loading data from hf-mfu-045A397ADF6180-dump.bin
game: nil
[-] ⛔ error - .../../share/proxmark3/luascripts/hf_mfu_amiibo_restore.lua:119: attempt to index field '?' (a nil value)

[+] finished hf_mfu_amiibo_restore
```

Should I remove the header data?

saltbo, 2023-01-09:
header
----
00040402
01001103
01000086
92580B4C
45A9C42F
A90145CE
5E5F9C43
09A43D47
D232A3D1
68CBADE6
7F8185C6
00000000
00000000
00000000

data
------
045a39ef
7adf6180
44480fe0
f110ffee
a5000600
dea80bdd
b13c3a52
a239506d
4ce2778d
cae45f43
952c4b26
99408148
a5656e82
58e2ecda
27d4c63d
9cb6ccfb
c47288d3
b6071142
3ce4e5c9
4b51aac3
63592e3c
01030000
024f0902
0d123604
f59bce70
bcaf328b
b4592bf0
e339b27d
f0ac8417
ee91f6d9
9ea425d1
725453d3
eb963364
deedc629
c0cd4854
dd4d60c5
93ae34d9
1ae1c0c6
8a88f72f
24a82df5
9966d7a7
793829a8
6c1a7f21
504e75ae
b1143d6f
7582277d
7b01b925
b50bdf9b
630b537d
1bd3403f
b252d7bf
4fb5b2c9
0cedd864
d4e8413b
3beaa78f
2f4a8209
d4acd8d5
b2f04762
3372f6a8
2b261d9f
6aba3e31
be47f549
ba5048c9
268a5c92
7decad03
854e6f5d
76f720ea
f5d85d98
8be6b669
114abdcd
ea9d0aa5
02c6e165
53cb91e3
16aa97c1
e90e2ae6
a479b967
324e0e96
6e315eb3
4ca502cf
b56996e8
33a39b78
7f519a4d
292ebe13
3102a548
a9f81434
722aec69
0139f8c7
306590a5
6ae0b2cd
911520f9
e0d16d32
8ecbf149
240c225b
6d0b40d1
602c45f7
6ed5f192
43c68da7
95a99472
6beba208
993ef007
bf6c6b69
d252136b
5a4d3004
5abdcdab
6d7e3ad0
e3944f20
2978e87a
5bfbc20b
7ecc3a41
bb8f75eb
84cdadcc
33f75533
27ed9335
870c06e6
abfad8f7
39ed30be
50c5a5c0
b877e97c
91f5af3d
03537227
e032ae43
ad3f7b34
16479283
e6818d66
c6f4b019
d2455443
f8f8236f
991c0dd7
26f9eb88
0dc3fa34
01000fbd
00000004
5f000000
8ab3b10a
80800000

gtalusan, 2023-01-09:
Looks like an off-by-1.

Code change here: https://github.com/RfidResearchGroup/proxmark3/pull/1863

Dumping an Amiibo:
```
[usb] pm3 --> hf mfu dump -k 7F4CD4ED
[+] TYPE: NTAG 215 504bytes (NT2H1511G0DU)  
[+] Reading tag memory...

[=] MFU dump file information
[=] -------------------------------------------------------------
[=] Version..... 00 04 04 02 01 00 11 03 
[=] TBD 0....... 00 00 
[=] TBD 1....... 00 
[=] Signature... 9F 87 F9 58 18 B2 4E 90 7D 41 00 96 CE B9 A7 4A 
[=]              B0 9E 3A 1A FA 30 E7 B2 AC BD AF 4E 89 68 38 04 
[=] Counter 0... 00 00 00 
[=] Tearing 0... 00 
[=] Counter 1... 00 00 00 
[=] Tearing 1... 00 
[=] Counter 2... 00 00 00 
[=] Tearing 2... 00 
[=] Max data page... 133 ( 536 bytes )
[=] Header size..... 56 bytes

[=] -------------------------------------------------------------
[=] block#   | data        |lck| ascii
[=] ---------+-------------+---+------
[=]   0/0x00 | 04 CA 20 66 |   | .. f
[=]   1/0x01 | 1F 39 61 81 |   | .9a.
[=]   2/0x02 | C6 48 0F E0 |   | .H..
[=]   3/0x03 | F1 10 FF EE | 1 | ....
[=]   4/0x04 | A5 95 3C 00 | 0 | ..<.
[=]   5/0x05 | 25 35 1B 6F | 0 | %5.o
[=]   6/0x06 | C4 DC 57 93 | 0 | ..W.
[=]   7/0x07 | A0 62 C1 7A | 0 | .b.z
[=]   8/0x08 | 91 D0 2C 88 | 0 | ..,.
[=]   9/0x09 | 14 CE 2F 16 | 0 | ../.
[=]  10/0x0A | CE 44 F6 1D | 0 | .D..
[=]  11/0x0B | AD 3E 3F CD | 0 | .>?.
[=]  12/0x0C | F8 E3 A8 5B | 0 | ...[
[=]  13/0x0D | 2F 44 FA 86 | 1 | /D..
[=]  14/0x0E | 2B E7 B9 DE | 1 | +...
[=]  15/0x0F | CD 35 9B 0F | 1 | .5..
[=]  16/0x10 | 24 C6 4F A4 | 0 | $.O.
[=]  17/0x11 | 4A 40 D9 C6 | 0 | J@..
[=]  18/0x12 | 68 48 CD 71 | 0 | hH.q
[=]  19/0x13 | 88 11 60 ED | 0 | ..`.
[=]  20/0x14 | 04 33 8A 34 | 0 | .3.4
[=]  21/0x15 | 01 01 03 00 | 0 | ....
[=]  22/0x16 | 04 14 09 02 | 0 | ....
[=]  23/0x17 | 0D 12 51 06 | 0 | ..Q.
[=]  24/0x18 | F8 50 37 6D | 0 | .P7m
[=]  25/0x19 | C9 54 A0 CA | 0 | .T..
[=]  26/0x1A | 85 25 17 D2 | 0 | .%..
[=]  27/0x1B | 9F C5 2B 92 | 0 | ..+.
[=]  28/0x1C | 8F 8D 3E D0 | 0 | ..>.
[=]  29/0x1D | 03 C0 02 84 | 0 | ....
[=]  30/0x1E | EE C5 60 40 | 0 | ..`@
[=]  31/0x1F | 13 53 70 C1 | 0 | .Sp.
[=]  32/0x20 | 3C A3 73 A0 | 0 | <.s.
[=]  33/0x21 | C2 E1 7F FA | 0 | ....
[=]  34/0x22 | 82 F4 D5 CE | 0 | ....
[=]  35/0x23 | EA F7 F9 10 | 0 | ....
[=]  36/0x24 | DE C3 AB 04 | 0 | ....
[=]  37/0x25 | 25 EB D4 45 | 0 | %..E
[=]  38/0x26 | 86 DC 2B AB | 0 | ..+.
[=]  39/0x27 | 5B 58 95 13 | 0 | [X..
[=]  40/0x28 | ED BD 72 14 | 0 | ..r.
[... 264 lines cut ...]
[=] 121/0x79 | 58 D6 97 25 | 0 | X..%
[=] 122/0x7A | 83 D2 A9 35 | 0 | ...5
[=] 123/0x7B | 7D 0C EE 86 | 0 | }...
[=] 124/0x7C | BA 52 0E 66 | 0 | .R.f
[=] 125/0x7D | 66 01 E1 87 | 0 | f...
[=] 126/0x7E | AA 9B 5F D0 | 0 | .._.
[=] 127/0x7F | 78 43 CB 79 | 0 | xC.y
[=] 128/0x80 | 8C 5D 09 1C | 0 | .]..
[=] 129/0x81 | D5 05 F8 A7 | 0 | ....
[=] 130/0x82 | 01 00 0F BD | 0 | ....
[=] 131/0x83 | 00 00 00 04 | 0 | ....
[=] 132/0x84 | 5F 00 00 00 | 0 | _...
[=] 133/0x85 | 58 1D D4 CC | 0 | X...
[=] 134/0x86 | 80 80 00 00 | 0 | ....
[=] ---------------------------------
[=] Restoring data blocks.
[=] ...............................................................................................................................
[=] Restore finished
[=] Special block: 3 (0x03) [ F1 10 FF EE ]
[=] Using pwd FF FF FF FF 
[+] isOk:01
[=] Block: 134 (0x86) [ 80 80 00 00 ]
[=] Using pwd FF FF FF FF 
....
```

saltbo, 2023-01-09:
It looks like it has the same effect as my manual removal.

But the Switch can not identify the new card...

saltbo, 2023-01-10:
If anyone has the same problem as me, please feedback there...

gtalusan, 2023-01-10:
It's quite difficult to understand what's happening with very little context.

From your screenshot, it appears you have 4 dumps from 4 different cards.  There's not much use in comparing the differing bytes since some of those bytes are derived from the UID.  PWD which you've circled at offset 532 in particular is derived from the UID.

The bytes circled with ??? are generated by PyAmiibo/amiitool or whatever tool you used to flash the tag.

In your last two dumps, you have the magic bytes F110FFEE at byte offset 4.  These is the incorrect location.

From an Amiibo dump that I have laying on my hard disk, you can see F110FFEE is at offset 12.  This would correspond to block 3 in hf_mfu_restore_amiibo.

```
% xxd -g4 0.bin
00000000: 04107ae6 9a8e4b81 de480fe0 f110ffee  ..z...K..H......
00000010: a5000300 97c660ac ecb959bc ff57b0c1  ......`...Y..W..
00000020: 8228cf93 f4dbf96d 707e6aaa 605f0710  .(.....mp~j.`_..
00000030: 76843e5d db86cbb7 5d2f1bd5 85e8722c  v.>]....]/....r,
00000040: ebe6a1b4 7783cd4d ab999e5b 4133285a  ....w..M...[A3(Z
00000050: 3b3e9627 01030000 024f0902 0d12c50b  ;>.'.....O......
00000060: 9c4b2e64 83593d14 1bb38afd ee5f635f  .K.d.Y=......_c_
00000070: 4a2b70bd d918404c fda928cd 34a9a371  J+p...@L..(.4..q
00000080: 2c9c573a cb7ea3b1 012836ab e76972fa  ,.W:.~...(6..ir.
00000090: 2ad29828 ee908a67 5a6face3 1fcb031b  *..(...gZo......
000000a0: ead666e8 9f7d80b6 6346f0b8 257a2240  ..f..}..cF..%z"@
000000b0: 58a6ebe2 40e6e206 b9c96b64 97a47656  X...@.....kd..vV
000000c0: 5458732a 5c6777ff 5d69bcfa 6848f42f  TXs*\gw.]i..hH./
000000d0: f81201b6 304adb9f e58c25c8 45255923  ....0J....%.E%Y#
000000e0: d82df7e8 9d00383a 8ac15350 dee3db26  .-....8:..SP...&
000000f0: 3c5fbd7d 5c675b80 e3e64541 35be6b7e  <_.}\g[...EA5.k~
00000100: 19162008 e7e9ee7d 2e829a1f 4bb32601  .. ....}....K.&.
00000110: 5eb16677 41922cae bc885df4 0d96077d  ^.fwA.,...]....}
00000120: 9d737b93 828d589b 73828b21 95ed3366  .s{...X.s..!..3f
00000130: c3d0ef91 c24dc042 ff9d9fcb 41267251  .....M.B....A&rQ
00000140: ec451808 620eb1b6 561961e4 e5bd59c3  .E..b...V.a...Y.
00000150: 47e1ef56 fab9ce3f 664132ea 24cc6d4c  G..V...?fA2.$.mL
00000160: 4cb8490d 8ce1261e a9e4d755 65eca687  L.I...&....Ue...
00000170: ad9a8abb 75759057 035b7857 0ce2196b  ....uu.W.[xW...k
00000180: 80c47694 6c76d6d9 194779cd 3bd29fec  ..v.lv...Gy.;...
00000190: dbfaa8bb bad92c86 17e22976 33cce16d  ......,...)v3..m
000001a0: 65d1187c 2a8d3a53 99e57eb5 b07fd21c  e..|*.:S..~.....
000001b0: 6e9f6461 0dfb3f3f b4677a20 d1fe05a1  n.da..??.gz ....
000001c0: 8d22d6bd 393d8c50 1fba30c9 837cfb72  ."..9=.P..0..|.r
000001d0: 6cb2d801 944ed679 27eef1f3 fba304fc  l....N.y'.......
000001e0: 4c1c44db 748ad018 e64e71a0 f0c9a25b  L.D.t....Nq....[
000001f0: f5b3eccb 532e3f0c 7b65775e de03ecb6  ....S.?.{ew^....
00000200: 6f1830d7 46f1a35b 01000fbd 00000004  o.0.F..[........
00000210: 5f000000 00000000 00000000           _...........
```

gtalusan, 2023-01-10:
Also, does "-de" happen to refer to decrypted?  If you're using amiitool or an amiitool-like program to decrypt your Amiibo data, then it re-shuffles the offsets of where data is stored.

The UID is the first block on an NTAG215.  After the encrypted dump is run through amiitool for decryption then you can see the UID shifting toward the end of the file.

So as I said previously, you must re-encrypt the file before giving it to the proxmark3 script.

Here's the code from amiitool that shifts the layout of the data blocks:

```
void nfc3d_amiibo_tag_to_internal(const uint8_t * tag, uint8_t * intl) {
        memcpy(intl + 0x000, tag + 0x008, 0x008);
        memcpy(intl + 0x008, tag + 0x080, 0x020);
        memcpy(intl + 0x028, tag + 0x010, 0x024);
        memcpy(intl + 0x04C, tag + 0x0A0, 0x168);
        memcpy(intl + 0x1B4, tag + 0x034, 0x020);
        memcpy(intl + 0x1D4, tag + 0x000, 0x008); ##### HERE THE UID IS COPIED FROM INTERNAL TO "HARDWARE" LAYOUT
        memcpy(intl + 0x1DC, tag + 0x054, 0x02C);
}

void nfc3d_amiibo_internal_to_tag(const uint8_t * intl, uint8_t * tag) {
        memcpy(tag + 0x008, intl + 0x000, 0x008);
        memcpy(tag + 0x080, intl + 0x008, 0x020);
        memcpy(tag + 0x010, intl + 0x028, 0x024);
        memcpy(tag + 0x0A0, intl + 0x04C, 0x168);
        memcpy(tag + 0x034, intl + 0x1B4, 0x020);
        memcpy(tag + 0x000, intl + 0x1D4, 0x008); # HERE THE HARDWARE LAYOUT UID IS COPIED TO INTERNAL LAYOUT
        memcpy(tag + 0x054, intl + 0x1DC, 0x02C);
}
```

saltbo, 2023-01-11:
I'm sorry I didn't make myself clear.

There are two cards, one can work, another one not. And got the` _de.bin` is decrypted from them

gtalusan, 2023-01-11:
TBH, I don't usually deal with the decrypted data so I can't offer help there.  All of my pre-generated Amiibo's have been encrypted already.

If I clone an existing Amiibo then writing its encrypted image to a fresh NTAG215 has always worked for me unless I get errors while writing to the card.  This usually happens if the tag is not positioned correctly.  If you watch for errors from proxmark3, then you can cancel it and re-flash the same card if you're quick enough.  Once the lock bytes have been written out, then parts of the tag become readonly.

You haven't shared any of your proxmark3 output.  Can you?  What game and Amiibos are you using?

saltbo, 2023-01-21:
I try using a big card, and it worked!!

Maybe because not fully restore the data into the small round card

saltbo, 2023-01-21:
One last question, Could we validate the data before writing the lock block?

saltbo, 2023-01-21:
```bash
[usb] pm3 --> script run hf_mfu_amiibo_restore -f hf-mfu-045A397ADF6180-dump.bin -k FFFFFFFF
[+] executing lua /usr/local/Cellar/proxmark3/4.15864/bin/../share/proxmark3/luascripts/hf_mfu_amiibo_restore.lua
[+] args '-f hf-mfu-045A397ADF6180-dump.bin -k FFFFFFFF'
----------------------------------------
----------------------------------------
Loading data from hf-mfu-045A397ADF6180-dump.bin
game: The Legend of Zelda
character: Midna & Wolf Link
type: Figure
series: Legend Of Zelda
generating new Amiibo binary for NTAG215 0443FA01B30503
[+] executing python /usr/local/Cellar/proxmark3/4.15864/bin/../share/proxmark3/pyscripts/amiibo_change_uid.py
[+] args '0443FA01B30503 /tmp/lua_FiFHcg.bin /tmp/lua_TKOLy1.bin /usr/local/Cellar/proxmark3/4.15864/bin/../share/proxmark3/resources/key_retail.bin'

[+] finished amiibo_change_uid

[+] loaded 540 bytes from binary file /tmp/lua_TKOLy1.bin
[=] Restoring /tmp/lua_TKOLy1.bin to card

[=] MFU dump file information
[=] -------------------------------------------------------------
[=] Version..... 00 00 00 00 00 00 00 00
[=] TBD 0....... 00 00
[=] TBD 1....... 00
[=] Signature... 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[=]              00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[=] Counter 0... 00 00 00
[=] Tearing 0... 00
[=] Counter 1... 00 00 00
[=] Tearing 1... 00
[=] Counter 2... 00 00 00
[=] Tearing 2... 00
[=] Max data page... 133 ( 536 bytes )
[=] Header size..... 56 bytes

[=] -------------------------------------------------------------
[=] block#   | data        |lck| ascii
[=] ---------+-------------+---+------
[=]   0/0x00 | 04 43 FA 35 |   | .C.5
[=]   1/0x01 | 01 B3 05 03 |   | ....
[=]   2/0x02 | B4 48 0F E0 |   | .H..
[=]   3/0x03 | F1 10 FF EE | 1 | ....
[=]   4/0x04 | A5 00 06 00 | 0 | ....
[=]   5/0x05 | B5 A8 6D 01 | 0 | ..m.
[=]   6/0x06 | 2E 94 EC 52 | 0 | ...R
[=]   7/0x07 | F7 8D 8F 43 | 0 | ...C
[=]   8/0x08 | C2 9C 05 B0 | 0 | ....
[=]   9/0x09 | A2 74 25 D5 | 0 | .t%.
[=]  10/0x0A | BA FF FD A9 | 0 | ....
[=]  11/0x0B | DB 6D 59 6A | 0 | .mYj
[=]  12/0x0C | AF 21 48 CA | 0 | .!H.
[=]  13/0x0D | FC C1 76 E6 | 1 | ..v.
[=]  14/0x0E | 5F E0 9E 47 | 1 | _..G
[=]  15/0x0F | 59 F3 01 89 | 1 | Y...
[=]  16/0x10 | F9 D9 C8 25 | 0 | ...%
[=]  17/0x11 | 48 1B EB D1 | 0 | H...
[=]  18/0x12 | 42 FF CE 6D | 0 | B..m
[=]  19/0x13 | B2 83 C3 BB | 0 | ....
[=]  20/0x14 | EB 2E 15 73 | 0 | ...s
[=]  21/0x15 | 01 03 00 00 | 0 | ....
[=]  22/0x16 | 02 4F 09 02 | 0 | .O..
[=]  23/0x17 | 0D 12 36 04 | 0 | ..6.
[=]  24/0x18 | F5 9B CE 70 | 0 | ...p
[=]  25/0x19 | BC AF 32 8B | 0 | ..2.
[=]  26/0x1A | B4 59 2B F0 | 0 | .Y+.
[=]  27/0x1B | E3 39 B2 7D | 0 | .9.}
[=]  28/0x1C | F0 AC 84 17 | 0 | ....
[=]  29/0x1D | EE 91 F6 D9 | 0 | ....
[=]  30/0x1E | 9E A4 25 D1 | 0 | ..%.
[... 312 lines cut ...]
[=] Using pwd FF FF FF FF
[+] isOk:01
[=] Block: 134 (0x86) [ 80 80 00 00 ]
[=] Using pwd FF FF FF FF
[+] isOk:01
[=] Block: 133 (0x85) [ E8 46 AE B9 ]
[=] Using pwd FF FF FF FF
[+] isOk:01
[=] Block: 131 (0x83) [ 00 00 00 04 ]
[=] Using pwd E8 46 AE B9
[+] isOk:01
[=] Block: 132 (0x84) [ 5F 00 00 00 ]
[=] Using pwd E8 46 AE B9
[+] isOk:01
[=] Special block: 2 (0x02) [ E8 48 0F E0 ]
[=] Using pwd E8 46 AE B9
[+] isOk:01
[=] Block: 130 (0x82) [ 01 00 0F BD ]
[=] Using pwd E8 46 AE B9
[+] isOk:01

[+] finished hf_mfu_amiibo_restore
```

I restore two cards at different heights, one work, another not.

gtalusan, 2023-01-21:
Interesting!  By heights do you mean different diameter/size?  I've been using the white coloured 2.5cm diameter NTAG215 cards - I have a batch from Aliexpress and another from Amazon with no problems, with a few dozen successful clones.

The `wrbl` command already waits for an ACK from the card that the data has been written.  In your above logs it's clear that no errors occurred so perhaps something else is wrong with the non-working cards.

Unfortunately, without the lock bytes written then I don't believe the Amiibo will function correctly either so it may not be a good way to determine a functioning card other than testing it with an Amiibo reading device.

iceman1001, 2023-01-29:
is this still an issue after the PR?

saltbo, 2023-02-01:
[image]

I have used all my cards, and only three cards work fine...

Because we only can restore once, I don't know the failed reason if it's the card or the location of the card during the recovery process.

iceman1001, 2023-02-01:
Why would you only be able to restore it once?    You got the pwd...

saltbo, 2023-02-02:
because the special block is set to lock...

iceman1001, 2023-02-02:
aha...  Long time I looked into this, but...  cant you modify the lock bytes and see if the switch allows it even it the blocks isn't locked?

and head over to the rfid hacking discord instead to ask questions..   there is a toy2life channel

saltbo, 2023-02-05:
I bought the third batch cards, it's worked fine. I think there is a quality problem with the two batches of cards I bought before.

closed 2023-02-05 by saltbo

Omarorj3, 2025-12-23:
> **To Reproduce**
> `script run hf_mfu_amiibo_restore -f myfile -k password`
> 
> **Expected behavior**
> Restore a valid amiibo card
> 
> **Screenshots**
> [image]
> 
> [image]
> 
> [image]
> 
> dump eml content
> ```bash
> 00040402
> 01001103
> 00000086
> 4829121E
> 80649494
> 4B56C266
> 2CF4DD2E
> 7FE50E10
> 96FE7E08
> 1E048CAA
> BCAEFC4A
> 00000000
> 00000000
> 00000000
> 0491E5F8
> 01EA4503
> AD480FE0
> F110FFEE
> A5000600
> 3CE83A1F
> 907079D2
> ED936922
> 235238B3
> 31D48C6C
> 26FA452F
> 80323FD8
> E5C890C7
> FA6E13BF
> 60A40B10
> 511C77AF
> 892B685D
> 0EC58BB7
> D9F4F00C
> 86306CBC
> EA099510
> 01030000
> 024F0902
> 0D123604
> F59BCE70
> BCAF328B
> B4592BF0
> E339B27D
> F0AC8417
> EE91F6D9
> 9EA425D1
> 725453D3
> 937BD2CE
> 7E8748E5
> 9DFEC870
> 12553F65
> ECDD9360
> C15AC77E
> F18AA040
> 2C4835E5
> EAE9F871
> 9D656E08
> 993AB0B6
> 72DC0F4F
> C8BA1241
> 415A1193
> AC7EB658
> B424FA75
> 04DBDFFE
> 8B208ED2
> 2E2D130E
> 3963493A
> 40153AD4
> 8F31DE32
> C686450A
> 9DC92512
> 73C6939C
> 3B911793
> 69A251D7
> E4589A7F
> 6ACA863E
> EFB403EF
> 1BE14E32
> 326319BD
> 7DD44C4B
> D9981862
> 7DA8D43F
> 7765C8E5
> 3C97807A
> 37E94176
> A527BC0C
> D456EE31
> AF4B3F81
> 081243E9
> 67DD4504
> B3EA201B
> E5AB5B56
> 1268CFB5
> B0ECDBBE
> 69A26A96
> 4E331264
> A6E2FF77
> 27F92B22
> 99277616
> 24B26C2F
> BA0B4C88
> F3900557
> 75C9808A
> D01A348D
> 603303EC
> DC1CB013
> 9A77B528
> B6C7C8F0
> E37690F5
> 769DADF5
> EAECB22D
> 059DEB72
> C97C1355
> C0EE84D1
> 89E3FF19
> A2134296
> 7124B7BC
> 4B574E0B
> 715F564E
> 8A4C5F62
> D9F164B5
> 33521253
> 8732E1DD
> BFD6FF1A
> B3850B17
> A14D84EC
> EDBF2318
> 96A98203
> 43E5734F
> CDC4D05C
> 6377ADB0
> D9565DEA
> 8476FC1F
> 29E4B294
> 93901E33
> B13EA257
> F538678B
> FEC97750
> 2BB8FA44
> 638FF9D8
> CD6824FB
> 6474C0E6
> B678594E
> 6359664F
> CEE4F96E
> 01000FBD
> 00000004
> 5F000000
> 3A5AEEBC
> 80800000
> ```
> 
> **Additional context**
> 
> Hello @gtalusan, thanks for your scripts about amiibo.
> 
> I use the script `hf_mfu_amiibo_sim.lua` is very great, It can sim an amiibo card to use. 
> 
> When I use the script `hf_mfu_amiibo_restore.lua`, It restores success, but I don't know why it can't be recognized by Switch.
> 
> I reviewed your script code with the post articles https://farewell-ladmin.com/backing-up-your-amiibo-with-a-proxmark3/ and https://forum.dangerousthings.com/t/cloning-amiibo-onto-new-cards/11585. I have not found any problem...
> 
> With all due respect, is there no problem for you to successfully clone it using this script?
>