RfidResearchGroup/proxmark3 #1866: cmdpiv build error

JohnDoeIsKIng, 2023-01-13:
Dear community,

with the actual repo I'm getting a build error:

`[-] CC src/cmdpiv.c
src/cmdpiv.c: In function ‘PivGetDataByCidAndPrint’:
src/cmdpiv.c:617:13: error: case label does not reduce to an integer constant
             case APDU_RES_SUCCESS:
             ^~~~
src/cmdpiv.c:624:13: error: case label does not reduce to an integer constant
             case APDU_RES_NOT_FOUND:
             ^~~~
src/cmdpiv.c:627:13: error: case label does not reduce to an integer constant
             case APDU_RES_SECURITY:
             ^~~~
Makefile:919: recipe for target 'obj/cmdpiv.o' failed
make[1]: *** [obj/cmdpiv.o] Error 1
Makefile:157: recipe for target 'client/all' failed
make: *** [client/all] Error 2
`

Build system is Ubuntu 18.04, bug appeared with the repo from yesterday.
Any hints ?
Regards

JD.

JohnDoeIsKIng, 2023-01-13:
Cheers Iceman,

sure:
`make PLATFORM=PM3GENERIC VERBOSE=1 all

===================================================================
Version info:      Iceman/master/v4.15864-199-g73a80fb07
Platform name:     Proxmark3 generic target
PLATFORM:          PM3GENERIC
PLATFORM_FPGA:     xc2s30
PLATFORM_SIZE:     512
Platform extras:   No extra selected
Included options:  LF HITAG EM4x50 EM4x70 ZX8211 GENERAL_HF ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   LF_SAMYRUN
===================================================================
[*] MAKE client/all
===================================================================
Version info:      Iceman/master/v4.15864-199-g73a80fb07
Client platform:   Linux
GUI support:       QT5 found, enabled (Qt version 5.9.5 in /usr/lib/x86_64-linux-gnu)
native BT support: Bluez not found, disabled
Jansson library:   system library not found, using local library
Lua library:       system library not found, using local library
Python3 library:   Python3 not found, disabled
Readline library:  enabled
Whereami library:  system library not found, using local library
Lua SWIG:          wrapper found
compiler version:  cc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
===================================================================
[-] CC src/cmdpiv.c
src/cmdpiv.c: In function ‘PivGetDataByCidAndPrint’:
src/cmdpiv.c:617:13: error: case label does not reduce to an integer constant
             case APDU_RES_SUCCESS:
             ^~~~
src/cmdpiv.c:624:13: error: case label does not reduce to an integer constant
             case APDU_RES_NOT_FOUND:
             ^~~~
src/cmdpiv.c:627:13: error: case label does not reduce to an integer constant
             case APDU_RES_SECURITY:
             ^~~~
Makefile:919: recipe for target 'obj/cmdpiv.o' failed
make[1]: *** [obj/cmdpiv.o] Error 1
Makefile:157: recipe for target 'client/all' failed
make: *** [client/all] Error 2
`

Hope that helps!
Best regards

JD.

jmichelp, 2023-01-13:
I'm not iceman, but I'll take it as a compliment 😁 

Thanks for the output. If this means you're compiling using gcc-7.5, it seems to be a pretty old compiler (2019), which matches your distribution (Ubuntu 18.04 LTS) which will go end-of-life in April this year.
You're probably better upgrading at that point, which will bring a more recent version of the compiler, without this false-positive bug (the reported piece of code is matching `uint16_t` against `uint16_t` so they all are integer constants).

iceman1001, 2023-01-14:
I swapped to DEFINES instead,  lets see if this solves your compilation error.   Even if I agree that you are on an ancient GCC version.  I believe the latest is GCC13.     I would manually upgrade it just in order to get a more modern one.

JohnDoeIsKIng, 2023-01-16:
Thanks, issue solved !
Perfect work!

Best regards

JD.

JohnDoeIsKIng, 2023-01-16:
Problem solved.

closed 2023-01-16 by JohnDoeIsKIng