RfidResearchGroup/proxmark3 #1868: [RELEASE 4.16078] Checklist - release name: Nitride

iceman1001, 2023-01-14:
# Checklist

- [ ] CHANGELOG.md: add title: `## [Nitride][2023-01-14]`
- [ ] `make style`
- [ ] `make miscchecks`
- [ ] `make clean; make client CC=clang CXX=clang++ LD=clang++` on recent Debian or Ubuntu
- [ ] `mymanualchecks.sh`
- [ ] `mycppcheck.sh` no alarming warning?
- [ ] `tools/build_all_firmwares.sh` check that the script contains all standalone modes then compile all standalone modes (linux only)
- [ ] `experimental_lib` compilation & tests
- [ ] `experimental_client_with_swig` compilation & tests
- [ ] Check Android `CMakeLists.txt` list of source file
- [ ] GitHub Actions - green across the board ( MacOS, Ubuntu, Windows)

# OS compilation and tests

```bash
#!/usr/bin/env bash

make clean && make -j PLATFORM=PM3GENERIC PLATFORM_EXTRAS= && tools/pm3_tests.sh --long || exit 1
make clean && make -j PLATFORM=PM3RDV4 PLATFORM_EXTRAS= || exit 1
make clean && make -j PLATFORM=PM3RDV4 PLATFORM_EXTRAS=BTADDON || exit 1
make -j PLATFORM=PM3RDV4 PLATFORM_EXTRAS=BTADDON && sudo make install PLATFORM=PM3RDV4 PLATFORM_EXTRAS=BTADDON && ( cd /tmp; proxmark3 -c 'data load -f lf_EM4x05.pm3;lf search -1'|grep 'Valid FDX-B ID found' ) && sudo make uninstall || exit 1

( cd client; rm -rf build; mkdir build;cd build;cmake .. && make -j PLATFORM=PM3GENERIC PLATFORM_EXTRAS= && cp -a ../*scripts ../*libs . && ../../tools/pm3_tests.sh --clientbin $(pwd)/proxmark3 client ) || exit 1
( cd client; rm -rf build; mkdir build;cd build;cmake .. && make -j PLATFORM=PM3RDV4 PLATFORM_EXTRAS= ) || exit 1
( cd client; rm -rf build; mkdir build;cd build;cmake .. && make -j PLATFORM=PM3RDV4 PLATFORM_EXTRAS=BTADDON ) || exit 1

# Hitag2crack, optionally with --long and --opencl ...
make hitag2crack/clean && make hitag2crack && tools/pm3_tests.sh hitag2crack || exit 1
```

- [ ] RPI Zero
- [ ] Jetson Nano
- [ ] WSL
- [ ] PSv3.10
- [ ] Archlinux
- [ ] Kali
- [ ] Debian Stable
- [ ] Debian Testing
- [ ] Ubuntu21
- [ ] ParrotOS
- [ ] Fedora
- [ ] OpenSuse Leap
- [ ] OpenSuse Tumbleweed
- [ ] OSX (MacPorts)
- [ ] OSX (Homebrew)
- [ ] Android
- [ ] Termux

# creating release

- [ ] `make release RELEASE_NAME="Nitride"`
  - last line of output,  gives you next command to run.
  - Sample:  `git push && git push origin v4.16078`
- [ ] CHANGELOG.md: edit title to add version info: `## [Nitride.4.16078][2023-01-14]`

## Step Github releases

- [ ] Go to Github releases,  create release based on the new created tag and publish

## Step Homebrew updates

- [ ] update homebrew repo, file `proxmark3.rb`
  - with a SHA256 sum of the file `v4.16078.tar.gz`
  - with updated list of standalone modes

## Step package maintains

- [ ] make a list of new standalone modes, so when we alert package maintainers they have a sporting chance of adding them

doegox, 2023-01-24:
I fixed a few things since my last tests, but I won't re-run all tests every time there are new changes... Please fix a RC and then we only fix things then make a release, but no new stuff.

iceman1001, 2023-01-24:
That is a fair request, 

As I know of now,  Its one more thing,  the preference.json saving ,  that is needed for a release.

iceman1001, 2023-01-29:
Alright,  done,  

all compiles on : 

RPI Zero
Jetson Nano
WSL
Archlinux
Kali
Ubuntu 22
ParrotOS
Fedora 37
OpenSuse Tumbleweed
Termux

the build_all firmware fails,   since the generic is too big for 256kb, 

I think its more or less ready for a release.   If you don't find anything,  I be releasing tomorrow

doegox, 2023-01-29:
> the build_all firmware fails, since the generic is too big for 256kb

Ok then we need to skip sth else for 256kb devices, so we can at least test the compilation of all the other firmwares. I chose arbitrarily to skip NFCBARCODE support: https://github.com/RfidResearchGroup/proxmark3/commit/2633a54e11670a0144bcf11ce308b3eba644773c

iceman1001, 2023-01-29:
Ok, 

There are some new entries for coverity I pushed fixes to,   and then its done.

iceman1001, 2023-01-29:
:(  skip nfcbarcode isn't enough.  
I added  `SKIP_ZX8211=1`  to make it work.

iceman1001, 2023-01-29:
New standalone modes in this release:
``` 
LF_NEDAP_SIM
LF_PROX2BRUTE
```

closed 2023-01-29 by iceman1001