RfidResearchGroup/proxmark3 #1870: Python pm3 API support broken in client with SWIG

doegox, 2023-01-15:
Hello, while preparing the release, I noticed this PR https://github.com/RfidResearchGroup/proxmark3/pull/1833 breaks the ability to run Python scripts from within the client and reach the pm3 API/module.
```sh
cd client/experimental_client_with_swig
./00make_swig.sh
./01make_client_cmake.sh
./02b_run_test_py.sh
[...]
Traceback (most recent call last):
  File "testembedded.py", line 3, in <module>
    import pm3
ModuleNotFoundError: No module named 'pm3'
```
while the commit just before this change is working on the same test.

According to https://docs.python.org/3/c-api/init_config.html#initialization-with-pyconfig `PyImport_AppendInittab()` must be moved between pre-initialization and initialization, so I tried to move it like this:
```c
        PyConfig_SetString(&py_conf, &py_conf.program_name, program);

#ifdef HAVE_PYTHON_SWIG
        // hook Proxmark3 API
        PyImport_AppendInittab("_pm3", PyInit__pm3);
#endif
        Py_InitializeFromConfig(&py_conf);
```
but that didn't change anything.

@jmichelp as you seem familiar with the new Python API and you are author of this merged PR, could you have a look ?

Thanks!

jmichelp, 2023-01-15:
Sure. I'll have a look and issue a fix during the week.

closed 2023-01-22 by iceman1001

doegox, 2023-01-24:
Thanks a lot @jmichelp !

PR #1873 (2023-01-22): Fix python initialization to have SWIG work
Fixes #1870