RfidResearchGroup/proxmark3 #1909: Compiling Error Makefile:57 Expecing ")")

SpeedGP, 2023-02-11:
Hi, I'm kind of a noob, I tried googling and searching the troubleshooting section, but couldn't find much.

Running kali

Sorry if I didn't provide enough info, let me know if I can help out any more.

┌──(james㉿DESKTOP-)-[~/proxmark3]
└─$ sudo make install         

===================================================================
Version info:      Iceman/master/v4.16191-46-g6df86578c
Platform name:     Proxmark3 RDV4
PLATFORM:          PM3RDV4
PLATFORM_FPGA:     xc2s30
PLATFORM_SIZE:     512
Platform extras:   No extra selected
Included options:  SMARTCARD FLASH -DRDV4 LF HITAG EM4x50 EM4x70 ZX8211 GENERAL_HF ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   LF_SAMYRUN
===================================================================
[*] MAKE client/install
===================================================================
Version info:      Iceman/master/v4.16191-46-g6df86578c
Client platform:   Linux
GUI support:       QT5 found, enabled (Qt version 5.15.8 in /usr/lib/x86_64-linux-gnu)
native BT support: Bluez found, enabled
Jansson library:   system library not found, using local library
Lua library:       system library not found, using local library
Python3 library:   Python3 v3.10 found, enabled
Readline library:  enabled
Whereami library:  system library not found, using local library
Lua SWIG:          wrapper found
Python SWIG:       wrapper found
compiler version:  cc (Debian 10.2.1-6) 10.2.1 20210110
===================================================================
[=] GEN src/version_pm3.c
[-] CC src/version_pm3.c
[*] MAKE deps/amiitool/libamiibo.a
make[2]: Nothing to be done for 'all'.
[*] MAKE deps/cliparser/libcliparser.a
PLATFORM Linux
make[2]: Nothing to be done for 'all'.
[*] MAKE deps/hardnested/libhardnested.a
make[2]: Nothing to be done for 'all'.
[*] MAKE deps/liblua/liblua.a for linux
make[3]: Nothing to be done for 'all'.
[*] MAKE deps/jansson/libjansson.a
make[2]: Nothing to be done for 'all'.
[*] MAKE obj/libmbedtls.a
make[2]: Nothing to be done for 'all'.
[*] MAKE deps/reveng/libreveng.a
make[2]: Nothing to be done for 'all'.
[*] MAKE deps/tinycbor/tinycbor.a
make[2]: Nothing to be done for 'all'.
[*] MAKE deps/whereami/libwhereami.a
make[2]: Nothing to be done for 'all'.
[=] CXX proxmark3
[@] Installing client to /usr/local...
[*] MAKE bootrom/install
compiler version:  arm-none-eabi-gcc (15:12.2.rel1-1) 12.2.1 20221205
[=] GEN version_pm3.c
[-] CC version_pm3.c
[=] LD obj/bootrom.elf
[=] GEN obj/bootrom.s19
[@] Installing bootrom to /usr/local...
[*] MAKE fpga_compress/install
[*] MAKE armsrc/install
compiler version:  arm-none-eabi-gcc (15:12.2.rel1-1) 12.2.1 20221205
[-] GEN version_pm3.c
[-] CC version_pm3.c
[=] LD obj/fullimage.stage1.elf
[-] GEN obj/fullimage.data.bin
[-] GEN obj/fullimage.data.bin.z
[=] LD obj/fullimage.elf
[=] GEN obj/fullimage.s19
[@] Installing fullimage to /usr/local...
[*] MAKE bootrom/all
compiler version:  arm-none-eabi-gcc (15:12.2.rel1-1) 12.2.1 20221205
[=] GEN version_pm3.c
[-] CC version_pm3.c
[=] LD obj/bootrom.elf
[=] GEN obj/bootrom.s19
[*] MAKE fpga_compress/all
make[1]: Nothing to be done for 'all'.
[*] MAKE armsrc/all
compiler version:  arm-none-eabi-gcc (15:12.2.rel1-1) 12.2.1 20221205
[-] GEN version_pm3.c
[-] CC version_pm3.c
[=] LD obj/fullimage.stage1.elf
[-] GEN obj/fullimage.data.bin
[-] GEN obj/fullimage.data.bin.z
[=] LD obj/fullimage.elf
[=] GEN obj/fullimage.s19
[*] MAKE recovery/install
[=] GEN bootrom.bin
[=] GEN fullimage.bin
[=] GEN proxmark3_recovery.bin
[@] Installing recovery to /usr/local...
[*] MAKE mfkey/install
[@] Installing mfkey32 mfkey32v2 mfkey64 to /usr/local...
[*] MAKE nonce2key/install
[@] Installing nonce2key to /usr/local...
[*] MAKE mf_nonce_brute/install
[@] Installing mf_nonce_brute mf_trace_brute to /usr/local...
[*] MAKE mfd_aes_brute/install
[@] Installing brute_key mfd_aes_brute mfd_multi_brute to /usr/local...
[*] MAKE client/all
===================================================================
Version info:      Iceman/master/v4.16191-46-g6df86578c
Client platform:   Linux
GUI support:       QT5 found, enabled (Qt version 5.15.8 in /usr/lib/x86_64-linux-gnu)
native BT support: Bluez found, enabled
Jansson library:   system library not found, using local library
Lua library:       system library not found, using local library
Python3 library:   Python3 v3.10 found, enabled
Readline library:  enabled
Whereami library:  system library not found, using local library
Lua SWIG:          wrapper found
Python SWIG:       wrapper found
compiler version:  cc (Debian 10.2.1-6) 10.2.1 20210110
===================================================================
[=] GEN src/version_pm3.c
[-] CC src/version_pm3.c
[*] MAKE deps/amiitool/libamiibo.a
make[2]: Nothing to be done for 'all'.
[*] MAKE deps/cliparser/libcliparser.a
PLATFORM Linux
make[2]: Nothing to be done for 'all'.
[*] MAKE deps/hardnested/libhardnested.a
make[2]: Nothing to be done for 'all'.
[*] MAKE deps/liblua/liblua.a for linux
make[3]: Nothing to be done for 'all'.
[*] MAKE deps/jansson/libjansson.a
make[2]: Nothing to be done for 'all'.
[*] MAKE obj/libmbedtls.a
make[2]: Nothing to be done for 'all'.
[*] MAKE deps/reveng/libreveng.a
make[2]: Nothing to be done for 'all'.
[*] MAKE deps/tinycbor/tinycbor.a
make[2]: Nothing to be done for 'all'.
[*] MAKE deps/whereami/libwhereami.a
make[2]: Nothing to be done for 'all'.
[=] CXX proxmark3
[*] MAKE recovery/all
[*] MAKE mfkey/all
make[1]: Nothing to be done for 'all'.
[*] MAKE nonce2key/all
make[1]: Nothing to be done for 'all'.
[*] MAKE mf_nonce_brute/all
make[1]: Nothing to be done for 'all'.
[*] MAKE mfd_aes_brute/all
make[1]: Nothing to be done for 'all'.
[@] Installing common resources to /usr/local...
ifneq (,)
/bin/sh: 1: Syntax error: word unexpected (expecting ")")
make: *** [Makefile:57: common/install] Error 2

ggallagh, 2023-02-11:
Makefile has been broken. Indentation at line 70 of Makefile implies a shell command but it isn't. Need to remove indentation for ifneq

iceman1001, 2023-02-11:
Pull latest and it should be fixed

closed 2023-02-11 by iceman1001

SpeedGP, 2023-02-12:
Thank You!!!

If anyone else finds this, it was line 70 and also line 74 need the indent removed.

commit 8f977ef9 (2023-02-11): fix #1909  intendation breaks the makefile