RfidResearchGroup/proxmark3 #1911: Segmentation Fault when running Python script in offline mode

pastacolsugo, 2023-02-13:
**Describe the bug**
Clients seg faults when executing a python script in offline mode.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a folder
2. Create this script (`a.py`)
```python
import pm3
p = pm3.pm3()
```

3. Cd to said folder
4. Run `pm3 --offline`
5. Run `script run a.py`

Output:

```
/usr/local/bin/pm3: row 248: 96831 Segmentation fault: 11  $CLIENT "$@"
````

**Expected behavior**
Client outputs an error to the screen and doesn't crash.

**Desktop (please complete the following information):**
 - OS: `Mac OS Monterey 12.6.2 (21G320)`
 - PM3: `Release v4.16191 - Nitride`
 - Python: `Python 3.10.9`
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version:
```
 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  Iceman/master/v4.16191 2023-01-29 20:35:17 3545f4f98
  compiled with............. Clang/LLVM Apple LLVM 14.0.0 (clang-1400.0.29.202)
  platform.................. OSX / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... absent
  Python script support..... present
  Lua SWIG support.......... present
  Python SWIG support....... present
```
 - hw status: `This command is not available in this mode`
 - data tune: `This command is not available in this mode`

**Additional context**
This problem came up trying to diagnose another issue with python script execution.

iceman1001, 2023-02-14:
I got all kinds of errors trying to run the  theremin.py script..   
needs some installing and adapting inside,   atleast now the client doesnt seqfault.

iceman1001, 2023-10-13:
I get the feeling you are trying to execute the swig build...  with a shared libary and you are calling the pm3 client which hasn't access to it...

[image]

so you need to look inside  `client/experimental_lib/example_py\`  folder and understand..

Setting it up like you wanted,   I can make it work...
[image]

closed 2023-10-13 by iceman1001