RfidResearchGroup/proxmark3 #1915: `pm3 --list`  doesnt work on WSL1 with more than one device

iceman1001, 2023-02-18:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Connect two proxmark3 devices to a WSL1,   run the `./pm3 --list`   and it doesn't give you two suggestions. 
```
iceman@TAU:~/pm3/henrygab$ ./pm3 --list
[!!] No port found
```

**Expected behavior**
```
./pm3 --list
1: /dev/ttyS5
2: /dev/ttyS3
```

**Desktop (please complete the following information):**
Latest source

**Additional context**
When looking at the comport identification part for WSL in pm3.sh file,

the output from the identification part doesn't do well in the for-loop,   since you get both com ports in one line.
```
function get_pm3_list_WSL {
    N=$1
    PM3LIST=()

    # Normal SERIAL PORTS (COM)
    for DEV in $($PSHEXE -command "Get-CimInstance -ClassName Win32_serialport | Where-Object {\$_.PNPDeviceID -like '*VID_9AC4&PID_4B8F*' -or \$_.PNPDeviceID -like '*VID_2D2D&PID_504D*'} | Select -expandproperty DeviceID" 2>/dev/null | tr -dc '[:print:]'); do
        _comport=$DEV
        DEV=$(echo $DEV | sed -nr 's#^COM([0-9]+)\b#/dev/ttyS\1#p')

```

With two devices hooked up,  the output is: 
```
iceman@TAU:~/pm3/henrygab$ powershell.exe -command "Get-CimInstance -ClassName Win32_serialport | Where-Object {\$_.PNPDeviceID -like '*VID_9AC4&PID_4B8F*' -or \$_.PNPDeviceID -like '*VID_2D2D&PID_504D*'} | Select -expandproperty DeviceID" 2>/dev/null | tr -dc '[:print:]'

COM5COM3
```

If I remove the `tr`  part it gives two lines,   not sure if that will help the for-loop..
```
iceman@TAU:~/pm3/henrygab$ powershell.exe -command "Get-CimInstance -ClassName Win32_serialport | Where-Object {\$_.PNPDeviceID -like '*VID_9AC4&PID_4B8F*' -or \$_.PNPDeviceID -like '*VID_2D2D&PID_504D*'} | Select -expandproperty DeviceID" 2>/dev/null
COM5
COM3
```

I have pushed some fixes for this behavior now

iceman1001, 2023-02-18:
Pushed some fixes for this,   should be working better on WSL-1 now. 
@henrygab   you might wanna test this out on WSL-2...

closed 2023-02-18 by iceman1001

henrygab, 2023-02-18:
I'm getting each device listed twice:

```
┌──(idic㉿q)-[~/proxmark3]
└─$ ./pm3 --list                                                                                                                      
1: /dev/ttyACM0
2: /dev/ttyACM0
3: /dev/ttyACM1
4: /dev/ttyACM1

┌──(idic㉿q)-[~/proxmark3]
└─$ ls -lFA /dev/ttyA* /dev/pm3*
lrwxrwxrwx 1 root root         7 Feb 18 12:31 /dev/pm3-0 -> ttyACM0
lrwxrwxrwx 1 root root         7 Feb 18 14:52 /dev/pm3-1 -> ttyACM1
crw-rw---- 1 root dialout 166, 0 Feb 18 12:33 /dev/ttyACM0
crw-rw---- 1 root dialout 166, 1 Feb 18 14:52 /dev/ttyACM1
```

I don't think this is intentional?  I'll wait to let things settle a bit, but... it's probably not quite right.

henrygab, 2023-02-18:
And we now get links that are constant under `/dev/serial/by-id/`

```
┌──(idic㉿q)-[~/proxmark3]
└─$ ls -lFA /dev/serial/by-id/
total 0
lrwxrwxrwx 1 root root 13 Feb 18 14:52 usb-proxmark.org_proxmark3_iceman__30************DC-if00 -> ../../ttyACM1
lrwxrwxrwx 1 root root 13 Feb 18 12:31 usb-proxmark.org_proxmark3_iceman__35************E1-if00 -> ../../ttyACM0
```

Success!

iceman1001, 2023-02-19:
Thanks for finding the bug...  
pushed a fix for it!