RfidResearchGroup/proxmark3 #1924: "hf 14a list" could not sniff traffic from reader

BensonYang1999, 2023-02-21:
**Describe the bug**
hf 14a sniff can not sniff the conversation from reader to card(Mifare Classic 1K).

**To Reproduce**
Steps to reproduce the behavior:
I used the version v4.15864 (also tried on v4.16191) trying to sniff the data from reader to card, steps are as follows:

1. launch pm3
2. run "hf 14a sniff"
3. place PM3 between reader and card
4. after reading, the led C is lit
5. use "trace list" or "hf list" to show the results

The results only contain data from tag to reader, no data from reader to card, which is annoying for decrypting.

**Expected behavior**
"hf 14a sniff" should be able to capture both direction traffic.

**Desktop (please complete the following information):**

- OS: Windows 11
- hw version
```
[ Proxmark3 RFID instrument ]

 [ CLIENT ]
  Iceman/master/v4.15864 2022-10-29 01:34:05 844ca091a
  compiled with............. MinGW-w64 10.3.0
  platform.................. Windows (64b) / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... absent
  Python script support..... present
  Lua SWIG support.......... present
  Python SWIG support....... present

 [ PROXMARK3 ]
```
- hw status
```
[#] Memory
[#]   BigBuf_size............. 42780
[#]   Available memory........ 42780
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode.................... HF image 2s30vq100 2022-03-23 17:21:16
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0
[#]
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
[#]     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
[#]               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A |
[#]    1 of 4 coding reference |  31 |  20 |  18 |  34 |  15 |  50 |  66 |
[#]
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[... 75 lines cut ...]
            |            |     |13                                                                       |     |
   20779200 |   20779392 | Tag |01                                                                       |     |
   20935296 |   20935552 | Tag |00!                                                                      |     |
   20990928 |   20991376 | Tag |07                                                                       |     |
   22016624 |   22025264 | Tag |03! 03! 2f  a3! f2  63! 20  0a!                                          |     |
   22031792 |   22031984 | Tag |01                                                                       |     |
   22431824 |   22474576 | Tag |02  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00   |     |
            |            |     |00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  90  00  5f   |     |
            |            |     |a2                                                                       |     |
   22876800 |   22877120 | Tag |03!                                                                      |     |
   23288896 |   23289088 | Tag |01                                                                       |     |
   23707424 |   23750112 | Tag |03  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00   |     |
            |            |     |00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  90  00  66   |     |
            |            |     |af                                                                       |     |
   27336176 |   27339696 | Tag |a3  6f  c6                                                               |     |
   30917488 |   30921008 | Tag |a3  6f  c6                                                               |     |
   32683600 |   32687120 | Tag |a3  6f  c6                                                               |     |
   34488432 |   34489072 | Tag |03!                                                                      |     |
   36264464 |   36267984 | Tag |a3  6f  c6                                                               |     |
   39832704 |   39833792 | Tag |a3!                                                                      |     |
   41584656 |   41588176 | Tag |a3  6f  c6                                                               |     |
   43346672 |   43350192 | Tag |a3  6f  c6                                                               |     |
   45085936 |   45089456 | Tag |a3  6f  c6                                                               |     |
   46826768 |   46827856 | Tag |a3!                                                                      |     |
```

iceman1001, 2023-02-22:
Looks more like a user related issue than a bug.   Better to ask user related questions in the discord server.

BensonYang1999, 2023-02-22:
> Looks more like a user related issue than a bug.   Better to ask user related questions in the discord server. 

OK!
Thanks!

closed 2023-02-23 by iceman1001