RfidResearchGroup/proxmark3 #1939: [RELEASE 4.16352] Checklist - release name: Seven

iceman1001, 2023-03-11:
# Checklist

- [ ] CHANGELOG.md: add title: `## [seven][YYYY-MM-DD]`
- [ ] `make style`
- [ ] `make miscchecks`
- [ ] `make clean; make client CC=clang CXX=clang++ LD=clang++` on recent Debian or Ubuntu
- [ ] `mymanualchecks.sh`
- [ ] `mycppcheck.sh` no alarming warning?
- [ ] `tools/build_all_firmwares.sh` check that the script contains all standalone modes then compile all standalone modes (linux only)
- [ ] `experimental_lib` compilation & tests
- [ ] `experimental_client_with_swig` compilation & tests
- [ ] GitHub Actions - green across the board ( MacOS, Ubuntu, Windows )

# OS compilation and tests

Run `tools/release_tests.sh` on:

- [ ] RPI Zero
- [ ] Jetson Nano
- [ ] WSL
- [ ] PSv3.10
- [ ] Archlinux
- [ ] Kali
- [ ] Debian Stable
- [ ] Debian Testing
- [ ] Ubuntu 22
- [ ] ParrotOS
- [ ] Fedora 37
- [ ] OpenSuse Leap
- [ ] OpenSuse Tumbleweed
- [ ] OSX (MacPorts)
- [ ] OSX (Homebrew)
- [ ] Android
- [ ] Termux

# creating release

- [ ] `make release RELEASE_NAME="seven"`
  - last line of output,  gives you next command to run.
  - Sample:  `git push && git push origin v4.16352`
- [ ] CHANGELOG.md: edit title to add version info: `## [seven.4.16352][YYYY-MM-DD]`

## Step Github releases

- [ ] Go to Github releases,  create release based on the new created tag and publish

## Step Homebrew updates

- [ ] update homebrew repo, file `proxmark3.rb`
  - with a SHA256 sum of the file `v4.16352.tar.gz`
  - with updated list of standalone modes

## Step package maintains

- [ ] make a list of new standalone modes, so when we alert package maintainers they have a sporting chance of adding them

SnakePin, 2023-03-11:
Tried the following command on Ubuntu 22.04 with Clang 17.0.0(9548af05):
`make clean; make client CC=clang CXX=clang++ LD=clang++`
and I just got a single warning, other than that it compiles fine.
```
src/flash.c:349:32: warning: cast from 'uint8_t *' (aka 'unsigned char *') to 'uint32_t *' (aka 'unsigned int *') increases required alignment from 1 to 4 [-Wcast-align]
            uint32_t offset = *(uint32_t *)(ctx->elf + le32(shdrs[i].sh_offset) + le32(shdrs[i].sh_size) - 4);
```

iceman1001, 2023-03-14:
Thank you for mention it.  We are aware of it existence when compiling with CLANG. 

The assigned array isnt checked for being evenly dividable with four so the compiler rightfully complains a bit

iceman1001, 2023-06-26:
@wh201906    
Nice android build script you made.

iceman1001, 2023-06-26:
Release done!
Thank you everyone for the efforts to bring this release out.

closed 2023-06-26 by iceman1001