RfidResearchGroup/proxmark3 #1944: HF antenna is UNUSABLE

ghecko, 2023-03-16:
**Describe the bug**

Hello, I obtain the "HF antenna is UNUSABLE" when I run the "hw tune". I have tried with 2 antenna (Iceman collection):

hw tune
```[=] ---------- Reminder ------------------------
[=] `hw tune` doesn't actively tune your antennas,
[=] it's only informative.
[=] Measuring antenna characteristics, please wait...
 🕚  10
[=] ---------- LF Antenna ----------
[+] LF antenna: 69.20 V - 125.00 kHz
[+] LF antenna: 34.28 V - 134.83 kHz
[+] LF optimal: 69.20 V - 125.00 kHz
[+] Approx. Q factor (*): 11.8 by frequency bandwidth measurement
[+] Approx. Q factor (*): 12.1 by peak voltage measurement
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[!] ⚠️  HF antenna is UNUSABLE

(*) Q factor must be measured without tag on the antenna

[+] Displaying LF tuning graph. Divisor 88 (blue) is 134.83 kHz, 95 (red) is 125.00 kHz.
```

**Desktop (please complete the following information):**

OS: Windows 11 (from WSL - same on Android Termux)

hw version

``` [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  Iceman/master/v4.16191-183-g1cc19c342 2023-03-16 14:47:58 051b138f6
  compiled with............. GCC 12.2.0
  platform.................. Linux / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... present
  Python script support..... present
  Lua SWIG support.......... present
  Python SWIG support....... present

 [ PROXMARK3 ]
  device.................... RDV4
  firmware.................. RDV4
  external flash............ present
  smartcard reader.......... present
  FPC USART for BT add-on... absent

 [ ARM ]
  bootrom: Iceman/master/v4.16191-183-g1cc19c342 2023-03-16 14:47:12 051b138f6
       os: Iceman/master/v4.16191-183-g1cc19c342 2023-03-16 14:47:49 051b138f6
  compiled with GCC 12.2.1 20221205

 [ FPGA ]
  LF image 2s30vq100 2022-03-23 17:21:05
  HF image 2s30vq100 2022-03-23 17:21:16
  HF FeliCa image 2s30vq100 2022-03-23 17:21:27
  HF 15 image 2s30vq100 2022-03-23 17:21:38

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 66% used )
```

hw status

```[#] Memory
[#]   BigBuf_size............. 40696
[#]   Available memory........ 40696
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode.................... HF image 2s30vq100 2022-03-23 17:21:16
[#] Flash memory
[#]   Baudrate................ 24 MHz
[#]   Init.................... OK
[#]   Memory size............. 2 mbits / 256 kb
[#]   Unique ID (be).......... 0x2671AFA782A867D5
[#] Smart card module (ISO 7816)
[#]   version................. v4.13
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0
[#]
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A |
[#]     long leading reference |  29 |  17 |  18 |  50 |  15 | N/A | N/A |
[#]               leading zero |  29 |  17 |  18 |  40 |  15 | N/A | N/A |
[#]    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 |
[#]
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 277504
[#]   Transfer Speed PM3 -> Client... 555008 bytes/s
[#] Various
[#]   Max stack usage......... 4088 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... -1
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 30839 Hz
[#] Installed StandAlone Mode
[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[#] Flash memory dictionary loaded
[#]   Mifare.................. 1508 / 2047 keys
[#]   T55x7................... 122 / 1023 keys
[#]   iClass.................. 28 / 511 keys
[#]
```

data tune
```[=] ---------- Reminder ------------------------
[=] `hw tune` doesn't actively tune your antennas,
[=] it's only informative.
[=] Measuring antenna characteristics, please wait...
 🕛   9
[=] ---------- LF Antenna ----------
[+] LF antenna: 69.50 V - 125.00 kHz
[+] LF antenna: 34.34 V - 134.83 kHz
[+] LF optimal: 69.50 V - 125.00 kHz
[+] Approx. Q factor (*): 11.8 by frequency bandwidth measurement
[+] Approx. Q factor (*): 12.1 by peak voltage measurement
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[!] ⚠️  HF antenna is UNUSABLE

(*) Q factor must be measured without tag on the antenna

[+] Displaying LF tuning graph. Divisor 88 (blue) is 134.83 kHz, 95 (red) is 125.00 kHz.
```

ghecko, 2023-03-16:
[image]
It's look like some component leaked here.
Where can I found schematics and part number to try replacing them?

iceman1001, 2023-03-19:
You would need to ask Proxgrind about that.
Closing, since its not a issue related to this firmware repo.

closed 2023-03-19 by iceman1001