RfidResearchGroup/proxmark3 #1953: M1 compilation Error

Talkwondo, 2023-03-30:
M1 mac can not compile when make -j.
 
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)

Intel works

iceman1001, 2023-03-30:
try cmake?

Talkwondo, 2023-04-01:
What is the option of the command i need to put?

gtalusan, 2023-04-03:
I have an M1 and it seems to work fine..

```
% make clean && make -j PLATFORM=PM3GENERIC

===================================================================
Version info:      Iceman/master/v4.16191-236-gbb4dad45d
Platform name:     Proxmark3 RDV4
PLATFORM:          PM3RDV4
PLATFORM_FPGA:     xc2s30
PLATFORM_SIZE:     512
Platform extras:   No extra selected
Included options:  SMARTCARD FLASH -DRDV4 LF HITAG EM4x50 EM4x70 ZX8211 GENERAL_HF ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   LF_SAMYRUN
===================================================================
[*] MAKE client/clean
===================================================================
Version info:      Iceman/master/v4.16191-236-gbb4dad45d
Client platform:   Darwin
GUI support:       QT5 found, enabled (Qt version 5.15.8 in /opt/homebrew/Cellar/qt@5/5.15.8_2/lib)
native BT support: Bluez not found, disabled
Jansson library:   system library not found, using local library
Lua library:       system library not found, using local library
Python3 library:   Python3 v3.11 found, enabled
Readline library:  enabled
Whereami library:  system library not found, using local library
Lua SWIG:          wrapper found
Python SWIG:       wrapper found
compiler version:  Apple clang version 13.1.6 (clang-1316.0.21.2.5)
===================================================================
PLATFORM Darwin
[*] MAKE bootrom/clean
[*] MAKE fpga_compress/clean
[*] MAKE armsrc/clean
[*] MAKE recovery/clean
[*] MAKE mfkey/clean
[*] MAKE nonce2key/clean
[*] MAKE mf_nonce_brute/clean
[*] MAKE mfd_aes_brute/clean

===================================================================
Version info:      Iceman/master/v4.16191-236-gbb4dad45d
Platform name:     Proxmark3 generic target
PLATFORM:          PM3GENERIC
PLATFORM_FPGA:     xc2s30
PLATFORM_SIZE:     512
Platform extras:   No extra selected
Included options:  LF HITAG EM4x50 EM4x70 ZX8211 GENERAL_HF ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   LF_SAMYRUN
===================================================================
[!] Platform definitions changed, cleaning bootrom/armsrc/recovery first...
[*] MAKE mfkey/all
[*] MAKE nonce2key/all
[*] MAKE mf_nonce_brute/all
[*] MAKE mfd_aes_brute/all
[-] CC nonce2key.c
[-] CC ../../common/crapto1/crypto1.c
[-] CC brute_key.c
[-] CC ../../common/crapto1/crapto1.c
[-] CC mfkey32.c
[-] CC util_posix.c
[-] CC ../../common/bucketsort.c
[-] CC ../../common/crapto1/crypto1.c
[-] CC randoms.c
[-] CC ../../common/crapto1/crapto1.c
[-] CC mf_nonce_brute.c
[-] CC mfd_aes_brute.c
[-] CC ../../common/crapto1/crypto1.c
[-] CC ../../common/bucketsort.c
[-] CC mfd_multi_brute.c
[... 429 lines cut ...]
[-] GEN obj/fpga_all.o
[=] AR libhardnested.a
[=] AR libjansson.a
[=] AR libreveng.a
[=] AR liblua.a
[=] AR libcliparser.a
[=] AR libmbedtls.a
[-] GEN version_pm3.c
[-] CC version_pm3.c
[=] LD obj/fullimage.stage1.elf
[-] GEN obj/fullimage.data.bin
[=] CXX proxmark3
[-] GEN obj/fullimage.data.bin.z
[=] LD obj/fullimage.elf
[=] GEN obj/fullimage.s19
[*] MAKE recovery/all
[=] GEN bootrom.bin
[=] GEN fullimage.bin
[=] GEN proxmark3_recovery.bin
```

```
% file client/proxmark3
client/proxmark3: Mach-O 64-bit executable arm64
```

iceman1001, 2023-04-06:
Since the issue with M1 and compiling findings are in contradiction,   where one can and another can't,    if someone has some more insights in M1 compilation fixes,  I will close this one

closed 2023-04-06 by iceman1001

lazybloke, 2024-03-11:
Same compile problem here : 

```
% make clean && make -j                    

===================================================================
Version info:      Iceman/master/v4.18218-86-gd0b3b8d6b
Platform name:     Proxmark3 generic target
PLATFORM:          PM3GENERIC
PLATFORM_FPGA:     xc2s30
PLATFORM_SIZE:     512
Platform extras:   No extra selected
Included options:  -DLED_ORDER_PM3EASY LF HITAG EM4x50 EM4x70 ZX8211 GENERAL_HF ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   LF_SAMYRUN
===================================================================
[*] MAKE client/clean
===================================================================
Version info:      Iceman/master/v4.18218-86-gd0b3b8d6b
Client platform:   Darwin
GUI support:       QT not found, disabled
native BT support: Bluez not found, disabled
Jansson library:   system library not found, using local library
Lua library:       system library not found, using local library
Python3 library:   Python3 v3.12 found, enabled
GD library:        GD v2.3.3 found, enabled
Readline library:  enabled
Whereami library:  system library not found, using local library
Lua SWIG:          wrapper found
Python SWIG:       wrapper found
compiler version:  Apple clang version 15.0.0 (clang-1500.3.9.4)
===================================================================
PLATFORM Darwin
[*] MAKE bootrom/clean
[*] MAKE fpga_compress/clean
[*] MAKE armsrc/clean
[*] MAKE recovery/clean
[*] MAKE mfkey/clean
[*] MAKE nonce2key/clean
[*] MAKE mf_nonce_brute/clean
[*] MAKE mfd_aes_brute/clean
[*] MAKE cryptorf/clean

===================================================================
Version info:      Iceman/master/v4.18218-86-gd0b3b8d6b
Platform name:     Proxmark3 generic target
PLATFORM:          PM3GENERIC
PLATFORM_FPGA:     xc2s30
PLATFORM_SIZE:     512
Platform extras:   No extra selected
Included options:  -DLED_ORDER_PM3EASY LF HITAG EM4x50 EM4x70 ZX8211 GENERAL_HF ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   LF_SAMYRUN
===================================================================
[*] MAKE client/all
[*] MAKE bootrom/all
[*] MAKE fpga_compress/all
[*] MAKE mfkey/all
[*] MAKE nonce2key/all
[*] MAKE mf_nonce_brute/all
[*] MAKE mfd_aes_brute/all
[*] MAKE cryptorf/all
[-] CC fpga_compress.c
[-] CC nonce2key.c
[-] CC ../../common/lz4/lz4hc.c
[-] CC ../../common/crapto1/crypto1.c
[-] CC ../../common/lz4/lz4.c
[-] CC ../../common/crapto1/crapto1.c
[-] CC ../../common/bucketsort.c
[-] CC mfkey32.c
[-] CC brute_key.c
[-] CC ../../common/crapto1/crypto1.c
[... 712 lines cut ...]
      _Pm3PyRun_SimpleFileNoExit in cmdscript.o
      _Pm3PyRun_SimpleFileNoExit in cmdscript.o
      _Pm3PyRun_SimpleFileNoExit in cmdscript.o
      _Pm3PyRun_SimpleFileNoExit in cmdscript.o
      _Pm3PyRun_SimpleFileNoExit in cmdscript.o
      _Pm3PyRun_SimpleFileNoExit in cmdscript.o
      _Pm3PyRun_SimpleFileNoExit in cmdscript.o
      ...
  "__Py_NoneStruct", referenced from:
      _Pm3PyRun_SimpleFileNoExit in cmdscript.o
      __wrap_delete_pm3 in pm3_pywrap.o
      __wrap_delete_pm3 in pm3_pywrap.o
      __wrap_pm3_name_get in pm3_pywrap.o
      __wrap_pm3_name_get in pm3_pywrap.o
      _pm3_swigregister in pm3_pywrap.o
      _pm3_swiginit in pm3_pywrap.o
      ...
  "__Py_NotImplementedStruct", referenced from:
      _SwigPyObject_richcompare in pm3_pywrap.o
      _SwigPyObject_richcompare in pm3_pywrap.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [proxmark3] Error 1
make: *** [client/all] Error 2
```