RfidResearchGroup/proxmark3 #1979: autopwn store incorrect key value when hardnested fail

reliveinfire, 2023-05-06:
Hi sir,
in some caes, the hf mf hardnested may fail to get the correct keys
in autopwn, it stores the incorrect key value
(i've use hf mf hardnested with script to find all correct keys)

case 1

[=]       28 |    1883 | Apply Sum(a8) and all bytes bitflip properties          |             nan |  nand
[+] target sector   0 key type B -- found valid key [ 000000000000 ]
correct one

[=]       35 |    2930 | Brute force phase completed.  Key found: 8642289466E5   |               0 |    0s
[+] target sector   2 key type A -- found valid key [ 8642289466E5 ]
but next fail case, it stores previous correct key

[=]       63 |    2210 | Apply Sum(a8) and all bytes bitflip properties          |             nan |  nand
[+] target sector   2 key type B -- found valid key [ 8642289466E5 ]

[=]       26 |    1662 | Apply Sum(a8) and all bytes bitflip properties          |             nan |  nand
[+] target sector   3 key type A -- found valid key [ 8642289466E5 ]

[=]       26 |    1662 | Apply Sum(a8) and all bytes bitflip properties          |             nan |  nand
[+] target sector   3 key type A -- found valid key [ 8642289466E5 ]

[=]       26 |    1662 | Apply Sum(a8) and all bytes bitflip properties          |             nan |  nand
[+] target sector   3 key type A -- found valid key [ 8642289466E5 ]
for nan cases, the fail found keys are supposed to store with nothing or special symbol.

is there better way to auto retry hf mf hardnest? i wrote one lua script to do this, but i dont know how to intercept the log.
i just try every valid key many times and use terminal find to check if proxmark found the key.

after trace the code, the autopwn did not re run the hardnest for the incorrect key,
the problems seem that found_key varible did not handle correctly in mfnestedhard

if the brute_force value is 0,
the correct key may be dropped in previous scan, the brute_force function can not find the correct key from the pending keys
the easy way to fix this, it may clear the found_key variable every time in autopwn loop
and autopwn can add new step for auto hardnest check for not found keys.

maybe we can add re-try mechanism or add retry parameter in hardnest command to recover this card type

i did this with script which issue hf md hardnest command. and it has chance to recover the correct key.
also, i found that test_grab.py in experimental_lib can parse the terminal log, it would be helpful to auto update the key when brute force run hf md hardnest in script.
thanks

iceman1001, 2023-05-06:
I pushed https://github.com/RfidResearchGroup/proxmark3/commit/31b01fff5a7634424d27123883fff47b341996fb   
would you mind checking if this solves your issue?

reliveinfire, 2023-05-06:
thanks a lot, it's working, the failure key would be stored with 000000000000.

maybe the autopwn should just stop the final dump if  any keys are 000000000000?
cause the dump value is invalid.

we can use the known keys to do further recovering.

closed 2023-05-06 by reliveinfire

iceman1001, 2023-05-06:
Not really,   only if both A/B keys are 00,  that sector is skipped.    Hence the partial text.

Autopwn does what it should do very well.   
If you discuss it more then the discord is the place for it.