RfidResearchGroup/proxmark3 #1980: [idea] hf mf hardnested force re-try

reliveinfire, 2023-05-06:
hf mf hardnested fails to get the correct key in some card type.
but if we re-try hardnested many times, it's possible to retrieve the correct key.
i guess in the hardnested flow, the correct key may be dropped due to physical signal reason, or Nonce is too random.

if hf mf hardnested can set re-try times, it would be very helpful.
and it stops at finding the key or counters are reaching target value.
the more features, it can auto-check every unknown keys, the user dont have to set unknown key manually.

iceman1001, 2023-05-06:
Easier to re-run the command instead of re-trying when it comes to hardnested.   Despite its name it only targets one key.

Normally for failure is static nonces, static encrypted nonces,  or bad coupling with tag.  
all of which can not be solved by re-running.

reliveinfire, 2023-05-06:
not quite sure why, i am newbie in proxmark3, just have this for 3 days.
but re-run hardnested, it did work.
my friend shared me the information, he said i can tried different known key and move the card.
so i wrote one easy script to re-run for every known keys, and move the card sometimes

reliveinfire, 2023-05-06:
do you mean, the bad couping is can be fixed with moving the card?
can we have one auto re-run hardnested,? user just need to move the card, dont have to issue the command, and wait for the result :)

iceman1001, 2023-05-06:
yes, bad coupling is dealt with trying different positions.

Normally hardnested runs for 5mins,  max 15min,  if longer than there is something else wrong.

reliveinfire, 2023-05-06:
every 5 mins, user need to click up arrow and enter, and check if hardnested is successful,
if it's successful, user need to change another target. the behavior is a little bit annoying.
prefer the more automation for this.

btw, use different known key in hardnested is useless, right? only try different positions work?

iceman1001, 2023-05-06:
We have autopwn for that.   

Your claim that hardnest recovers keys that autopwn does not,  is confusing since both use the same code.

reliveinfire, 2023-05-06:
do you mean that autopwn keeps previous result? all i need to do is re-run autopwn until all keys recovered?

iceman1001, 2023-05-06:
no,  I mean autopwn should find all keys in one run.    

if you are experiencing that it recovers partial keys and different ones with each run,   then its bad coupling that is the cause.

Find the sweetspot,  run the command and done.

reliveinfire, 2023-05-06:
ok i got it, i'll try it.
but in this case, why not just support keep keys in autopwn cache? 
if the card has 32 keys, it takes long time to complete one round, and every trial it waste time to re-calcuate previous known result.

iceman1001, 2023-05-06:
there is nothing saying you can't make a dictionary file,   and use it as input for the next run....

reliveinfire, 2023-05-06:
oh, i think this is supposed to be the correct usage for this case.  this is very helpful for my case. thanks.
although this is done manully:
copy and paste into dictionary file with editor,
start next round, then check which keys are new manually,
copy paste, then loop
in my opniton, your autopwn design goal is to auto complete tasks.
i'll try to make this more automative.

closed 2023-05-06 by reliveinfire

reliveinfire, 2023-05-06:
from your comment, maybe add new keys into dic file is easy to implement.

reliveinfire, 2023-05-06:
flow could be:
1. check if all keys found at the end
2. if not, add keys to a uid-dictionary
3. re start the autopwn procedure
until all keys found, dump the card content

it seems that this does not need to modify much original flow.

reliveinfire, 2023-05-06:
one idea is that it could infinite waiting at the fail end
if there are ways to notify user to move the card, maybe like board beep, or laptop system information.
after move the card, user can trigger to start next round :D