RfidResearchGroup/proxmark3 #1981: hitag write  copy proxton fob

sisaladin, 2023-05-07:
hello 
I am trying to copy proxton fob 
I need manage to write page 4 and page   6 but when I did try to write  page  7 and page   5 its come FFFFFFFF 
any idea why ?
thank you

[=] --- Tag Information ---------------------------
[=] -------------------------------------------------------------
[+]      UID: C0155713
[+]     TYPE: PCF 7936
[=] ------------------------------------------------------------- 

[usb] pm3 --> lf hitag reader --21 -k BDF5E846
[+]  UID: c0155713

[=] Hitag2 tag information

[=] ------------------------------------
[+] Config byte : 0x06 [ 00000110 ]
[+] Encoding    : Manchester
[+] Version     : Hitag2
[+] Coding in HITAG 2 operation: manchester
[+] Tag is in   : Password mode
[+] Page 6,7    : RW
[+] Page 4,5    : RW
[+] Page 3      : RW
[+] Page 1,2    : RW
[=] ------------------------------------
[=] 00 | C0 15 57 13 | ..W.
[=] 01 | BD F5 E8 46 | ...F
[=] 02 | 20 F0 4F 4E |  .ON
[=] 03 | 06 F9 07 C2 | ....
[=] 04 | 84 03 01 08 | ....
[=] 05 | FF FF FF FF | ....
[=] 06 | 0C 04 88 40 | ...@
[=] 07 | FF FF FF FF | ....
[=] 08 | 00 00 00 00 | ....
[=] 09 | 00 00 00 00 | ....
[=] 10 | 00 00 00 00 | ....
[=] 11 | 00 00 00 00 | ....
[=] -------- Possible de-scramble patterns ---------
[+] Paxton id: 102365 | 0x18fdd

jenningsreeve, 2023-05-09:
Could be coil tuning problem or code problem.
Code check:
Login to the fob, write pages 4 to 7 separately.
If they all program correctly, not a code problem.
Coil check:
Send AUTH "11000" continuously and move fob for best reception of IDE response.
Hold fob in that position, try write all pages again.

sisaladin, 2023-05-09:
thank you for reply sorry I am in you in this system 
can you explain more please.
how i can write pages 4 to 7 separately
Send AUTH "11000" continuously

sisaladin, 2023-05-09:
any idea maybe how I can write in challenge mode or crypto mode

jenningsreeve, 2023-05-09:
Hi
Can you post the script from your write to tag.
It is difficult to help without knowing what you have done.

sisaladin, 2023-05-09:
this the tag i want to copy it [usb] pm3 --> lf hitag dump -k BDF5E846
[=] Authenticating in password mode
[+] Dumping tag memory...
[=] 00 | 68 45 96 12 | hE..
[=] 01 | BD F5 E8 46 | ...F
[=] 02 | 20 F0 4F 4E |  .ON
[=] 03 | 06 F9 07 C2 | ....
[=] 04 | 84 03 01 08 | ....
[=] 05 | 44 CE 20 C0 | D. .
[=] 06 | 0C 04 88 40 | ...@
[=] 07 | 0C 20 10 84 | . ..
[=] 08 | 00 00 00 00 | ....
[=] 09 | 00 00 00 00 | ....
[=] 10 | 00 00 00 00 | ....
[=] 11 | 00 00 00 00 | ....
[=] FILE PATH:  lf-hitag-68459612-dump.bin
[+] saved 48 bytes to binary file lf-hitag-68459612-dump.bin
[=] FILE PATH:  lf-hitag-68459612-dump.eml
[+] saved 12 blocks to text file lf-hitag-68459612-dump.eml
[=] FILE PATH:  lf-hitag-68459612-dump.json
[+] saved to json file lf-hitag-68459612-dump.json

jenningsreeve, 2023-05-09:
Not the lf hitag dump
You started this post saying you have written the data to the fob but it is incorrect.
Please post the result of the write to tag process.

sisaladin, 2023-05-09:
this is the tag im copy into trying
[usb] pm3 --> lf hitag writer --27 -k BDF5E846 -p 7 -d 0C201084
[#] Authenticating using password:
[#] bd f5 e8 46
[#] Configured for hitag2 writer
[usb] pm3 --> lf hitag reader --21 -k BDF5E846
[+]  UID: aa8b4813

[=] Hitag2 tag information

[=] ------------------------------------
[+] Config byte : 0x06 [ 00000110 ]
[+] Encoding    : Manchester
[+] Version     : Hitag2
[+] Coding in HITAG 2 operation: manchester
[+] Tag is in   : Password mode
[+] Page 6,7    : RW
[+] Page 4,5    : RW
[+] Page 3      : RW
[+] Page 1,2    : RW
[=] ------------------------------------
[=] 00 | AA 8B 48 13 | ..H.
[=] 01 | BD F5 E8 46 | ...F
[=] 02 | 20 F0 4F 4E |  .ON
[=] 03 | 06 F9 07 C2 | ....
[=] 04 | FF FF FF FF | ....
[=] 05 | FF FF FF FF | ....
[=] 06 | FF FF FF FF | ....
[=] 07 | FF FF FF FF | ....
[=] 08 | 00 00 00 00 | ....
[=] 09 | 00 00 00 00 | ....
[=] 10 | 00 00 00 00 | ....
[=] 11 | 00 00 00 00 | ....
[=] -------- Possible de-scramble patterns ---------
[+] Paxton id: 166666665 | 0x9ef21a9
[usb] pm3 -->

jenningsreeve, 2023-05-09:
Let me study the data.
From the data you posted, the fob has a green band, it is used on Compact/ Switch2.
The Paxton site code is 01287112
The fob pack number is 0
The fob is number 23 from the pack

jenningsreeve, 2023-05-09:
You seem to be testing different fobs.
The first script shows fob IDE as C0 15 57 13
The second script shows fob IDE as 68 45 96 12
The third script shows fob IDE as AA 8B 48 13
Are you testing different fobs or is there an error in reading.

sisaladin, 2023-05-09:
first one I cannot access anymore
[=] --- Tag Information ---------------------------
[=] -------------------------------------------------------------
[+]      UID: C0155713
[+]     TYPE: PCF 7936
[=] -------------------------------------------------------------
[usb] pm3 --> lf hitag reader --21 -k BDF5E846
[#] Password failed!
[usb] pm3 -->

jenningsreeve, 2023-05-09:
If the writer has changed the password by mistake you have bricked the fob.
I think you have used the Hitag S writer which is not the same as Hitag 2.
The PM3, as far as I'm aware, cannot write all pages to a Hitag 2.
Do you you only need 1 fob copied?.
If yes, where in the world are you?

sisaladin, 2023-05-09:
the second one 68 45 96 12  is the fob I want to copy it 
the third AA 8B 48 13 one is the fob im  trying to copy to

sisaladin, 2023-05-09:
yes Do you you only need 1 fob copied for now im in london

jenningsreeve, 2023-05-09:
Post a read of the fob you need copied, I will run off a copy and send it to you free.
Give me your address and postcode.

sisaladin, 2023-05-09:
thank you very much this is very nice for you I don't want to bother you
 I'm trying to learn and do it myself in case I needed in the future

jenningsreeve, 2023-05-09:
It would be no problem for me.
I have the ultimate Paxton cloning device, the Retag-UK PX1 Duplicator.
It takes less than 1 second for it to read and duplicate any Paxton or GDX token.

sisaladin, 2023-05-09:
where I can get one please ?

jenningsreeve, 2023-05-09:
Message the manufacturer directly info@retag.co.uk

sisaladin, 2023-05-10:
this tool is very expensive can I have your email please

sisaladin, 2023-05-10:
Please email me sampalvaroo@ .
g.
m.
ail.com

sisaladin, 2023-05-10:
This is the tag 
this the tag i want to copy it [usb] pm3 --> lf hitag dump -k BDF5E846
[=] Authenticating in password mode
[+] Dumping tag memory...
[=] 00 | 68 45 96 12 | hE..
[=] 01 | BD F5 E8 46 | ...F
[=] 02 | 20 F0 4F 4E | .ON
[=] 03 | 06 F9 07 C2 | ....
[=] 04 | 84 03 01 08 | ....
[=] 05 | 44 CE 20 C0 | D. .
[=] 06 | 0C 04 88 40 | ...@
[=] 07 | 0C 20 10 84 | . ..
[=] 08 | 00 00 00 00 | ....
[=] 09 | 00 00 00 00 | ....
[=] 10 | 00 00 00 00 | ....
[=] 11 | 00 00 00 00 | ....
[=] FILE PATH: lf-hitag-68459612-dump.bin
[+] saved 48 bytes to binary file lf-hitag-68459612-dump.bin
[=] FILE PATH: lf-hitag-68459612-dump.eml
[+] saved 12 blocks to text file lf-hitag-68459612-dump.eml
[=] FILE PATH: lf-hitag-68459612-dump.json
[+] saved to json file lf-hitag-68459612-dump.json

sisaladin, 2023-05-22:
Hi jareckib yes I did it very easy thank you very much

jenningsreeve, 2023-05-22:
Yes the Hitag2 v3.1 can copy a Paxton but you also need a PC, serial drivers, unzip the Russian software and hope it is clean, and follow the dodgy instructions on how to make it all work.
That's fine if you want to use your time working it all out.
The PX1 copies and verifies a Paxton token in under a second with no software, external power or instructions.
It is a serious tool for Locksmiths and system administrators not a hobbyist toy.
If you want to roll up your sleeves, I can tell you how to make your own Paxton copier for £20.

jareckib, 2023-05-22:
I have your PX1 device - page 3 is locked for writing after copying. I also have TMD5s and Hitag2 v 3.1. All my devices copy the PAXTON fob with no problem.

https://ibb.co/vBq7gcW
https://ibb.co/Bc7pgyQ

jenningsreeve, 2023-05-22:
Page 3 Config being changed to 46 only prevents the Page 3 Config from being changed again.
It does not apply Read Only to any other page.
It does that for a very good reason, to prevent a copy from being used to produce another copy.
Basic security, only an original issued Paxton can produce a copy.
1 complaint was made about this function in the very early version of the PX1.
All PX1 owners were offered a free software upgrade which made the "copy a copy" optional.
If you want that function upgraded, or even removed completely, free of charge, just sent it back to me.

PX1 now has many other functions that no other copier can do:
Produce a Compact/ Switch2 access token when a Shadow card is read.
Produce a Paxton Net2 token with unique random Net2 number, from any other Paxton or generic Hitag2 token.
It is the only copier that can tell the difference between Compact/ Switch2 fobs and the identical GDX Indigo fobs. 
It is the only copier that can duplicate GDX Indigo fobs.
It is the only copier that can duplicate a Paxton10 token.

I challenge you to make any of your other copiers do any of the above.

iceman1001, 2023-05-23:
No ads to other software thank you.

And pull latest,  the hitag2 writing just got some more love.

jenningsreeve, 2023-05-23:
To iceman1001
What does this mean? "And pull latest, the hitag2 writing just got some more love".

iceman1001, 2023-05-23:
that you should pull latest source in order to see if your hitag2 writes become better.

jenningsreeve, 2023-05-23:
To iceman1001
The latest source pull will not correct the inherent problems of PM3's ability to communicate with a Hitag2 transponder.
The LF coil is not suitable for Hitag2 as the physical transponders vary from an IMMO carbon coffin transponder to an ISO card.
Almost every Hitag2 programmer uses the NXP HTRC110/ PCF7991 front end transceiver which adapts its communication characteristics to the different transponders.
The PM3 front end is only discrete components and no amount of software will account for the hardware shortfall.

jareckib, 2023-05-23:
@iceman thanks a lot,.......works perfect

 `[=] Hitag2 tag information

[=] ------------------------------------
[+] Config byte : 0x46 [ 01000110 ]
[+] Encoding    : Manchester
[+] Version     : Hitag2
[+] Coding in HITAG 2 operation: manchester
[+] Tag is in   : Password mode
[+] Page 6,7    : RW
[+] Page 4,5    : RW
[+] Page 3      : read only. Configuration byte and password tag FIXED / IRREVERSIBLE
[+] Page 1,2    : RW
[=] ------------------------------------
[=] 00 | FA 32 39 12 | .29.
[=] 01 | BD F5 E8 46 | ...F
[=] 02 | 20 F0 4F 4E |  .ON
[=] 03 | 46 F9 07 C2 | F...
[=] 04 | 01 23 45 67 | .#Eg
[=] 05 | 01 23 45 67 | .#Eg
[=] 06 | 01 23 45 67 | .#Eg
[=] 07 | 01 23 45 67 | .#Eg
[=] 08 | 00 00 00 00 | ....
[=] 09 | 00 00 00 00 | ....
[=] 10 | 00 00 00 00 | ....
[=] 11 | 00 00 00 00 | ....
[=] -------- Possible de-scramble patterns ---------

jenningsreeve, 2023-05-23:
Yes I have seen the read script many times, that is not addressing the problem of writing new data to the Hitag2

iceman1001, 2023-05-23:
Smells like an sales pitch to me.  If the RFidler can handle hitag2, I am sure the Proxmark3 will do just fine.
One reason I haven't fixed it is because of the purpose.   Easy cloning of car fobs isn't research according to me. 

Anyway,  @sisaladin   @jareckib   this issue looks more like a user related issue  and for that we have the discord server and the pm3 forum for such discussions. 

The latest fix was related to writing, you tested reading.  

I will close this issue.

jenningsreeve, 2023-05-23:
To iceman1001
Yep that is the answer you give every time you are confronted with the Hitag2 problems, avoid the issue!

jenningsreeve, 2023-05-23:
To iceman1001
If you didn't have anything constructive to add to this issue, then you should not have intervened at all, butt out!

jenningsreeve, 2023-05-23:
To jareckib
Do you have a PM3 script to share of a write to a Hitag2?
I know you can do it with your other devices, this is a PM3 discussion

jareckib, 2023-05-23:
@iceman1001 .... That's right, but first I used the command -

  pm3 --> lf hitag writer --27 -k bdf5e846 -p 7 -d 01234567 ....so :)

jenningsreeve, 2023-05-23:
Ok, lets deal with what the PM3 is actually doing;
It has logged into the Hitag2 with normal auth 11000 then Hitag2 or Paxton password,
It has overwritten the Hitag2 password with the Paxton password without any verify after write,
If there was any error in the password that was written, the Hitag2 will return to wait state and the password is now unknown, BRICKED.  You have to verify the write was successful without logging out or you can loose the password. 
Any data page write should send the page to be written and then verify the mirrored page response confirming it will write to the correct page, the PM3 doesn't do that.
Each page written should be verified by a read immediately to confirm the data was actually saved by the transponder and that it is correct, the PM3 doesn't do that.
The position of the transponder in a read scenario is far less critical that a write scenario.  The transponder requires much more power from the PM3 coil (and time) to write to its internal memory, PM3 accounts for none of this.

jenningsreeve, 2023-05-23:
if the pages are written incorrectly then it won't open the door daaaaah

jenningsreeve, 2023-05-23:
What is the point of it all if you really don't care whether it works or not.

jareckib, 2023-05-23:
jennigsreeve@   
I meant the fob won't be locked for writing. You can correct the position of the fob on the coil and enter the correct values for pages 4 to 7. I checked the record 20 times .... fob centered on the coil, each time without writing errors. Have a nice day.