RfidResearchGroup/proxmark3 #2006: Question hf mf restore command key.bin file path

fraski30, 2023-06-11:
Hello everyone, sorry if is not a bug.
I'm using lastest version of proxspace on windows.
When i use commands like hf mf autopwn, or hf mf dump, files are saved into pm3 folder and all works perfect.
When i use hf mf restore command, i specify dump file(Which it´s created inside of pm3 folder) and key file also inside of pm3 folder. I have and error saying key file not found. If i move key file from pm3 folder to proxmark/client folder command works perfect without change any parameter.
Is this normal? Dump command works perfect with files inside pm3 folder. 
Thank you and sorry for my English!

iceman1001, 2023-06-11:
If you set the default paths by using the `pref` command,  it should be better
```
pref show
```

fraski30, 2023-06-11:
> If you set the default paths by using the `pref` command, it should be better
> 
> ```
> pref show
> ```

[=] Using C:\Users\davir\Desktop\ProxSpace\ProxSpace\pm3/.proxmark3/preferences.json
[=] Current settings
[=]     emoji.................. alttext
[=]     hints.................. on
[=]     color.................. ansi
[=]     Plot window............ X    2 Y   89 H  400 W  800
[=]     Slider/Overlay window.. X   10 Y  490 H  243 W  800
[=]     default save path...... C:\Users\davir\Desktop\ProxSpace\ProxSpace\pm3
[=]     dump save path......... C:\Users\davir\Desktop\ProxSpace\ProxSpace\pm3
[=]     trace save path........ C:\Users\davir\Desktop\ProxSpace\ProxSpace\pm3
[=]     client debug........... off
[=]     show plot sliders...... on
[=]     barmode................ value
[=]     Cmd execution delay.... 0
[=]     output................. normal
paths are default. Can i add path for restore?
Thanks iceman

iceman1001, 2023-06-12:
I pushed a fix for that for about two weeks ago,   are you using the latest proxspace/source code?

fraski30, 2023-06-12:
> I pushed a fix for that for about two weeks ago, are you using the latest proxspace/source code?

Oh my fault i updated from commit 351 to commit 449 and now is working perfect!
Thank you very much @iceman1001, problem solved

iceman1001, 2023-06-12:
Good to hear your problems has been resolved!

closed 2023-06-12 by iceman1001