RfidResearchGroup/proxmark3 #2007: PM3 Easy reads maybe 1 of 20 tries

xandmann, 2023-06-12:
Strange problem with PM3 Easy, recently purchased from KSEC, UK.
Out of the box the device reads maybe 1 of 20 tries no matter what tag
I try HF or LF (can’t even read the ones that came in the package).

Flashed latest Iceman fork from GitHub, with PM3GENERIC without issues
and according to “hw tune” the antennas are fine. I’m on Ubuntu 20.04.5 LTS.

Most of the time the device complains about multiple tags, collision after x bits.
There is nothing close by, nearest object 20 cm away. 

Tried removing the LF antenna and the middle board, placed the device with a long USB
cable in the middle of the workbench. Placed DUT at various distances 0-5 cm (on both sides of the PM3)
with the same (bad) results.

[image]

iceman1001, 2023-06-25:
Have you downloaded (pulled)  latest source,  compiled,  flashed and tested again?

iceman1001, 2023-07-27:
ping!

iceman1001, 2023-08-03:
closing because of inactivity

closed 2023-08-03 by iceman1001