RfidResearchGroup/proxmark3 #2013: Macports Build Process will not work on newer versions of MacOS, arm64 or x86_64.

ipreferpi, 2023-06-24:
**Describe the bug**
Macports build process does not work, due to a dependency issue for `arm-none-eabi-gcc`. Port health is showing failures for Ventura, Monterey, and Big Sur. https://ports.macports.org/port/arm-none-eabi-gcc/details/

**To Reproduce**
Steps to reproduce the behavior:
Attempt MacPorts installation process.
Get stuck on `Building arm-none-eabi-gcc`

**Expected behavior**
Code should build.

**Additional context**
Not really sure of a fix other than a documentation note to check for updates on the MacPorts health page for `arm-none-eabi-gcc`

iceman1001, 2023-06-24:
...  building arm-none-eabi-gcc  sounds like its outside of the scope of this repo.

Sounds like arm-none-eabl-gcc project has some issues?

linuxgemini, 2023-06-27:
(Full disclosure, I am the `proxmark3-iceman` port maintainer.)

_Out of scope for this repository._

The `arm-none-eabi-gcc` port (well, I guess any GCC cross compiler port) was broken-ish for some time and looks like recent changes to related meta Ports/PortGroups did fix this (I just built the Portfile yesterday). However, the MacPorts buildbot hasn't re-tested building `arm-none-eabi-gcc` since late April. Hence why `port` tries to compile it for you.

The build process for `arm-none-eabi-gcc` may take _half a day_ as GCC has to bootstrap itself from stage0.

Just so you know, please create a issue/ticket at https://trac.macports.org for further issues like this one.

closed 2023-06-27 by iceman1001