RfidResearchGroup/proxmark3 #2015: `pm3_help2json.py` doesn't work for commands with empty lines in description

wh201906, 2023-06-27:
**Describe the bug**
The `pm3_help2json.py` cannot parse some of the command helps into json
output from client:
```
---------------------------------------------------------------------------------------
usart btfactory          
available offline: no

Reset BT add-on to factory settings
This requires
    1) BTpower to be turned ON
    2) BT add-on to NOT be connected
      => the add-on blue LED must blink

WARNING: process only if strictly needed!

usage:
    usart btfactory [-h]

options:
    -h, --help                     This help

examples/notes:
    usart btfactory               

```
parse result:
```
        "usart btfactory": {
            "command": "usart btfactory",
            "description": "Reset BT add-on to factory settings This requires 1) BTpower to be turned ON 2) BT add-on to NOT be connected => the add-on blue LED must blink",
            "notes": [],
            "offline": false,
            "options": [],
            "usage": ""
        },
``` 

**Expected behavior**
```
        "usart btfactory": {
            "command": "usart btfactory",
            "description": "Reset BT add-on to factory settings This requires 1) BTpower to be turned ON 2) BT add-on to NOT be connected => the add-on blue LED must blink WARNING: process only if strictly needed!",
            "notes": [
                "usart btfactory"
            ],
            "offline": false,
            "options": [
                "-h, --help This help"
            ],
            "usage": "usart btfactory [-h]"
        },
```

**Additional context**
I think it's caused by the regular expression in the `pm3_help2json.py`.
```
re_description = r'(?P<description>(?:.+\n)+)\n+'
```
It doesn't match the empty lines.

iceman1001, 2023-06-27:
After playing with regex online...   The problem seem to be description regex.    When the description has empty lines in it,  the usage part fails..

[image]

wh201906, 2023-06-27:
Yes. I think so.

iceman1001, 2023-06-27:
complex regex..    there lots of mismatches.

wh201906, 2023-06-27:
I tried to match every line with Positive Lookahead like this
```
(?P<description>(?:.*\n)+)\n+(?=usage)
```
It works on regex101.com, but not in the `pm3_help2json.py`

<details><summary>regex101.com</summary>

[image]

</details>

iceman1001, 2023-06-27:
:)  funny,   I just pushed a fix based on the same idea.

This one remove all empty sub command help items, like  "lf help".   It also fixes a bunch of misses before.

If you don't mind to pull and test?

iceman1001, 2023-06-27:
For the regex101.com to test properly you will need to give it the whole help text file.   Its lot to look through.

closed 2023-06-27 by wh201906