RfidResearchGroup/proxmark3 #2018: ELECTRA fob cloning

jareckib, 2023-06-30:
[image]

I don't know if it's in the topic and I can't check because I don't have this type of key. But on one of the forums I found something like this:  https://github.com/ClaraCrazy/Flipper-Xtreme/issues/199

iceman1001, 2023-07-05:
do you have a copy of the signal and which data that is printed?
```
lf read
data save -f lf_elektra
```

jareckib, 2023-07-05:
``-102
-93
-88
-82
-76
-70
-66
-60
-55
-52
-48
-44
-41
-39
-37
-35
-32
-30
-28
-27
-24
-23
-21
-21
-19
-18
-18
-17
-16
-15
-14
-13
-13
-13
-14
-12
-11
-11
-10
-11
-11
-11
-10
38
127
127
127
127
127
127
127
127
127
127
127
127
127
127
127
127
124
113
104
94
85
77
73
66
61
55
[... 41249 lines cut ...]
-3
-2
-3
-3
-79
-112
-112
-112
-112
-112
-112
-112
-112
-112
-112
-112
-112
-112
-112
-112
-112
-112
-112
-112
-109

jareckib, 2023-07-05:
[image]

jareckib, 2023-07-07:
[lf_elektra.zip](https://github.com/RfidResearchGroup/proxmark3/files/11985790/lf_elektra.zip)

jareckib, 2023-07-07:
[lf_elektra.zip](https://github.com/RfidResearchGroup/proxmark3/files/11985833/lf_elektra.zip)

jareckib, 2023-07-12:
Any chance of a new ELECTRA fob?

closed 2023-07-15 by jareckib