RfidResearchGroup/proxmark3 #2019: mifare dictionary keys

evilpete, 2023-07-01:
# Travelodge by Wyndham Berkeley
0000FFFFFFFF
4663ACD2FFFF
EDC317193709

# Hotel Santa Cruz
75FAB77E2E5B

# saflok brand HOTEL key
32F093536677

# A WaterFront Hotel in  Oakland
3351916B5A77

# Ballys (2018)
336E34CC2177

# Random Hawaiian Hotel
A1670589B2AF

# SF Hotel (SoMa area)
2E0F00700000

# TAP LA transit card (mf classic card from 2014)
073EA622C9CC
0B1E863497F0
0BB9EE9022D9
0E3824E8C3C6
0F2791EDA281
27A6FA15ED2C
35C3F7DE783A
375A22C53D43
3D51DE4AC40F
458D5B2A9EEC
554B2EFDD9C4
5621022CD098
5C063BD5579F
62545F654D9F
6324521C7AFC
689766D777D0
68F933727D62
71E014238723
879DC03B759A
9095D0C3C344
92A7AD43C437
962437C8C45A
A160D18CA5C8
A39A65A72D6E
A4EBB44208C2
AB5F0F829695
AF0601A02DC7
BEF14EC1C1C9
C91DDF322410
CCEA733B0FD3
E372C0DFA525
EA0389A32D72

iceman1001, 2023-07-05:
Nice list of keys,     first of all its very easy to make the changes to the dictionary and then make PR.

Second,   are these keys verified to be static in those systems and not just one card?

iceman1001, 2023-07-20:
ping

closed 2023-09-06 by iceman1001