RfidResearchGroup/proxmark3 #2027: Buffer overflow crash when writing to files on pentoo

dperret, 2023-07-09:
**Describe the bug**
A clear and concise description of what the bug is.

Writing to files is causing the proxmark3 client to crash on pentoo, with the error message ```*** buffer overflow detected ***: terminated
Aborted```

**To Reproduce**
Steps to reproduce the behavior:

Each of the 3 commands below caused the client to crash with the same error.
```hf mfu dump -k <key>```
```hf mf autopwn```
```trace save -f <filename>```

**Expected behavior**
A clear and concise description of what you expected to happen.

The rest of the commands seem to be working, just not writing the data to a file.

**Desktop (please complete the following information):**
 - OS: pentoo, Linux kernel 6.3.3
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
 - hw status
 - data tune

```
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  Iceman/master/v4.16717 2023-06-26 13:10:45 de506fd18
  compiled with............. GCC 12.3.1 20230526
  platform.................. Linux / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... present
  Python script support..... absent
  Lua SWIG support.......... present
  Python SWIG support....... absent

 [ PROXMARK3 ]
  device.................... RDV4
  firmware.................. RDV4
  external flash............ present
  smartcard reader.......... present
  FPC USART for BT add-on... present

 [ ARM ]
  bootrom: Iceman/master/v4.16717 2023-06-26 13:10:45 de506fd18
       os: Iceman/master/v4.16717 2023-06-26 13:10:45 de506fd18
  compiled with GCC 10.3.1 20210824 (release)

 [ FPGA ] 
  LF image 2s30vq100 2023-05-24 14:12:56
  HF image 2s30vq100 2023-05-26 19:50:53
  HF FeliCa image 2s30vq100 2023-05-24 14:12:58
  HF 15 image 2s30vq100 2023-05-24 14:12:57

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 67% used )

[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 38708
[#]   Available memory........ 38708
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode.................... HF image 2s30vq100 2023-05-26 19:50:53
[#] Flash memory
[#]   Baudrate................ 24 MHz
[#]   Init.................... OK
[#]   Memory size............. 2 mbits / 256 kb
[#]   Unique ID (be).......... 0x26A6A5A782A867D5
[#] Smart card module (ISO 7816)
[#]   version................. v2.06
[#]     Outdated firmware. Please upgrade to v4.x or above.
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] 
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A | 
[#]     long leading reference |  29 |  17 |  15 |  47 |  15 | N/A | N/A | 
[#]               leading zero |  29 |  17 |  15 |  40 |  15 | N/A | N/A | 
[#]    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 | 
[#] 
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 284160
[#]   Transfer Speed PM3 -> Client... 568320 bytes/s
[#] Various
[#]   Max stack usage......... 4088 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... -1
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 29425 Hz
[#] Installed StandAlone Mode
[#]   No standalone mode present
[#] Flash memory dictionary loaded
[#] 
[usb] pm3 --> data tune
[=] ---------- Reminder ------------------------
[=] `hw tune` doesn't actively tune your antennas,
[=] it's only informative.
[=] Measuring antenna characteristics, please wait...
 🕛   9
[=] ---------- LF Antenna ----------
[+] LF antenna: 67.04 V - 125.00 kHz
[+] LF antenna: 33.00 V - 134.83 kHz
[+] LF optimal: 67.04 V - 125.00 kHz
[+] Approx. Q factor (*): 11.7 by frequency bandwidth measurement
[+] Approx. Q factor (*): 11.7 by peak voltage measurement
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[+] HF antenna: 41.72 V - 13.56 MHz
[+] Approx. Q factor (*): 7.3 by peak voltage measurement
[+] HF antenna is OK

(*) Q factor must be measured without tag on the antenna

[+] Displaying LF tuning graph. Divisor 88 (blue) is 134.83 kHz, 95 (red) is 125.00 kHz.

[usb] pm3 --> 

```

**Additional context**
Add any other context about the problem here.

When running the same commands on Windows using the prebuilt binaries from https://www.proxmarkbuilds.org/ the client software did not crash, and successfully wrote to files.

iceman1001, 2023-07-09:
I am confused of your actions. 
you write a MFU card then run autopwn against the same card?

dperret, 2023-07-09:
Sorry, no, those were run against different cards. I ran hf 14a sniff against a MFU card, tried to write the trace and got the crash. I tried to dump that MFU card with the sniffed password and got another crash. I ran autopwn against a completely different mifare classic 1K card.

iceman1001, 2023-07-09:
Try write down the exact commands which is needed to replicate the bug.

dperret, 2023-07-09:
The commands to reproduce the bug were 
With a MFU card and reader
```
hf 14a sniff -c -r
trace save -f mfutracefile
```
With the MFU card only
```
hf mfu dump -k 9FD5E9A9
```
and with a different MF Classic 1K card only
```
hf mf autopwn
```

Here is the terminal output reproducing the bug, showing both the commands and the output from the proxmark3 client. 
```
pentoo ~ # proxmark3 /dev/ttyACM0 
[=] Session log /root/.proxmark3/logs/log_20230709.txt
[+] loaded from JSON file /root/.proxmark3/preferences.json
[=] Using UART port /dev/ttyACM0
[=] Communicating with PM3 over USB-CDC

  8888888b.  888b     d888  .d8888b.   
  888   Y88b 8888b   d8888 d88P  Y88b  
  888    888 88888b.d88888      .d88P  
  888   d88P 888Y88888P888     8888"  
  8888888P"  888 Y888P 888      "Y8b.  
  888        888  Y8P  888 888    888  
  888        888   "   888 Y88b  d88P 
  888        888       888  "Y8888P"    [ ☕ ]

Release v4.16717 - seven
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
  [ Proxmark3 RFID instrument ]

    MCU....... AT91SAM7S512 Rev A
    Memory.... 512 KB ( 67% used )

    Client.... Iceman/master/v4.16717 2023-06-26 13:10:45
    Bootrom... Iceman/master/v4.16717 2023-06-26 13:10:45 
    OS........ Iceman/master/v4.16717 2023-06-26 13:10:45 
    Target.... RDV4

[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  Iceman/master/v4.16717 2023-06-26 13:10:45 de506fd18
  compiled with............. GCC 12.3.1 20230526
  platform.................. Linux / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... present
  Python script support..... absent
  Lua SWIG support.......... present
  Python SWIG support....... absent

 [ PROXMARK3 ]
  device.................... RDV4
  firmware.................. RDV4
  external flash............ present
  smartcard reader.......... present
  FPC USART for BT add-on... present

 [ ARM ]
  bootrom: Iceman/master/v4.16717 2023-06-26 13:10:45 de506fd18
       os: Iceman/master/v4.16717 2023-06-26 13:10:45 de506fd18
  compiled with GCC 10.3.1 20210824 (release)
[... 628 lines cut ...]
[+] -----+-----+--------------+---+--------------+----
[+]  000 | 003 | 00AE371F0EE8 | N | FFFFFFFFFFFF | D
[+]  001 | 007 | 2A2C13CC242A | D | FFFFFFFFFFFF | D
[+]  002 | 011 | FFFFFFFFFFFF | D | FFFFFFFFFFFF | D
[+]  003 | 015 | FFFFFFFFFFFF | D | FFFFFFFFFFFF | D
[+]  004 | 019 | 00AE371F0EE8 | R | FFFFFFFFFFFF | D
[+]  005 | 023 | 00AE371F0EE8 | R | FFFFFFFFFFFF | D
[+]  006 | 027 | 00AE371F0EE8 | R | FFFFFFFFFFFF | D
[+]  007 | 031 | 00AE371F0EE8 | R | FFFFFFFFFFFF | D
[+]  008 | 035 | 00AE371F0EE8 | R | FFFFFFFFFFFF | D
[+]  009 | 039 | 00AE371F0EE8 | R | FFFFFFFFFFFF | D
[+]  010 | 043 | 00AE371F0EE8 | R | FFFFFFFFFFFF | D
[+]  011 | 047 | 00AE371F0EE8 | R | FFFFFFFFFFFF | D
[+]  012 | 051 | 00AE371F0EE8 | R | FFFFFFFFFFFF | D
[+]  013 | 055 | 00AE371F0EE8 | R | FFFFFFFFFFFF | D
[+]  014 | 059 | 00AE371F0EE8 | R | FFFFFFFFFFFF | D
[+]  015 | 063 | 00AE371F0EE8 | R | FFFFFFFFFFFF | D
[+] -----+-----+--------------+---+--------------+----
[=] ( D:Dictionary / S:darkSide / U:User / R:Reused / N:Nested / H:Hardnested / C:statiCnested / A:keyA  )

*** buffer overflow detected ***: terminated
Aborted
pentoo ~ # 

```

iceman1001, 2023-07-09:
so after bad readings / coupling with your toothbrush,  the dump command wanted to use the UID,   and then crashed.
```
[=] Using UID as filename
*** buffer overflow detected ***: terminated
Aborted
```

and the second time it seems to be almost the same,  with  autopwn wanting to write files....

dperret, 2023-07-09:
Yes, that is correct. I also tried unplugging and replugging the RDV4 to reset it, and running autopwn again against another MF Classic 1K card, and got the same result. I reproduced this behavior on a second machine running pentoo as well.

iceman1001, 2023-07-09:
can you run `gdb` ?

dperret, 2023-07-09:
Yes I can. Running proxmark3 in gdb and reproducing the autopwn crash ended with this output:
```
*** buffer overflow detected ***: terminated

Thread 5 "WorkerThread" received signal SIGABRT, Aborted.
[Switching to Thread 0x7ffff23046c0 (LWP 28564)]
0x00007ffff66afa4c in ?? () from /lib64/libc.so.6
(gdb) 

```

dperret, 2023-07-09:
and the backtrace:
```
(gdb) backtrace
#0  0x00007ffff66afa4c in ?? () from /lib64/libc.so.6
#1  0x00007ffff6660d96 in raise () from /lib64/libc.so.6
#2  0x00007ffff664a87c in abort () from /lib64/libc.so.6
#3  0x00007ffff664b5b7 in ?? () from /lib64/libc.so.6
#4  0x00007ffff673ef3b in __fortify_fail () from /lib64/libc.so.6
#5  0x00007ffff673d766 in __chk_fail () from /lib64/libc.so.6
#6  0x00007ffff673d345 in __snprintf_chk () from /lib64/libc.so.6
#7  0x00005555556c7ebe in snprintf (__fmt=0x55555582c675 "%.*s%s", __n=45, __s=0x7fffe803c546 "") at /usr/include/bits/stdio2.h:54
#8  newfilenamemcopyEx (preferredName=preferredName@entry=0x7fffe803aed0 "hf-mf-5683802B-key.bin", suffix=suffix@entry=0x55555581b07a ".bin", e_save_path=e_save_path@entry=spDump) at src/fileutils.c:221
#9  0x00005555556c973c in createMfcKeyDump (preferredName=preferredName@entry=0x7fffe803aed0 "hf-mf-5683802B-key.bin", sectorsCnt=sectorsCnt@entry=16 '\020', e_sector=e_sector@entry=0x7fffe803ad40) at src/fileutils.c:777
#10 0x000055555563a098 in CmdHF14AMfAutoPWN (Cmd=<optimized out>) at src/cmdhfmf.c:3090
#11 0x00005555556a2f3c in CmdsParse (Commands=0x5555558d23a0 <CommandTable>, Cmd=0x7fffe801c4b6 "autopwn") at src/cmdparser.c:321
#12 0x00005555556a2f3c in CmdsParse (Commands=0x5555558ceb40 <CommandTable>, Cmd=0x7fffe801c4b3 "mf autopwn") at src/cmdparser.c:321
#13 0x00005555556a2f3c in CmdsParse (Commands=Commands@entry=0x5555558d6b40 <CommandTable>, Cmd=Cmd@entry=0x7fffe801c4b0 "hf mf autopwn") at src/cmdparser.c:321
#14 0x00005555556a1e02 in CommandReceived (Cmd=Cmd@entry=0x7fffe801c4b0 "hf mf autopwn") at src/cmdmain.c:365
#15 0x00005555556ed43c in main_loop (script_cmds_file=<optimized out>, script_cmd=<optimized out>, stayInCommandLoop=true) at src/proxmark3.c:404
#16 0x00007ffff6ca567f in ?? () from /usr/lib64/libQt5Core.so.5
#17 0x00007ffff66add14 in ?? () from /lib64/libc.so.6
#18 0x00007ffff672f010 in ?? () from /lib64/libc.so.6
(gdb) 

```

dperret, 2023-07-09:
Here's the crash and backtrace from the mfu dump command

```
[usb] pm3 --> hf mfu dump -k 9FD5E9A9
[+] TYPE: NTAG 213C 144bytes (NT2H1311C1DTL)  
[+] Reading tag memory...
[#] Cmd Error: card timeout. len: 0
[#] Cmd Error: card timeout. len: 0
[#] Cmd Error: card timeout. len: 0
[#] Cmd Error: card timeout. len: 0
[#] Cmd Error: card timeout. len: 0
[#] Cmd Error: card timeout. len: 0
[!] ⚠️  iso14443a card select failed
[#] Warning: HF field is off, ignoring TransmitFor14443a command
[#] Warning: HF field is off, ignoring TransmitFor14443a command

[=] MFU dump file information
[=] -------------------------------------------------------------
[=] Version..... 00 00 00 00 00 00 00 00 
[=] TBD 0....... 00 00 
[=] TBD 1....... 00 
[=] Signature... 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[=]              00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[=] Counter 0... 00 00 00 
[=] Tearing 0... 00 
[=] Counter 1... 00 00 00 
[=] Tearing 1... 00 
[=] Counter 2... 00 00 00 
[=] Tearing 2... BD 
[=] Max data page... 43 ( 176 bytes )
[=] Header size..... 56 bytes

[=] -------------------------------------------------------------
[=] block#   | data        |lck| ascii
[=] ---------+-------------+---+------
[=]   0/0x00 | 04 4C 8F 4F |   | .L.O
[=]   1/0x01 | BA A0 11 90 |   | ....
[=]   2/0x02 | 9B 48 FF FF |   | .H..
[=]   3/0x03 | E1 10 12 00 | 1 | ....
[=]   4/0x04 | 03 20 D1 01 | 1 | . ..
[=]   5/0x05 | 1C 55 02 70 | 1 | .U.p
[=]   6/0x06 | 68 69 6C 69 | 1 | hili
[=]   7/0x07 | 70 73 2E 63 | 1 | ps.c
[=]   8/0x08 | 6F 6D 2F 6E | 1 | om/n
[=]   9/0x09 | 66 63 62 72 | 1 | fcbr
[=]  10/0x0A | 75 73 68 68 | 1 | ushh
[=]  11/0x0B | 65 61 64 74 | 1 | eadt
[=]  12/0x0C | 61 70 FE 00 | 1 | ap..
[=]  13/0x0D | 00 00 00 00 | 1 | ....
[=]  14/0x0E | 00 00 00 00 | 1 | ....
[=]  15/0x0F | 00 00 00 00 | 1 | ....
[=]  16/0x10 | 00 00 00 00 | 0 | ....
[=]  17/0x11 | 00 00 00 00 | 0 | ....
[=]  18/0x12 | 00 00 00 00 | 0 | ....
[=]  19/0x13 | 00 00 00 00 | 0 | ....
[=]  20/0x14 | 00 00 00 00 | 0 | ....
[=]  21/0x15 | 00 00 00 00 | 0 | ....
[=]  22/0x16 | 00 00 00 00 | 0 | ....
[=]  23/0x17 | 00 00 00 00 | 0 | ....
[=]  24/0x18 | 00 00 00 00 | 0 | ....
[=]  25/0x19 | 00 00 00 00 | 0 | ....
[=]  26/0x1A | 00 00 00 00 | 0 | ....
[=]  27/0x1B | 00 00 00 00 | 0 | ....
[=]  28/0x1C | 00 00 00 00 | 0 | ....
[=]  29/0x1D | 00 00 00 00 | 0 | ....
[=]  30/0x1E | 00 00 00 00 | 0 | ....
[=]  31/0x1F | 00 01 09 00 | 0 | ....
[=]  32/0x20 | 00 00 01 02 | 0 | ....
[=]  33/0x21 | 60 54 32 32 | 0 | `T22
[=]  34/0x22 | 30 37 30 39 | 0 | 0709
[=]  35/0x23 | 20 34 33 54 | 0 |  43T
[=]  36/0x24 | E5 61 02 01 | 0 | .a..
[=]  37/0x25 | 00 00 04 00 | 0 | ....
[=]  38/0x26 | AD 2F 7D 04 | 0 | ./}.
[=]  39/0x27 | 00 00 00 01 | 0 | ....
[=]  40/0x28 | 00 03 30 BD | 0 | ..0.
[=]  41/0x29 | 04 00 00 10 | 0 | ....
[=]  42/0x2A | 43 00 00 00 | 0 | C...
[=]  43/0x2B | 9F D5 E9 A9 | 0 | ....
[=]  44/0x2C | B2 52 00 00 | 0 | .R..
[=] ---------------------------------
[=] Using UID as filename
*** buffer overflow detected ***: terminated

Thread 5 "WorkerThread" received signal SIGABRT, Aborted.
[Switching to Thread 0x7ffff23046c0 (LWP 29535)]
0x00007ffff66afa4c in ?? () from /lib64/libc.so.6
(gdb) backtrace
#0  0x00007ffff66afa4c in ?? () from /lib64/libc.so.6
#1  0x00007ffff6660d96 in raise () from /lib64/libc.so.6
#2  0x00007ffff664a87c in abort () from /lib64/libc.so.6
#3  0x00007ffff664b5b7 in ?? () from /lib64/libc.so.6
#4  0x00007ffff673ef3b in __fortify_fail () from /lib64/libc.so.6
#5  0x00007ffff673d766 in __chk_fail () from /lib64/libc.so.6
#6  0x00007ffff673d345 in __snprintf_chk () from /lib64/libc.so.6
#7  0x00005555556c7ebe in snprintf (__fmt=0x55555582c675 "%.*s%s", __n=53, __s=0x7fffe803d226 "") at /usr/include/bits/stdio2.h:54
#8  newfilenamemcopyEx (preferredName=preferredName@entry=0x7ffff2303010 "hf-mfu-044C8FBAA01190-dump", suffix=suffix@entry=0x55555581b07a ".bin", e_save_path=e_save_path@entry=spDefault) at src/fileutils.c:221
#9  0x00005555556c7f1b in newfilenamemcopy (preferredName=preferredName@entry=0x7ffff2303010 "hf-mfu-044C8FBAA01190-dump", suffix=suffix@entry=0x55555581b07a ".bin") at src/fileutils.c:185
#10 0x00005555556c7f49 in saveFile (preferredName=preferredName@entry=0x7ffff2303010 "hf-mfu-044C8FBAA01190-dump", suffix=suffix@entry=0x55555581b07a ".bin", data=data@entry=0x7ffff2302b97, datalen=datalen@entry=236) at src/fileutils.c:240
#11 0x00005555556cc230 in pm3_save_dump (fn=fn@entry=0x7ffff2303010 "hf-mfu-044C8FBAA01190-dump", d=d@entry=0x7ffff2302b97 "", n=236, jsft=jsft@entry=jsfMfuMemory, blocksize=blocksize@entry=4) at src/fileutils.c:2125
#12 0x000055555565d2ca in CmdHF14AMfUDump (Cmd=<optimized out>) at src/cmdhfmfu.c:2680
#13 0x00005555556a2f3c in CmdsParse (Commands=0x5555558d3360 <CommandTable>, Cmd=0x7fffe803ae27 "dump -k 9FD5E9A9") at src/cmdparser.c:321
#14 0x00005555556a2f3c in CmdsParse (Commands=0x5555558ceb40 <CommandTable>, Cmd=0x7fffe803ae23 "mfu dump -k 9FD5E9A9") at src/cmdparser.c:321
#15 0x00005555556a2f3c in CmdsParse (Commands=Commands@entry=0x5555558d6b40 <CommandTable>, Cmd=Cmd@entry=0x7fffe803ae20 "hf mfu dump -k 9FD5E9A9") at src/cmdparser.c:321
#16 0x00005555556a1e02 in CommandReceived (Cmd=Cmd@entry=0x7fffe803ae20 "hf mfu dump -k 9FD5E9A9") at src/cmdmain.c:365
#17 0x00005555556ed43c in main_loop (script_cmds_file=<optimized out>, script_cmd=<optimized out>, stayInCommandLoop=true) at src/proxmark3.c:404
#18 0x00007ffff6ca567f in ?? () from /usr/lib64/libQt5Core.so.5
#19 0x00007ffff66add14 in ?? () from /lib64/libc.so.6
#20 0x00007ffff672f010 in ?? () from /lib64/libc.so.6
(gdb) 

```

dperret, 2023-07-09:
One more with the backtrace from trying to save the trace file
```
[usb] pm3 --> trace save -f mfutracefile
*** buffer overflow detected ***: terminated

Thread 5 "WorkerThread" received signal SIGABRT, Aborted.
[Switching to Thread 0x7ffff23046c0 (LWP 29606)]
0x00007ffff66afa4c in ?? () from /lib64/libc.so.6
(gdb) backtrace
#0  0x00007ffff66afa4c in ?? () from /lib64/libc.so.6
#1  0x00007ffff6660d96 in raise () from /lib64/libc.so.6
#2  0x00007ffff664a87c in abort () from /lib64/libc.so.6
#3  0x00007ffff664b5b7 in ?? () from /lib64/libc.so.6
#4  0x00007ffff673ef3b in __fortify_fail () from /lib64/libc.so.6
#5  0x00007ffff673d766 in __chk_fail () from /lib64/libc.so.6
#6  0x00007ffff673d345 in __snprintf_chk () from /lib64/libc.so.6
#7  0x00005555556c7ebe in snprintf (__fmt=0x55555582c675 "%.*s%s", __n=41, __s=0x7fffe803ca76 "") at /usr/include/bits/stdio2.h:54
#8  newfilenamemcopyEx (preferredName=preferredName@entry=0x7ffff2303510 "mfutracefile", suffix=suffix@entry=0x55555581be30 ".trace", e_save_path=e_save_path@entry=spDefault) at src/fileutils.c:221
#9  0x00005555556c7f1b in newfilenamemcopy (preferredName=preferredName@entry=0x7ffff2303510 "mfutracefile", suffix=suffix@entry=0x55555581be30 ".trace") at src/fileutils.c:185
#10 0x00005555556c7f49 in saveFile (preferredName=preferredName@entry=0x7ffff2303510 "mfutracefile", suffix=suffix@entry=0x55555581be30 ".trace", data=0x7fffe803c400, datalen=1637) at src/fileutils.c:240
#11 0x00005555556a7f46 in CmdTraceSave (Cmd=<optimized out>) at src/cmdtrace.c:1129
#12 0x00005555556a2f3c in CmdsParse (Commands=0x5555558d7880 <CommandTable>, Cmd=0x7fffe803bd86 "save -f mfutracefile") at src/cmdparser.c:321
#13 0x00005555556a2f3c in CmdsParse (Commands=Commands@entry=0x5555558d6b40 <CommandTable>, Cmd=Cmd@entry=0x7fffe803bd80 "trace save -f mfutracefile") at src/cmdparser.c:321
#14 0x00005555556a1e02 in CommandReceived (Cmd=Cmd@entry=0x7fffe803bd80 "trace save -f mfutracefile") at src/cmdmain.c:365
#15 0x00005555556ed43c in main_loop (script_cmds_file=<optimized out>, script_cmd=<optimized out>, stayInCommandLoop=true) at src/proxmark3.c:404
#16 0x00007ffff6ca567f in ?? () from /usr/lib64/libQt5Core.so.5
#17 0x00007ffff66add14 in ?? () from /lib64/libc.so.6
#18 0x00007ffff672f010 in ?? () from /lib64/libc.so.6
(gdb) 

```

iceman1001, 2023-07-09:
Somewhere around there...
```
 newfilenamemcopyEx (preferredName=preferredName@entry=0x7ffff2303010 "hf-mfu-044C8FBAA01190-dump", suffix=suffix@entry=0x55555581b07a ".bin", e_save_path=e_save_path@entry=spDefault) at src/fileutils.c:221
```

iceman1001, 2023-07-09:
I pushed a fix,  if you pull latest,  compile and test.   Lets see if it still complains

dperret, 2023-07-10:
Still crashing at least during autopwn
```
(gdb) run /dev/ttyACM0
Starting program: /home/redbaron/source/proxmark3/client/proxmark3 /dev/ttyACM0
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
[=] Session log /root/.proxmark3/logs/log_20230710.txt
[+] loaded from JSON file /root/.proxmark3/preferences.json
[=] Using UART port /dev/ttyACM0
[New Thread 0x7ffff40d76c0 (LWP 11595)]
[=] Communicating with PM3 over USB-CDC

  8888888b.  888b     d888  .d8888b.   
  888   Y88b 8888b   d8888 d88P  Y88b  
  888    888 88888b.d88888      .d88P  
  888   d88P 888Y88888P888     8888"  
  8888888P"  888 Y888P 888      "Y8b.  
  888        888  Y8P  888 888    888  
  888        888   "   888 Y88b  d88P 
  888        888       888  "Y8888P"    [ ☕ ]

[New Thread 0x7ffff33266c0 (LWP 11596)]
[Detaching after fork from child process 11597]
[New Thread 0x7ffff2b256c0 (LWP 11598)]
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[New Thread 0x7ffff127c6c0 (LWP 11599)]
  [ Proxmark3 RFID instrument ]

    MCU....... AT91SAM7S512 Rev A
    Memory.... 512 KB ( 67% used )

    Client.... Iceman/master/v4.16717-63-g17be2f6c3 2023-07-09 19:56:01
    Bootrom... Iceman/master/v4.16717-63-g17be2f6c3 2023-07-09 19:55:56 
    OS........ Iceman/master/v4.16717-63-g17be2f6c3 2023-07-09 19:56:05 
    Target.... RDV4

[usb] pm3 --> hf mf autopwn
[!] ⚠️  no known key was supplied, key recovery might fail
[+] loaded 56 keys from hardcoded default array
[=] running strategy 1
[=] .
[=] Chunk 2.7s | found 31/32 keys (56)
[=] running strategy 2
[=] .
[=] Chunk 2.6s | found 28/32 keys (56)
[+] target sector   0 key type A -- found valid key [ FFFFFFFFFFFF ] (used for nested / hardnested attack)
[+] target sector   0 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   1 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   1 key type B -- found valid key [ 2A2C13CC242A ]
[+] target sector   2 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   2 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   3 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   3 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   4 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   4 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   5 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   5 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   6 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   6 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   7 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   7 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   8 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   8 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   9 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   9 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector  10 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector  10 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector  11 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector  11 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector  12 key type A -- found valid key [ FFFFFFFFFFFF ]
[... 35 lines cut ...]
[Switching to Thread 0x7ffff127c6c0 (LWP 11599)]
0x00007ffff66afa4c in ?? () from /lib64/libc.so.6
(gdb) backtrace
#0  0x00007ffff66afa4c in  () at /lib64/libc.so.6
#1  0x00007ffff6660d96 in raise () at /lib64/libc.so.6
#2  0x00007ffff664a87c in abort () at /lib64/libc.so.6
#3  0x00007ffff664b5b7 in  () at /lib64/libc.so.6
#4  0x00007ffff673ef3b in  () at /lib64/libc.so.6
#5  0x00007ffff673d766 in  () at /lib64/libc.so.6
#6  0x00007ffff673d345 in __snprintf_chk () at /lib64/libc.so.6
#7  0x00005555557250b1 in snprintf (__fmt=0x5555558facdb "%.*s%s", __n=50, __s=0x7fffe803c456 "") at /usr/include/bits/stdio2.h:54
#8  newfilenamemcopyEx (preferredName=preferredName@entry=0x7fffe803af40 "hf-mf-3631862B-key.bin", suffix=suffix@entry=0x5555558e8fee ".bin", e_save_path=e_save_path@entry=spDump) at src/fileutils.c:221
#9  0x0000555555727141 in createMfcKeyDump (preferredName=preferredName@entry=0x7fffe803af40 "hf-mf-3631862B-key.bin", sectorsCnt=sectorsCnt@entry=16 '\020', e_sector=e_sector@entry=0x7fffe803adb0) at src/fileutils.c:777
#10 0x0000555555658567 in CmdHF14AMfAutoPWN (Cmd=<optimized out>) at src/cmdhfmf.c:3097
#11 0x00005555556ef25e in CmdsParse (Commands=0x5555559a82c0 <CommandTable>, Cmd=0x7fffe801c526 "autopwn") at src/cmdparser.c:321
#12 0x00005555556ef25e in CmdsParse (Commands=0x5555559a4b00 <CommandTable>, Cmd=0x7fffe801c523 "mf autopwn") at src/cmdparser.c:321
#13 0x00005555556ef25e in CmdsParse (Commands=Commands@entry=0x5555559ac920 <CommandTable>, Cmd=Cmd@entry=0x7fffe801c520 "hf mf autopwn") at src/cmdparser.c:321
#14 0x00005555556edd23 in CommandReceived (Cmd=Cmd@entry=0x7fffe801c520 "hf mf autopwn") at src/cmdmain.c:365
#15 0x000055555575dba5 in main_loop (script_cmds_file=<optimized out>, script_cmd=<optimized out>, stayInCommandLoop=true) at src/proxmark3.c:404
#16 0x00007ffff6ca567f in  () at /usr/lib64/libQt5Core.so.5
#17 0x00007ffff66add14 in  () at /lib64/libc.so.6
#18 0x00007ffff672f010 in  () at /lib64/libc.so.6
(gdb) 

```

dperret, 2023-07-10:
And the mfu dump
```
[usb] pm3 --> hf mfu dump -k 9FD5E9A9
[+] TYPE: NTAG 213C 144bytes (NT2H1311C1DTL)  
[+] Reading tag memory...
[#] Cmd Error: card timeout. len: 0
[#] Cmd Error: card timeout. len: 0
[#] Cmd Error: card timeout. len: 0
[#] Cmd Error: card timeout. len: 0
[!] ⚠️  iso14443a card select failed
[#] Warning: HF field is off, ignoring TransmitFor14443a command
[!] ⚠️  iso14443a card select failed
[#] Warning: HF field is off, ignoring TransmitFor14443a command

[=] MFU dump file information
[=] -------------------------------------------------------------
[=] Version..... 00 00 00 00 00 00 00 00 
[=] TBD 0....... 00 00 
[=] TBD 1....... 00 
[=] Signature... 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[=]              00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[=] Counter 0... 00 00 00 
[=] Tearing 0... 00 
[=] Counter 1... 00 00 00 
[=] Tearing 1... 00 
[=] Counter 2... 00 00 00 
[=] Tearing 2... BD 
[=] Max data page... 43 ( 176 bytes )
[=] Header size..... 56 bytes

[=] -------------------------------------------------------------
[=] block#   | data        |lck| ascii
[=] ---------+-------------+---+------
[=]   0/0x00 | 04 4C 8F 4F |   | .L.O
[=]   1/0x01 | BA A0 11 90 |   | ....
[=]   2/0x02 | 9B 48 FF FF |   | .H..
[=]   3/0x03 | E1 10 12 00 | 1 | ....
[=]   4/0x04 | 03 20 D1 01 | 1 | . ..
[=]   5/0x05 | 1C 55 02 70 | 1 | .U.p
[=]   6/0x06 | 68 69 6C 69 | 1 | hili
[=]   7/0x07 | 70 73 2E 63 | 1 | ps.c
[=]   8/0x08 | 6F 6D 2F 6E | 1 | om/n
[=]   9/0x09 | 66 63 62 72 | 1 | fcbr
[=]  10/0x0A | 75 73 68 68 | 1 | ushh
[=]  11/0x0B | 65 61 64 74 | 1 | eadt
[=]  12/0x0C | 61 70 FE 00 | 1 | ap..
[=]  13/0x0D | 00 00 00 00 | 1 | ....
[=]  14/0x0E | 00 00 00 00 | 1 | ....
[=]  15/0x0F | 00 00 00 00 | 1 | ....
[=]  16/0x10 | 00 00 00 00 | 0 | ....
[=]  17/0x11 | 00 00 00 00 | 0 | ....
[=]  18/0x12 | 00 00 00 00 | 0 | ....
[=]  19/0x13 | 00 00 00 00 | 0 | ....
[=]  20/0x14 | 00 00 00 00 | 0 | ....
[=]  21/0x15 | 00 00 00 00 | 0 | ....
[=]  22/0x16 | 00 00 00 00 | 0 | ....
[=]  23/0x17 | 00 00 00 00 | 0 | ....
[=]  24/0x18 | 00 00 00 00 | 0 | ....
[=]  25/0x19 | 00 00 00 00 | 0 | ....
[=]  26/0x1A | 00 00 00 00 | 0 | ....
[=]  27/0x1B | 00 00 00 00 | 0 | ....
[=]  28/0x1C | 00 00 00 00 | 0 | ....
[=]  29/0x1D | 00 00 00 00 | 0 | ....
[=]  30/0x1E | 00 00 00 00 | 0 | ....
[=]  31/0x1F | 00 01 09 00 | 0 | ....
[=]  32/0x20 | 00 00 01 02 | 0 | ....
[=]  33/0x21 | 60 54 32 32 | 0 | `T22
[=]  34/0x22 | 30 37 30 39 | 0 | 0709
[=]  35/0x23 | 20 34 33 54 | 0 |  43T
[=]  36/0x24 | E5 61 02 01 | 0 | .a..
[=]  37/0x25 | 00 00 04 00 | 0 | ....
[=]  38/0x26 | AD 2F 7D 04 | 0 | ./}.
[=]  39/0x27 | 00 00 00 01 | 0 | ....
[=]  40/0x28 | 00 03 30 BD | 0 | ..0.
[=]  41/0x29 | 04 00 00 10 | 0 | ....
[=]  42/0x2A | 43 00 00 00 | 0 | C...
[=]  43/0x2B | 9F D5 E9 A9 | 0 | ....
[=]  44/0x2C | 00 00 00 00 | 0 | ....
[=] ---------------------------------
[=] Using UID as filename
*** buffer overflow detected ***: terminated

Thread 5 "WorkerThread" received signal SIGABRT, Aborted.
[Switching to Thread 0x7ffff127c6c0 (LWP 11627)]
0x00007ffff66afa4c in ?? () from /lib64/libc.so.6
(gdb) backtrace
#0  0x00007ffff66afa4c in  () at /lib64/libc.so.6
#1  0x00007ffff6660d96 in raise () at /lib64/libc.so.6
#2  0x00007ffff664a87c in abort () at /lib64/libc.so.6
#3  0x00007ffff664b5b7 in  () at /lib64/libc.so.6
#4  0x00007ffff673ef3b in  () at /lib64/libc.so.6
#5  0x00007ffff673d766 in  () at /lib64/libc.so.6
#6  0x00007ffff673d345 in __snprintf_chk () at /lib64/libc.so.6
#7  0x00005555557250b1 in snprintf (__fmt=0x5555558facdb "%.*s%s", __n=58, __s=0x7fffe803ba96 "") at /usr/include/bits/stdio2.h:54
#8  newfilenamemcopyEx (preferredName=0x7ffff127afa0 "hf-mfu-044C8FBAA01190-dump", preferredName@entry=0xec <error: Cannot access memory at address 0xec>, suffix=suffix@entry=0x5555558e8fee ".bin", e_save_path=e_save_path@entry=spDefault) at src/fileutils.c:221
#9  0x00005555557251bd in newfilenamemcopy (suffix=0x5555558e8fee ".bin", preferredName=0xec <error: Cannot access memory at address 0xec>) at src/fileutils.c:185
#10 saveFile (preferredName=preferredName@entry=0x7ffff127afa0 "hf-mfu-044C8FBAA01190-dump", suffix=suffix@entry=0x5555558e8fee ".bin", data=data@entry=0x7ffff127ab20, datalen=datalen@entry=236) at src/fileutils.c:240
#11 0x0000555555729c57 in pm3_save_dump (fn=fn@entry=0x7ffff127afa0 "hf-mfu-044C8FBAA01190-dump", d=d@entry=0x7ffff127ab20 "", n=236, jsft=jsft@entry=jsfMfuMemory, blocksize=blocksize@entry=4) at src/fileutils.c:2125
#12 0x000055555568f0cc in CmdHF14AMfUDump (Cmd=<optimized out>) at src/cmdhfmfu.c:2680
#13 0x00005555556ef25e in CmdsParse (Commands=0x5555559a9180 <CommandTable>, Cmd=0x7fffe803ba77 "dump -k 9FD5E9A9") at src/cmdparser.c:321
#14 0x00005555556ef25e in CmdsParse (Commands=0x5555559a4b00 <CommandTable>, Cmd=0x7fffe803ba73 "mfu dump -k 9FD5E9A9") at src/cmdparser.c:321
#15 0x00005555556ef25e in CmdsParse (Commands=Commands@entry=0x5555559ac920 <CommandTable>, Cmd=Cmd@entry=0x7fffe803ba70 "hf mfu dump -k 9FD5E9A9") at src/cmdparser.c:321
#16 0x00005555556edd23 in CommandReceived (Cmd=Cmd@entry=0x7fffe803ba70 "hf mfu dump -k 9FD5E9A9") at src/cmdmain.c:365
#17 0x000055555575dba5 in main_loop (script_cmds_file=<optimized out>, script_cmd=<optimized out>, stayInCommandLoop=true) at src/proxmark3.c:404
#18 0x00007ffff6ca567f in  () at /usr/lib64/libQt5Core.so.5
#19 0x00007ffff66add14 in  () at /lib64/libc.so.6
#20 0x00007ffff672f010 in  () at /lib64/libc.so.6
(gdb) 

```

iceman1001, 2023-07-10:
I am curious,  the reserved space should be more than enough now.
What is your default paths set to? 

`pref show`

dperret, 2023-07-10:
```
[usb] pm3 --> pref show

[=] Using /root/.proxmark3/preferences.json
[=] Current settings
[=]     emoji.................. emoji
[=]     hints.................. on
[=]     color.................. ansi
[=]     Plot window............ X   10 Y   56 H  400 W  800
[=]     Slider/Overlay window.. X   10 Y  490 H  200 W  800
[=]     default save path...... /root
[=]     dump save path......... /root
[=]     trace save path........ /root
[=]     client debug........... off
[=]     show plot sliders...... on
[=]     barmode................ value
[=]     Cmd execution delay.... 0
[=]     output................. normal

[usb] pm3 --> 

```

iceman1001, 2023-07-10:
I pushed another fix.  Pull latest and compile.  I am having a hard time seeing your values would have generated any overflows.

You would need to go into that frame when back tracking and look at variables.

dperret, 2023-07-10:
Trying to step through to where it crashes:
```
~/source/proxmark3 $ sudo gdb ./client/proxmark3 
GNU gdb (Gentoo 13.2 vanilla) 13.2
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://bugs.gentoo.org/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./client/proxmark3...
(gdb) break newfilenamemcopyEx
Breakpoint 1 at 0x1d0f60: file src/fileutils.c, line 188.
(gdb) run /dev/ttyACM0
Starting program: /home/redbaron/source/proxmark3/client/proxmark3 /dev/ttyACM0
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
[=] Session log /root/.proxmark3/logs/log_20230710.txt
[+] loaded from JSON file /root/.proxmark3/preferences.json
[=] Using UART port /dev/ttyACM0
[New Thread 0x7ffff40d76c0 (LWP 22323)]
[=] Communicating with PM3 over USB-CDC

  8888888b.  888b     d888  .d8888b.   
  888   Y88b 8888b   d8888 d88P  Y88b  
  888    888 88888b.d88888      .d88P  
  888   d88P 888Y88888P888     8888"  
  8888888P"  888 Y888P 888      "Y8b.  
  888        888  Y8P  888 888    888  
  888        888   "   888 Y88b  d88P 
  888        888       888  "Y8888P"    [ ☕ ]

[New Thread 0x7ffff33266c0 (LWP 22324)]
[Detaching after fork from child process 22325]
[New Thread 0x7ffff2b256c0 (LWP 22326)]
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[New Thread 0x7ffff127c6c0 (LWP 22329)]
  [ Proxmark3 RFID instrument ]

    MCU....... AT91SAM7S512 Rev A
    Memory.... 512 KB ( 67% used )

    Client.... Iceman/master/v4.16717-64-g588dea400 2023-07-10 02:49:32
    Bootrom... Iceman/master/v4.16717-64-g588dea400 2023-07-10 02:49:27 
    OS........ Iceman/master/v4.16717-64-g588dea400 2023-07-10 02:49:36 
    Target.... RDV4

[usb] pm3 --> hf mf autopwn
[!] ⚠️  iso14443a card select failed
[usb] pm3 --> hf mf autopwn
[!] ⚠️  iso14443a card select failed
[usb] pm3 --> hf mf autopwn
[!] ⚠️  no known key was supplied, key recovery might fail
[+] loaded 56 keys from hardcoded default array
[=] running strategy 1
[=] Chunk 1.0s | found 32/32 keys (56)
[+] target sector   0 key type A -- found valid key [ FFFFFFFFFFFF ] (used for nested / hardnested attack)
[+] target sector   0 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   1 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   1 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   2 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   2 key type B -- found valid key [ FFFFFFFFFFFF ]
[... 132 lines cut ...]
(gdb) s
219	    snprintf(pfn, len, "%.*s%s", p_namelen, preferredName, suffix);
(gdb) print pfn
$6 = 0x7fffe803b566 ""
(gdb) print len
$7 = 2000
(gdb) print p_namelen
$8 = 18
(gdb) print preferredName
$9 = 0x7fffe803b120 "hf-mf-3631862B-key.bin"
(gdb) print suffix
$10 = 0x5555558e8fee ".bin"
(gdb) s
0x0000555555725035 in snprintf (__fmt=0x5555558facdb "%.*s%s", __n=2000, __s=0x7fffe803b566 "") at /usr/include/bits/stdio2.h:54
54	  return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
(gdb) s
*** buffer overflow detected ***: terminated

Thread 5 "WorkerThread" received signal SIGABRT, Aborted.
0x00007ffff66afa4c in ?? () from /lib64/libc.so.6
(gdb) frame
#0  0x00007ffff66afa4c in ?? () from /lib64/libc.so.6
(gdb) 

```

iceman1001, 2023-07-10:
Thanks,   excellent debugging info. 

the line 219,     is meant to copy only preferedName without the extension. 
```
    snprintf(pfn, len, "%.*s%s", p_namelen, preferredName, suffix);
```

pfn  is a pointer to fileName array.
can you print it aswell.

dperret, 2023-07-10:
pfn looks like it's pointing to an empty string
```
(gdb) print pfn
$6 = 0x7fffe803b566 ""
```
Another run, stopping at line 219, before the crash
```
~/source/proxmark3 $ sudo gdb ./client/proxmark3 
GNU gdb (Gentoo 13.2 vanilla) 13.2
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://bugs.gentoo.org/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./client/proxmark3...
(gdb) break newfilenamemcopyEx
Breakpoint 1 at 0x1d0f60: file src/fileutils.c, line 188.
(gdb) run /dev/ttyACM0
Starting program: /home/redbaron/source/proxmark3/client/proxmark3 /dev/ttyACM0
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
[=] Session log /root/.proxmark3/logs/log_20230710.txt
[+] loaded from JSON file /root/.proxmark3/preferences.json
[=] Using UART port /dev/ttyACM0
[New Thread 0x7ffff40d76c0 (LWP 22404)]
[=] Communicating with PM3 over USB-CDC

  8888888b.  888b     d888  .d8888b.   
  888   Y88b 8888b   d8888 d88P  Y88b  
  888    888 88888b.d88888      .d88P  
  888   d88P 888Y88888P888     8888"  
  8888888P"  888 Y888P 888      "Y8b.  
  888        888  Y8P  888 888    888  
  888        888   "   888 Y88b  d88P 
  888        888       888  "Y8888P"    [ ☕ ]

[New Thread 0x7ffff33266c0 (LWP 22405)]
[Detaching after fork from child process 22406]
[New Thread 0x7ffff2b256c0 (LWP 22407)]
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[New Thread 0x7ffff127c6c0 (LWP 22408)]
  [ Proxmark3 RFID instrument ]

    MCU....... AT91SAM7S512 Rev A
    Memory.... 512 KB ( 67% used )

    Client.... Iceman/master/v4.16717-64-g588dea400 2023-07-10 02:49:32
    Bootrom... Iceman/master/v4.16717-64-g588dea400 2023-07-10 02:49:27 
    OS........ Iceman/master/v4.16717-64-g588dea400 2023-07-10 02:49:36 
    Target.... RDV4

[usb] pm3 --> hf search
 🕕  Searching for ISO14443-A tag...          
[+]  UID: 36 31 86 2B 
[+] ATQA: 00 04
[+]  SAK: 08 [2]
[+] Possible types:
[+]    MIFARE Classic 1K
[=] proprietary non iso14443-4 card found, RATS not supported
[+] Prng detection: weak
[#] Auth error
[... 119 lines cut ...]
1115	    size_t ls = strlen(s);
(gdb) s
1116	    size_t lsuffix = strlen(suffix);
(gdb) s
1117	    if (ls >= lsuffix) {
(gdb) s
1118	        return strncmp(suffix, s + (ls - lsuffix), lsuffix) == 0;
(gdb) s
newfilenamemcopyEx (preferredName=preferredName@entry=0x7fffe803bb70 "hf-mf-3631862B-key.bin", suffix=suffix@entry=0x5555558e8fee ".bin", e_save_path=e_save_path@entry=spDump) at src/fileutils.c:216
216	        p_namelen -= strlen(suffix);
(gdb) s
219	    snprintf(pfn, len, "%.*s%s", p_namelen, preferredName, suffix);
(gdb) print pfn
$2 = 0x7fffe803c6d6 ""
(gdb) print len
$3 = 2000
(gdb) print p_namelen
$4 = 18
(gdb) print preferredName
$5 = 0x7fffe803bb70 "hf-mf-3631862B-key.bin"
(gdb) print suffix
$6 = 0x5555558e8fee ".bin"
(gdb) 

```

iceman1001, 2023-07-10:
pfn is a pointer that jumps ahead in the fileName array. 
so I am curious if it jumps one too far
in this step
` pfn += save_path_len + strlen(PATHSEP);`

dperret, 2023-07-10:
Latest run with pfn from that step
```
~/source/proxmark3 $ sudo gdb ./client/proxmark3 
GNU gdb (Gentoo 13.2 vanilla) 13.2
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://bugs.gentoo.org/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./client/proxmark3...
(gdb) break newfilenamemcopyEx
Breakpoint 1 at 0x1d0f60: file src/fileutils.c, line 188.
(gdb) run /dev/ttyACM0
Starting program: /home/redbaron/source/proxmark3/client/proxmark3 /dev/ttyACM0
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
[=] Session log /root/.proxmark3/logs/log_20230710.txt
[+] loaded from JSON file /root/.proxmark3/preferences.json
[=] Using UART port /dev/ttyACM0
[New Thread 0x7ffff40d76c0 (LWP 22448)]
[=] Communicating with PM3 over USB-CDC

  8888888b.  888b     d888  .d8888b.   
  888   Y88b 8888b   d8888 d88P  Y88b  
  888    888 88888b.d88888      .d88P  
  888   d88P 888Y88888P888     8888"  
  8888888P"  888 Y888P 888      "Y8b.  
  888        888  Y8P  888 888    888  
  888        888   "   888 Y88b  d88P 
  888        888       888  "Y8888P"    [ ☕ ]

[New Thread 0x7ffff33266c0 (LWP 22449)]
[Detaching after fork from child process 22450]
[New Thread 0x7ffff2b256c0 (LWP 22451)]
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[New Thread 0x7ffff127c6c0 (LWP 22452)]
  [ Proxmark3 RFID instrument ]

    MCU....... AT91SAM7S512 Rev A
    Memory.... 512 KB ( 67% used )

    Client.... Iceman/master/v4.16717-64-g588dea400 2023-07-10 02:49:32
    Bootrom... Iceman/master/v4.16717-64-g588dea400 2023-07-10 02:49:27 
    OS........ Iceman/master/v4.16717-64-g588dea400 2023-07-10 02:49:36 
    Target.... RDV4

[usb] pm3 --> hf mf autopwn
[!] ⚠️  No tag found.
[!] ⚠️  no known key was supplied, key recovery might fail
[+] loaded 56 keys from hardcoded default array
[=] running strategy 1
[=] Chunk 1.5s | found 32/32 keys (56)
[+] target sector   0 key type A -- found valid key [ FFFFFFFFFFFF ] (used for nested / hardnested attack)
[+] target sector   0 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   1 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   1 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   2 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   2 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   3 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   3 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   4 key type A -- found valid key [ FFFFFFFFFFFF ]
[... 220 lines cut ...]
No symbol "PATHSEP" in current context.
(gdb) s
214	    uint16_t p_namelen = strlen(preferredName);
(gdb) print p_namelen
$12 = <optimized out>
(gdb) print preferredName
$13 = 0x7ffff127b3b0 "hf-mf-3631862B-dump"
(gdb) s
215	    if (str_endswith(preferredName, suffix)) {
(gdb) s
str_endswith (s=s@entry=0x7ffff127b3b0 "hf-mf-3631862B-dump", suffix=suffix@entry=0x5555558e8fee ".bin") at src/util.c:1114
1114	bool str_endswith(const char *s,  const char *suffix) {
(gdb) c
Continuing.
*** buffer overflow detected ***: terminated

Thread 5 "WorkerThread" received signal SIGABRT, Aborted.
0x00007ffff66afa4c in ?? () from /lib64/libc.so.6
(gdb) frame
#0  0x00007ffff66afa4c in ?? () from /lib64/libc.so.6
(gdb) print pfn
No symbol "pfn" in current context.
(gdb) 

```

iceman1001, 2023-07-10:
I cant replicate it. 

The offending line is:
```  snprintf(pfn, len, "%.*s%s", p_namelen, preferredName, suffix); ```

It seem that on pentoo it doesnt like to append the pfn point despite it points to valid reserved memory

iceman1001, 2023-07-23:
ping

dperret, 2023-07-24:
It is definitely failing on that line of code, but I haven't figured out why yet. Possibly related to a compiler option? I have that line and the necessary variables pulled out into a minimum test case, and can reproduce the crash consistently.

```
pentoo ~ # cat test.c 
#include <ctype.h>
#include <stdio.h>
#include <stdint.h>

int main()
{
char *pfn;
size_t len;
uint16_t p_namelen;
char *preferredName;
char *suffix;

pfn = "";
len = 1000;
p_namelen = 18;
preferredName = "hf-mf-3631862B-key.bin";
suffix = ".bin";

printf("pfn: %s \n", pfn);
printf("len: %lu \n", len);
printf("p_namelen: %u \n", p_namelen);
printf("preferredName: %s \n", preferredName);
printf("suffix: %s \n", suffix);

snprintf(pfn, len, "%.*s%s", p_namelen, preferredName, suffix);
return 0;
}
pentoo ~ # gcc -v -g -o test.out -Wall test.c
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-pc-linux-gnu/12/lto-wrapper
Target: x86_64-pc-linux-gnu
Configured with: /var/tmp/portage/sys-devel/gcc-12.3.1_p20230526/work/gcc-12-20230526/configure --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --prefix=/usr --bindir=/usr/x86_64-pc-linux-gnu/gcc-bin/12 --includedir=/usr/lib/gcc/x86_64-pc-linux-gnu/12/include --datadir=/usr/share/gcc-data/x86_64-pc-linux-gnu/12 --mandir=/usr/share/gcc-data/x86_64-pc-linux-gnu/12/man --infodir=/usr/share/gcc-data/x86_64-pc-linux-gnu/12/info --with-gxx-include-dir=/usr/lib/gcc/x86_64-pc-linux-gnu/12/include/g++-v12 --with-python-dir=/share/gcc-data/x86_64-pc-linux-gnu/12/python --enable-languages=c,c++,fortran --enable-obsolete --enable-secureplt --disable-werror --with-system-zlib --enable-nls --without-included-gettext --disable-libunwind-exceptions --enable-checking=release --with-bugurl=https://bugs.gentoo.org/ --with-pkgversion='Gentoo Hardened 12.3.1_p20230526 p2' --with-gcc-major-version-only --enable-libstdcxx-time --enable-lto --disable-libstdcxx-pch --enable-shared --enable-threads=posix --enable-__cxa_atexit --enable-clocale=gnu --enable-multilib --with-multilib-list=m32,m64 --disable-fixed-point --enable-targets=all --enable-libgomp --disable-libssp --disable-libada --enable-cet --disable-systemtap --disable-valgrind-annotations --disable-vtable-verify --disable-libvtv --with-zstd --without-isl --enable-default-pie --enable-default-ssp --with-build-config='bootstrap-lto bootstrap-cet'
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 12.3.1 20230526 (Gentoo Hardened 12.3.1_p20230526 p2) 
COLLECT_GCC_OPTIONS='-v' '-g' '-o' 'test.out' '-Wall' '-mtune=generic' '-march=x86-64' '-dumpdir' 'test.out-'
 /usr/libexec/gcc/x86_64-pc-linux-gnu/12/cc1 -quiet -v test.c -quiet -dumpdir test.out- -dumpbase test.c -dumpbase-ext .c -mtune=generic -march=x86-64 -g -Wall -version -fcf-protection -o /tmp/ccJfpL2V.s
GNU C17 (Gentoo Hardened 12.3.1_p20230526 p2) version 12.3.1 20230526 (x86_64-pc-linux-gnu)
	compiled by GNU C version 12.3.1 20230526, GMP version 6.2.1, MPFR version 4.2.0-p9, MPC version 1.3.1, isl version none
GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
ignoring nonexistent directory "/usr/local/include"
ignoring nonexistent directory "/usr/lib/gcc/x86_64-pc-linux-gnu/12/../../../../x86_64-pc-linux-gnu/include"
#include "..." search starts here:
#include <...> search starts here:
 /usr/lib/gcc/x86_64-pc-linux-gnu/12/include
 /usr/lib/gcc/x86_64-pc-linux-gnu/12/include-fixed
 /usr/include
End of search list.
GNU C17 (Gentoo Hardened 12.3.1_p20230526 p2) version 12.3.1 20230526 (x86_64-pc-linux-gnu)
	compiled by GNU C version 12.3.1 20230526, GMP version 6.2.1, MPFR version 4.2.0-p9, MPC version 1.3.1, isl version none
GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
Compiler executable checksum: bc4b694fa98acf2817ad97cab61bc32f
COLLECT_GCC_OPTIONS='-v' '-g' '-o' 'test.out' '-Wall' '-mtune=generic' '-march=x86-64' '-dumpdir' 'test.out-'
 /usr/lib/gcc/x86_64-pc-linux-gnu/12/../../../../x86_64-pc-linux-gnu/bin/as -v --gdwarf-5 --64 -o /tmp/ccFLp2m6.o /tmp/ccJfpL2V.s
GNU assembler version 2.40.0 (x86_64-pc-linux-gnu) using BFD version (Gentoo 2.40 p5) 2.40.0
COMPILER_PATH=/usr/libexec/gcc/x86_64-pc-linux-gnu/12/:/usr/libexec/gcc/x86_64-pc-linux-gnu/12/:/usr/libexec/gcc/x86_64-pc-linux-gnu/:/usr/lib/gcc/x86_64-pc-linux-gnu/12/:/usr/lib/gcc/x86_64-pc-linux-gnu/:/usr/lib/gcc/x86_64-pc-linux-gnu/12/../../../../x86_64-pc-linux-gnu/bin/
LIBRARY_PATH=/usr/lib/gcc/x86_64-pc-linux-gnu/12/:/usr/lib/gcc/x86_64-pc-linux-gnu/12/../../../../lib64/:/lib/../lib64/:/usr/lib/../lib64/:/usr/lib/gcc/x86_64-pc-linux-gnu/12/../../../../x86_64-pc-linux-gnu/lib/:/usr/lib/gcc/x86_64-pc-linux-gnu/12/../../../:/lib/:/usr/lib/
COLLECT_GCC_OPTIONS='-v' '-g' '-o' 'test.out' '-Wall' '-mtune=generic' '-march=x86-64' '-dumpdir' 'test.out.'
 /usr/libexec/gcc/x86_64-pc-linux-gnu/12/collect2 -plugin /usr/libexec/gcc/x86_64-pc-linux-gnu/12/liblto_plugin.so -plugin-opt=/usr/libexec/gcc/x86_64-pc-linux-gnu/12/lto-wrapper -plugin-opt=-fresolution=/tmp/ccnboGUh.res -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lgcc_s --eh-frame-hdr -m elf_x86_64 -dynamic-linker /lib64/ld-linux-x86-64.so.2 -pie -z now -o test.out /usr/lib/gcc/x86_64-pc-linux-gnu/12/../../../../lib64/Scrt1.o /usr/lib/gcc/x86_64-pc-linux-gnu/12/../../../../lib64/crti.o /usr/lib/gcc/x86_64-pc-linux-gnu/12/crtbeginS.o -L/usr/lib/gcc/x86_64-pc-linux-gnu/12 [... cut ...]
Starting program: /root/test.out 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
pfn:  
len: 1000 
p_namelen: 18 
preferredName: hf-mf-3631862B-key.bin 
suffix: .bin 

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7e39618 in ?? () from /lib64/libc.so.6
(gdb) backtrace
#0  0x00007ffff7e39618 in ?? () from /lib64/libc.so.6
#1  0x00007ffff7e14666 in snprintf () from /lib64/libc.so.6
#2  0x000055555555525a in main () at test.c:26
(gdb) 

```

iceman1001, 2023-07-24:
`char *pfn;`   is empty,   need to alloc mem for it to work..

dperret, 2023-07-25:
Fixing the test program to allocate *pfn in the same way that newfilenamemcopyEx does a few lines earlier fixes the test program, but the proxmark3 client built from the latest in master still crashes on the same line.

test program
```
pentoo ~ # cat test.c 
#include <ctype.h>
#include <stdio.h>
#include <stdint.h>
#include <stdlib.h>

int main()
{
size_t len;
uint16_t p_namelen;
char *preferredName;
char *suffix;

len = 1000;
p_namelen = 18;
preferredName = "hf-mf-3631862B-key.bin";
suffix = ".bin";

char *fileName = (char *) calloc(len, sizeof(uint8_t));
char *pfn = fileName;

printf("pfn: %s \n", pfn);
printf("len: %lu \n", len);
printf("p_namelen: %u \n", p_namelen);
printf("preferredName: %s \n", preferredName);
printf("suffix: %s \n", suffix);

snprintf(pfn, len, "%.*s%s", p_namelen, preferredName, suffix);
return 0;
}
pentoo ~ # gcc -v -g -o test.out -Wall test.c
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-pc-linux-gnu/12/lto-wrapper
Target: x86_64-pc-linux-gnu
Configured with: /var/tmp/portage/sys-devel/gcc-12.3.1_p20230526/work/gcc-12-20230526/configure --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --prefix=/usr --bindir=/usr/x86_64-pc-linux-gnu/gcc-bin/12 --includedir=/usr/lib/gcc/x86_64-pc-linux-gnu/12/include --datadir=/usr/share/gcc-data/x86_64-pc-linux-gnu/12 --mandir=/usr/share/gcc-data/x86_64-pc-linux-gnu/12/man --infodir=/usr/share/gcc-data/x86_64-pc-linux-gnu/12/info --with-gxx-include-dir=/usr/lib/gcc/x86_64-pc-linux-gnu/12/include/g++-v12 --with-python-dir=/share/gcc-data/x86_64-pc-linux-gnu/12/python --enable-languages=c,c++,fortran --enable-obsolete --enable-secureplt --disable-werror --with-system-zlib --enable-nls --without-included-gettext --disable-libunwind-exceptions --enable-checking=release --with-bugurl=https://bugs.gentoo.org/ --with-pkgversion='Gentoo Hardened 12.3.1_p20230526 p2' --with-gcc-major-version-only --enable-libstdcxx-time --enable-lto --disable-libstdcxx-pch --enable-shared --enable-threads=posix --enable-__cxa_atexit --enable-clocale=gnu --enable-multilib --with-multilib-list=m32,m64 --disable-fixed-point --enable-targets=all --enable-libgomp --disable-libssp --disable-libada --enable-cet --disable-systemtap --disable-valgrind-annotations --disable-vtable-verify --disable-libvtv --with-zstd --without-isl --enable-default-pie --enable-default-ssp --with-build-config='bootstrap-lto bootstrap-cet'
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 12.3.1 20230526 (Gentoo Hardened 12.3.1_p20230526 p2) 
COLLECT_GCC_OPTIONS='-v' '-g' '-o' 'test.out' '-Wall' '-mtune=generic' '-march=x86-64' '-dumpdir' 'test.out-'
 /usr/libexec/gcc/x86_64-pc-linux-gnu/12/cc1 -quiet -v test.c -quiet -dumpdir test.out- -dumpbase test.c -dumpbase-ext .c -mtune=generic -march=x86-64 -g -Wall -version -fcf-protection -o /tmp/ccVU4NFH.s
GNU C17 (Gentoo Hardened 12.3.1_p20230526 p2) version 12.3.1 20230526 (x86_64-pc-linux-gnu)
	compiled by GNU C version 12.3.1 20230526, GMP version 6.2.1, MPFR version 4.2.0-p9, MPC version 1.3.1, isl version none
GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
ignoring nonexistent directory "/usr/local/include"
ignoring nonexistent directory "/usr/lib/gcc/x86_64-pc-linux-gnu/12/../../../../x86_64-pc-linux-gnu/include"
#include "..." search starts here:
#include <...> search starts here:
 /usr/lib/gcc/x86_64-pc-linux-gnu/12/include
 /usr/lib/gcc/x86_64-pc-linux-gnu/12/include-fixed
 /usr/include
End of search list.
GNU C17 (Gentoo Hardened 12.3.1_p20230526 p2) version 12.3.1 20230526 (x86_64-pc-linux-gnu)
	compiled by GNU C version 12.3.1 20230526, GMP version 6.2.1, MPFR version 4.2.0-p9, MPC version 1.3.1, isl version none
GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
Compiler executable checksum: bc4b694fa98acf2817ad97cab61bc32f
COLLECT_GCC_OPTIONS='-v' '-g' '-o' 'test.out' '-Wall' '-mtune=generic' '-march=x86-64' '-dumpdir' 'test.out-'
 /usr/lib/gcc/x86_64-pc-linux-gnu/12/../../../../x86_64-pc-linux-gnu/bin/as -v --gdwarf-5 --64 -o /tmp/cc9pDPY4.o /tmp/ccVU4NFH.s
GNU assembler version 2.40.0 (x86_64-pc-linux-gnu) using BFD version (Gentoo 2.40 p5) 2.40.0
COMPILER_PATH=/usr/libexec/gcc/x86_64-pc-linux-gnu/12/:/usr/libexec/gcc/x86_64-pc-linux-gnu/12/:/usr/libexec/gcc/x86_64-pc-linux-gnu/:/usr/lib/gcc/x86_64-pc-linux-gnu/12/:/usr/lib/gcc/x86_64-pc-linux-gnu/:/usr/lib/gcc/x86_64-pc-linux-gnu/12/../../../../x86_64-pc-linux-gnu/bin/
LIBRARY_PATH=/usr/lib/gcc/x86_64-pc-linux-gnu/12/:/usr/lib/gcc/x86_64-pc-linux-gnu/12/../../../../lib64/:/lib/../lib64/:/usr/lib/../lib64/:/usr/lib/gcc/x86_64-pc-linux-gnu/12/../../../../x86_64-pc-linux-gnu/lib/:/usr/lib/gcc/x86_64-pc-linux-gnu/12/../../../:/lib/:/usr/lib/
COLLECT_GCC_OPTIONS='-v' '-g' '-o' 'test.out' '-Wall' '-mtune=generic' '-march=x86-64' '-dumpdir' 'test.out.'
 /usr/libexec/gcc/x86_64-pc-linux-gnu/12/collect2 -plugin /usr/libexec/gcc/x86_64-pc-linux-gnu/12/liblto_plugin.so -plugin-opt=/usr/libexec/gcc/x86_64-pc-linux-gnu/12/lto-wrapper -plugin-opt=-fresolution=/tmp/cc6Lx68G.res -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lgcc_s --eh-frame-hdr -m elf_x86_64 -dynamic-linker /lib64/ld-linux-x86-64.so.2 -pie -z now -o test.out /usr/lib/gcc/x86_64-pc-linux-gnu/12/../../../../lib64/Scrt1.o /usr/lib/gcc/x86_64-pc-linux-gnu/12/../../../../lib64/crti.o /usr/lib/gcc/x86_64- [... cut ...]
#13 0x00005555556f55fe in CmdsParse (Commands=Commands@entry=0x5555559b8c40 <CommandTable>, Cmd=Cmd@entry=0x7fffe801c680 "hf mf autopwn") at src/cmdparser.c:321
#14 0x00005555556f40c3 in CommandReceived (Cmd=Cmd@entry=0x7fffe801c680 "hf mf autopwn") at src/cmdmain.c:365
#15 0x0000555555765d65 in main_loop (script_cmds_file=<optimized out>, script_cmd=<optimized out>, stayInCommandLoop=true) at src/proxmark3.c:407
#16 0x00007ffff6ca59ff in  () at /usr/lib64/libQt5Core.so.5
#17 0x00007ffff66add14 in  () at /lib64/libc.so.6
#18 0x00007ffff672f010 in  () at /lib64/libc.so.6
(gdb) 

```

iceman1001, 2023-07-25:
Lets see,   try calling with with a prefered name without file extension in it.   I have an idea.  If that doesn't crash,   then its related to the removal of the suffix before.   I  think it could be the p_namelen var.  if suffix isn't zero term,  then strlen will be wrong.

```
    uint16_t p_namelen = strlen(preferredName);
    if (str_endswith(preferredName, suffix)) {
        p_namelen -= strlen(suffix);
    }
    // modify filename
    snprintf(pfn, len, "%.*s%s", p_namelen, preferredName, suffix);
    ```

iceman1001, 2023-07-25:
```
    // remove file extension if exist in name
    size_t p_namelen = strlen(preferredName);
    printf("p_namelen... %zu ", p_namelen);
    if (str_endswith(preferredName, suffix)) {
        printf("suffix len... %zu ", strlen(suffix));
        p_namelen -= strlen(suffix);
        printf("p_namelen after... %zu ", p_namelen);
    }
    ```
    
    p_namelen was u16,   strlen returns size_t which could be as large as 8bytes on some systems.

dperret, 2023-07-25:
Test program didn't crash with or without the file extension in the preferredName, and the latest is still crashing
commit https://github.com/RfidResearchGroup/proxmark3/commit/cb9ee94ed6aa63be94a351793ba49ac38244f722
```
~/source/proxmark3 $ sudo gdb ./client/proxmark3 
GNU gdb (Gentoo 13.2 vanilla) 13.2
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://bugs.gentoo.org/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./client/proxmark3...
(gdb) break src/fileutils.c:219
Breakpoint 1 at 0x1d8fdd: file src/fileutils.c, line 219.
(gdb) run /dev/ttyACM0
Starting program: /home/redbaron/source/proxmark3/client/proxmark3 /dev/ttyACM0
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
[=] Session log /root/.proxmark3/logs/log_20230725.txt
[+] loaded from JSON file /root/.proxmark3/preferences.json
[=] Using UART port /dev/ttyACM0
[New Thread 0x7ffff40d76c0 (LWP 14237)]
[=] Communicating with PM3 over USB-CDC

  8888888b.  888b     d888  .d8888b.   
  888   Y88b 8888b   d8888 d88P  Y88b  
  888    888 88888b.d88888      .d88P  
  888   d88P 888Y88888P888     8888"  
  8888888P"  888 Y888P 888      "Y8b.  
  888        888  Y8P  888 888    888  
  888        888   "   888 Y88b  d88P 
  888        888       888  "Y8888P"    [ ☕ ]

[New Thread 0x7ffff32fa6c0 (LWP 14238)]
[Detaching after fork from child process 14239]
[New Thread 0x7ffff2af96c0 (LWP 14240)]
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[New Thread 0x7ffff124f6c0 (LWP 14241)]
  [ Proxmark3 RFID instrument ]

    MCU....... AT91SAM7S512 Rev A
    Memory.... 512 KB ( 67% used )

    Client.... Iceman/master/v4.16717-218-gcb9ee94ed 2023-07-25 09:25:13
    Bootrom... Iceman/master/v4.16717-218-gcb9ee94ed 2023-07-25 09:25:08 
    OS........ Iceman/master/v4.16717-218-gcb9ee94ed 2023-07-25 09:25:16 
    Target.... RDV4

[usb] pm3 --> hf mf autopwn
[!] ⚠️  no known key was supplied, key recovery might fail
[+] loaded 56 keys from hardcoded default array
[=] running strategy 1
[=] ..
[=] Chunk 4.1s | found 19/32 keys (56)
[=] running strategy 2
[=] ..
[=] Chunk 4.1s | found 19/32 keys (56)
[+] target sector   0 key type B -- found valid key [ FFFFFFFFFFFF ] (used for nested / hardnested attack)
[+] target sector   1 key type A -- found valid key [ 2A2C13CC242A ]
[+] target sector   1 key type B -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   2 key type A -- found valid key [ FFFFFFFFFFFF ]
[+] target sector   2 key type B -- found valid key [ FFFFFFFFFFFF ]
[... 331 lines cut ...]
Thread 5 "WorkerThread" received signal SIGABRT, Aborted.
0x00007ffff66afa4c in ?? () from /lib64/libc.so.6
(gdb) backtrace
#0  0x00007ffff66afa4c in  () at /lib64/libc.so.6
#1  0x00007ffff6660d96 in raise () at /lib64/libc.so.6
#2  0x00007ffff664a87c in abort () at /lib64/libc.so.6
#3  0x00007ffff664b5b7 in  () at /lib64/libc.so.6
#4  0x00007ffff673ef3b in  () at /lib64/libc.so.6
#5  0x00007ffff673d766 in  () at /lib64/libc.so.6
#6  0x00007ffff673d345 in __snprintf_chk () at /lib64/libc.so.6
#7  0x000055555572d022 in snprintf (__fmt=0x5555559042b8 "%.*s%s", __n=1000, __s=0x7fffe803b826 "") at /usr/include/bits/stdio2.h:54
#8  newfilenamemcopyEx (preferredName=preferredName@entry=0x7fffe803b0a0 "hf-mf-3631862B-key.bin", suffix=suffix@entry=0x5555558f1fce ".bin", e_save_path=e_save_path@entry=spDump) at src/fileutils.c:222
#9  0x000055555572f121 in createMfcKeyDump (preferredName=preferredName@entry=0x7fffe803b0a0 "hf-mf-3631862B-key.bin", sectorsCnt=sectorsCnt@entry=16 '\020', e_sector=e_sector@entry=0x7fffe803af10) at src/fileutils.c:779
#10 0x000055555565d0a7 in CmdHF14AMfAutoPWN (Cmd=<optimized out>) at src/cmdhfmf.c:3105
#11 0x00005555556f55fe in CmdsParse (Commands=0x5555559b4560 <CommandTable>, Cmd=0x7fffe801c686 "autopwn") at src/cmdparser.c:321
#12 0x00005555556f55fe in CmdsParse (Commands=0x5555559b0b40 <CommandTable>, Cmd=0x7fffe801c683 "mf autopwn") at src/cmdparser.c:321
#13 0x00005555556f55fe in CmdsParse (Commands=Commands@entry=0x5555559b8c40 <CommandTable>, Cmd=Cmd@entry=0x7fffe801c680 "hf mf autopwn") at src/cmdparser.c:321
#14 0x00005555556f40c3 in CommandReceived (Cmd=Cmd@entry=0x7fffe801c680 "hf mf autopwn") at src/cmdmain.c:365
#15 0x0000555555765da5 in main_loop (script_cmds_file=<optimized out>, script_cmd=<optimized out>, stayInCommandLoop=true) at src/proxmark3.c:407
#16 0x00007ffff6ca59ff in  () at /usr/lib64/libQt5Core.so.5
#17 0x00007ffff66add14 in  () at /lib64/libc.so.6
#18 0x00007ffff672f010 in  () at /lib64/libc.so.6
(gdb) 

```

iceman1001, 2023-07-25:
Good,

Its nothing to do with length of the strings then.  Somehow pfn pointer doesn't point to valid memory anymore.

its down to track it:

```
    char *pfn = fileName;       <---- assigned

    // user preference save paths
    int save_path_len = path_size(e_save_path);
    if (save_path_len) {
        snprintf(pfn, len, "%s%s", g_session.defaultPaths[e_save_path], PATHSEP);
        pfn += save_path_len + strlen(PATHSEP);          <----  verify it doesn't point outside 
    }

  <----  here make a test to see where PFN points at.

    // remove file extension if exist in name
    size_t p_namelen = strlen(preferredName);
    if (str_endswith(preferredName, suffix)) {
        p_namelen -= strlen(suffix);
    }

```

iceman1001, 2023-07-26:
pushed some more checks.  Since I can't trigger it it is up to you to find a root cause and a fix

dperret, 2023-07-27:
Ok, I'll keep looking at it. Here's gdb output, stepping through with the checks from last night

```
pentoo ~/source/proxmark3 $ sudo gdb ./client/proxmark3 
GNU gdb (Gentoo 13.2 vanilla) 13.2
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://bugs.gentoo.org/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./client/proxmark3...
(gdb) break src/fileutils.c:207
Breakpoint 1 at 0x1d92ad: file src/fileutils.c, line 207.
(gdb) break src/fileutils.c:220
Breakpoint 2 at 0x1d9307: file src/fileutils.c, line 220.
(gdb) run /dev/ttyACM0 
Starting program: /home/redbaron/source/proxmark3/client/proxmark3 /dev/ttyACM0
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
[=] Session log /root/.proxmark3/logs/log_20230727.txt
[+] loaded from JSON file `/root/.proxmark3/preferences.json`
[=] Using UART port /dev/ttyACM0
[New Thread 0x7ffff40d76c0 (LWP 12876)]
[=] Communicating with PM3 over USB-CDC

  8888888b.  888b     d888  .d8888b.   
  888   Y88b 8888b   d8888 d88P  Y88b  
  888    888 88888b.d88888      .d88P  
  888   d88P 888Y88888P888     8888"  
  8888888P"  888 Y888P 888      "Y8b.  
  888        888  Y8P  888 888    888  
  888        888   "   888 Y88b  d88P 
  888        888       888  "Y8888P"    [ ☕ ]

[New Thread 0x7ffff32fa6c0 (LWP 12877)]
[Detaching after fork from child process 12878]
[New Thread 0x7ffff2af96c0 (LWP 12879)]
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[New Thread 0x7ffff124f6c0 (LWP 12880)]
  [ Proxmark3 RFID instrument ]

    MCU....... AT91SAM7S512 Rev A
    Memory.... 512 KB ( 67% used )

    Client.... Iceman/master/v4.16717-236-g7f5590d94 2023-07-27 00:32:09
    Bootrom... Iceman/master/v4.16717-236-g7f5590d94 2023-07-27 00:32:04 
    OS........ Iceman/master/v4.16717-236-g7f5590d94 2023-07-27 00:32:12 
    Target.... RDV4

[usb] pm3 --> hf mf autopwn
[!] ⚠️  no known key was supplied, key recovery might fail
[+] loaded 56 keys from hardcoded default array
[=] running strategy 1
[=] ..
[=] Chunk 4.1s | found 19/32 keys (56)
[=] running strategy 2
[=] ..
[=] Chunk 4.1s | found 19/32 keys (56)
[+] target sector   0 key type B -- found valid key [ FFFFFFFFFFFF ] (used for nested / hardnested attack)
[+] target sector   1 key type A -- found valid key [ 2A2C13CC242A ]
[+] target sector   1 key type B -- found valid key [ FFFFFFFFFFFF ]
[... 116 lines cut ...]
snprintf (__fmt=0x55555590433b "%.*s%s", __n=1000, __s=0x7fffe803b886 "") at /usr/include/bits/stdio2.h:54
54	  return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
(gdb) s
220	    snprintf(pfn, len, "%.*s%s", (int)p_namelen, preferredName, suffix);
(gdb) print pfn
$7 = 0x7fffe803b886 ""
(gdb) print len
$8 = 1000
(gdb) print p_namelen
$9 = <optimized out>
(gdb) print preferredName
$10 = 0x7fffe803b100 "hf-mf-3631862B-key.bin"
(gdb) print suffix
$11 = 0x5555558f208e ".bin"
(gdb) s
0x000055555572d331 in snprintf (__fmt=0x55555590433b "%.*s%s", __n=1000, __s=0x7fffe803b886 "") at /usr/include/bits/stdio2.h:54
54	  return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
(gdb) s
*** buffer overflow detected ***: terminated

Thread 5 "WorkerThread" received signal SIGABRT, Aborted.
0x00007ffff66afa4c in ?? () from /lib64/libc.so.6
(gdb) 

```

iceman1001, 2023-07-27:
`214	    size_t p_namelen = strlen(preferredName);`
can you print p_namelen after this line?

dperret, 2023-07-28:
Looks like https://github.com/RfidResearchGroup/proxmark3/commit/19251645e15e5701fd0af09d7bcf93dee20a9c20 fixed it! I tested both the `hf mf autopwn` and `hf mfu dump` commands (on different cards), building from source from commit https://github.com/RfidResearchGroup/proxmark3/commit/97a1f97308fed7231170b0aed864ae2db82c294f and both are running successfully now. Thank you!

closed 2023-07-28 by dperret

iceman1001, 2023-07-28:
That took way longer to fix than I predicted.   Those error messages could be a bit more informative.