RfidResearchGroup/proxmark3 #2037: New ELECTRA fob - em410x- any idea?

jareckib, 2023-07-13:
Any` chance of a new ELECTRA fob? If anything more is needed please let me know.
original ELECTRA fob---  
[lf_electra (1).zip](https://github.com/RfidResearchGroup/proxmark3/files/12027058/lf_electra.1.zip)

`

[usb] pm3 --> lf search

[=] NOTE: some demods output possible binary
[=] if it finds something that looks like a tag
[=] False Positives ARE possible
[=]
[=] Checking for known tags...
[=]
[+] EM 410x Short ID found on a 128b frame
[+]     Data after ID: 7C700000000
[+]     Possibly an Electra (RO), 0x7C7 = 1991
[+]     Short ID details:
[+] EM 410x ID 03E8A955A1
[+] EM410x ( RF/64 )
[=] -------- Possible de-scramble patterns ---------
[+] Unique TAG ID      : C01795AA85
[=] HoneyWell IdentKey
[+]     DEZ 8          : 11097505
[+]     DEZ 10         : 3903411617
[+]     DEZ 5.5        : 59561.21921
[+]     DEZ 3.5A       : 003.21921
[+]     DEZ 3.5B       : 232.21921
[+]     DEZ 3.5C       : 169.21921
[+]     DEZ 14/IK2     : 00016788313505
[+]     DEZ 15/IK3     : 000825029405317
[+]     DEZ 20/ZK      : 12000107090510100805
[=]
[+] Other              : 21921_169_11097505
[+] Pattern Paxton     : 62755745 [0x3BD93A1]
[+] Pattern 1          : 3590663 [0x36CA07]
[+] Pattern Sebury     : 21921 41 2708897  [0x55A1 0x29 0x2955A1]
[+] VD / ID            : 003 / 3903411617
[=] ------------------------------------------------

[+] Valid EM410x ID found!

[=] Couldn't identify a chipset
[usb] pm3 -->`

jareckib, 2023-07-14:
Any chance of copying the new ELECTRA fob? I have information that flipper zero works with the new ELECTRA.

jareckib, 2023-07-14:
I found a solution how to copy ELECTRA. Maybe it will help
lf search
ID:[+] EM 410x ID 03E8A955A1  
lf em 410x clone --id 03E8A955A1
lf t55 detect
lf t55 dump
We read blocks 1 and 2

lf t55 wipe

lf t55xx write -b 0 -d 00148080
lf t55xx write -b 1 -d FF80DD8D
lf t55xx write -b 2 -d 24A5507A
lf t55xx write -b 3 -d 7E1EAAAA
lf t55xx write -b 4 -d AAAAAAAA

jareckib, 2023-07-14:
I hope there will be no problem with the fix for ELECTRA

iceman1001, 2023-07-15:
You don't need to mix commands,  you can just use the t55xx commands
```
lf t55 detect
lf t55 dump
lf t55 restore
```

jareckib, 2023-07-15:
Of course, but I'm copying with EM4102

iceman1001, 2023-07-15:
yes, 
your way needs five different commands.   Mine three.  I tend to keep things simple.

jareckib, 2023-07-15:
Yes, but I have to change block 0 (00148040-00148080)and add block 3 (0x7E1EAAAA) and 4 (0xAAAAAAAA) for this new ELECTRA fob copy to work.

The fob has been tested and opens the door.

jareckib, 2023-07-15:
Maybe you could add the command: lf electra clone --id ???

iceman1001, 2023-07-15:
Feel free to contribute,

jareckib, 2023-07-15:
or add a lua script that will add 3 blocks after reading the ID

jareckib, 2023-07-15:
I've done this for now....A little help????

lf search
lf clone em 410x --id ........
script run electra
[electra.zip](https://github.com/RfidResearchGroup/proxmark3/files/12060828/electra.zip)

jareckib, 2023-07-15:
£5 for five minutes of your work?

closed 2023-07-15 by jareckib