RfidResearchGroup/proxmark3 #2038: Auth error on MIFARE Classic 4K with known keys

deimspb, 2023-07-13:
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  Iceman/master/v4.16717-33-ge43f6804a 2023-07-09 09:31:18 5e7ed8251
  compiled with............. MinGW-w64 10.3.0
  platform.................. Windows (64b) / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... absent
  Python script support..... present
  Lua SWIG support.......... present
  Python SWIG support....... present

 [ PROXMARK3 ]
  firmware.................. PM3 GENERIC

 [ ARM ]
  bootrom: Iceman/master/v4.16717-33-ge43f6804a 2023-07-09 02:15:36 5e7ed8251
       os: Iceman/master/v4.16717-33-ge43f6804a 2023-07-09 02:15:36 5e7ed8251
  compiled with GCC 12.2.1 20221205

 [ FPGA ]
  fpga_pm3_lf.ncd image 2s30vq100 2023-05-30 19:43:29
  fpga_pm3_hf.ncd image 2s30vq100 2023-05-30 19:43:44
  fpga_pm3_felica.ncd image 2s30vq100 2023-05-30 19:44:12
  fpga_pm3_hf_15.ncd image 2s30vq100 2023-05-30 19:43:58

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 60% used )

Card info:
[usb] pm3 --> hf 14a info

[+]  UID: 04 47 24 AA 06 ** **
[+] ATQA: 00 42
[+]  SAK: 18 [2]
[+] MANUFACTURER: NXP Semiconductors Germany
[+] Possible types:
[+]    MIFARE Classic 4K CL2
[=] -------------------------- ATS --------------------------
[+] ATS: 0C 75 77 80 02 C1 05 2F 2F 01 BC D6 [ D3 00 ]
[=]      0C...............  TL    length is 12 bytes
[=]         75............  T0    TA1 is present, TB1 is present, TC1 is present, FSCI is 5 (FSC = 64)
[=]            77.........  TA1   different divisors are supported, DR: [2, 4, 8], DS: [2, 4, 8]
[=]               80......  TB1   SFGI = 0 (SFGT = (not needed) 0/fc), FWI = 8 (FWT = 1048576/fc)
[=]                  02...  TC1   NAD is NOT supported, CID is supported

[=] -------------------- Historical bytes --------------------
[=]     C1 05 2F 2F 01 BC D6      MIFARE Plus X 4K (SL1)
[+]     C1.....................   Mifare or (multiple) virtual cards of various type
[+]        05..................   length is 5 bytes
[+]           2x...............   MIFARE Plus
[+]              2x............   Released
[+]                 x1.........   VCS, VCSL, and SVC supported
[?] Hint: try `hf mfp info`

first two sectors has default keys and they readble without any issue:
[usb] pm3 --> hf mf rdsc -s 0 -k FFFFFFFFFFFF

[=]   # | sector 00 / 0x00                                | ascii
[=] ----+-------------------------------------------------+-----------------
[=]   0 | 04 47 24 AA 06 ** ** 18 42 00 14 01 11 00 ** ** | .G$..U..B.......
[=]   1 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
[=]   2 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
[=]   3 | 00 00 00 00 00 00 7C 37 88 00 00 00 00 00 00 00 | ......|7........

[usb] pm3 --> hf mf rdsc -s 1 -k FFFFFFFFFFFF

[=]   # | sector 01 / 0x01                                | ascii
[=] ----+-------------------------------------------------+-----------------
[=]   4 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
[=]   5 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
[=]   6 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
[=]   7 | 00 00 00 00 00 00 7C 37 88 00 00 00 00 00 00 00 | ......|7........

other sectors return Auth error

[usb] pm3 --> hf mf rdsc -s 2 -k EAAC88E5****
[#] Auth error

atemt to read block return this error too.
so i can't dump this card

BUT if i use this command 
hf mf autopwn --4k -f keys.dic  where keys.dic containt all card keys
dump successfully created with correct data

[+] Generating binary key file
[+] Found keys have been dumped to D:\Downloads\Proxmark\ProxSpace\pm3/hf-mf-044724AA06****-key.bin
[=] FYI! --> 0xFFFFFFFFFFFF <-- has been inserted for unknown keys where res is 0
[+] transferring keys to simulator memory (Cmd Error: 04 can occur)
[=] downloading the card content from emulator memory
[+] saved 1024 bytes to binary file D:\Downloads\Proxmark\ProxSpace\pm3/hf-mf-044724AA06****-dump.bin
[+] saved 64 blocks to text file D:\Downloads\Proxmark\ProxSpace\pm3/hf-mf-044724AA06****-dump.eml
[+] saved to json file D:\Downloads\Proxmark\ProxSpace\pm3/hf-mf-044724AA06****-dump.json
[=] autopwn execution time: 4 seconds

i'll try to compile software(v4.16717-33) on Windows11 and Kali linux with the same result

iceman1001, 2023-07-13:
Pull latest source and compile / flash and try again.
There was a bug which got fixed yesterday.

deimspb, 2023-07-13:
After pull install latest FW "read block" and "read sector" command was begin work properly, but dump command still return auth error:
[usb] pm3 --> hf mf rdsc -s 16 -k 72F96BDD****

[=]   # | sector 16 / 0x10                                | ascii
[=] ----+-------------------------------------------------+-----------------
[=]  64 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
[=]  65 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
[=]  66 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
[=]  67 | 00 00 00 00 00 00 7C 37 88 00 00 00 00 00 00 00 | ......|7........

[usb] pm3 --> hf mf rdsc -s 5 -k 77DABC98****

[=]   # | sector 05 / 0x05                                | ascii
[=] ----+-------------------------------------------------+-----------------
[=]  20 | 2D 8A 6C 01 C2 35 24 13 00 00 00 00 00 00 5A 06 | -.l..5$.......Z.
[=]  21 | 02 00 2D 8A 6C 00 18 21 2F 23 00 00 00 00 00 00 | ..-.l..!/#......
[=]  22 | 02 00 2D 8A 6C 00 18 21 2F 23 00 00 00 00 00 00 | ..-.l..!/#......
[=]  23 | 00 00 00 00 00 00 7F 07 88 00 00 00 00 00 00 00 | ................

[usb] pm3 --> hf mf dump --keys hf-mf-044724AA06****-key.bin
[=] Using... hf-mf-044724AA06****-key.bin
[=] Reading sector access bits...
[=] ....[#] Auth error
.[#] Auth error
.[#] Auth error
.[#] Auth error
.[#] Auth error
.[#] Auth error

[!] trying with key B instead...

[=] ..[#] Auth error
.[#] Auth error
.[#] Auth error
.[#] Auth error

[-] could not get access rights for sector  2. Trying with defaults...

.[#] Auth error
.[#] Auth error
.[#] Auth error
........

hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  Iceman/master/v4.16717-96-gfa0db3014 2023-07-14 00:05:35 cdf68ab2f
  compiled with............. MinGW-w64 10.3.0
  platform.................. Windows (64b) / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... absent
  Python script support..... absent
  Lua SWIG support.......... present
  Python SWIG support....... absent

 [ PROXMARK3 ]
  firmware.................. PM3 GENERIC

 [ ARM ]
  bootrom: Iceman/master/v4.16717-33-ge43f6804a 2023-07-09 02:15:35 5e7ed8251
       os: Iceman/master/v4.16717-33-ge43f6804a 2023-07-09 02:15:35 5e7ed8251
  compiled with GCC 12.2.1 20221205

 [ FPGA ]
  fpga_pm3_lf.ncd image 2s30vq100 2023-05-30 19:43:29
  fpga_pm3_hf.ncd image 2s30vq100 2023-05-30 19:43:44
  fpga_pm3_felica.ncd image 2s30vq100 2023-05-30 19:44:12
  fpga_pm3_hf_15.ncd image 2s30vq100 2023-05-30 19:43:58

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 60% used )

[!] ARM firmware does not match the source at the time the client was compiled
[!] Make sure to flash a correct and up-to-date version

deimspb, 2023-07-14:
you're right, my mistake. After updating your need to dump all keys again. Without updating keys - don't work properly

closed 2023-07-14 by deimspb

iceman1001, 2023-07-14:
Good to hear it was resolved. 
And I pushed another fix just now, so do pull again.