RfidResearchGroup/proxmark3 #2046: ELECTRA - LUA

jareckib, 2023-07-20:
What is the problem with my LUA-ELECTRA. I report problems on a regular basis. Block 0 change and adding two blocks 3 and 4. I'm starting in LUA. Why is my script not added....spent some time....maybe it's worth appreciating?

jareckib, 2023-07-20:
A simple modification that changes block 0 and adds two blocks. You are not interested? I see in libraries----last error.....also added "*mfp" didn't work. What is your problem?

jareckib, 2023-07-20:
A simple modification to change block 0 and add two blocks. You are not interested? I see in libraries----last compile time error.....also added "*mfp" didn't work. I also read what a mistake it is. What is your problem? I can help

jareckib, 2023-07-20:
Give me two weeks. I'm on vacation, and since I'm starting to work in coding for Proxmark3, I'll give you the result files. I will make a modification for the ELECTRA fob myself.

jareckib, 2023-07-20:
Why is my script electa.lua not added to the build?

iceman1001, 2023-07-20:
....  if you want code to be merged,  you need to make a pull request.  
this here is the issues

jareckib, 2023-07-20:
I let go of the subject .... I have LUA for Proxmark3 for the new ELECTRA tag and TMD-5s device. There's no point in wasting  own precious time if you can't appreciate it. Have a nice day.

jareckib, 2023-07-20:
Never mind. I will put it on youtube on my channel. Anyone who is interested download it.

iceman1001, 2023-07-20:
You seem to fail to understand how github works

closed 2023-07-20 by iceman1001