RfidResearchGroup/proxmark3 #2047: Partial key view on hf mf super when key start with 0

TrixyCracker, 2023-07-20:
when key starts with zeroes, mfkey returns a partial key.
`
[usb] pm3 --> hf mf super
[+] UID: ED6F8104 Sector 00 key A [            0 ]  LOOK HERE
[usb] pm3 --> hf mf rdbl --blk 0 -k 000000000000
[#] Auth error

[usb] pm3 --> hf mf super
[+] UID: ED6F8104 Sector 00 key A [            0 ]  LOOK HERE
[usb] pm3 --> hf mf rdbl --blk 0 -k 000000000000
[#] Auth error

[usb] pm3 --> hf mf super
[+] UID: ED6F8104 Sector 00 key A [            0 ]  LOOK HERE
[usb] pm3 --> hf mf rdbl --blk 0 -k 100000000000
[#] Auth error

[usb] pm3 --> hf mf rdbl --blk 0 -k 100000000000
[#] Auth error

[usb] pm3 --> hf mf super
[+] UID: ED6F8104 Sector 00 key A [ 100000000000 ]
[usb] pm3 --> hf mf rdbl --blk 0 -k 000000000001
[#] Auth error

[usb] pm3 --> hf mf rdbl --blk 0 -k 000000000001
[#] Auth error

[usb] pm3 --> hf mf super
[+] UID: ED6F8104 Sector 00 key A [            1 ]  LOOK HERE
[usb] pm3 --> hf mf super
[+] UID: ED6F8104 Sector 00 key A [            1 ]  LOOK HERE
[usb] pm3 --> hf mf rdbl --blk 0 -k 000000000011
[#] Auth error

[usb] pm3 --> hf mf super
[-] ⛔ failed to recover any key
[usb] pm3 --> hf mf super
^[[A[-] ⛔ failed to recover any key
[usb] pm3 --> hf mf rdbl --blk 0 -k 000000000011
[#] Auth error

[usb] pm3 --> hf mf rdbl --blk 0 -k 000000000011
[#] Auth error

[usb] pm3 --> hf mf super
[+] UID: ED6F8104 Sector 00 key A [           11 ]  LOOK HERE
`

iceman1001, 2023-07-20:
I don't have a super card,  so first of all, which one do you have?
 
And it is hard to duplicate the issue with your samples.  
How about you compliment with proper details?

TrixyCracker, 2023-07-20:
> I don't have a super card, so first of all, which one do you have?
> 
> And it is hard to duplicate the issue with your samples. How about you compliment with proper details?

[image]
it's easier to understand with a picture, as u can see, extension with zeroes not working, in fact instead of 000011223344 i can see only 11223344.

iceman1001, 2023-07-20:
interesting,   all three possible places which prints that kind of line uses zero padding string formatters  (%012)
                                       
```
   PrintAndLogEx(SUCCESS, "UID: %s Sector %02x key %c [ " _GREEN_("%012" PRIX64) " ]",
```

TrixyCracker, 2023-07-20:
> interesting, all three possible places which prints that kind of line uses zero padding string formatters (%012)
> 
> ```
>    PrintAndLogEx(SUCCESS, "UID: %s Sector %02x key %c [ " _GREEN_("%012" PRIX64) " ]",
> ```

im using previous version, my bad, in new version code seems correct

closed 2023-07-21 by iceman1001