RfidResearchGroup/proxmark3 #2055: Compile Error on raspberry pi

lsellens, 2023-07-26:
There appears to be a potential buffer overflow in snprintf in function newfilenamemcopyEx of fileutils.c
I get
directive output may be truncated writing between 1 and 2147483645 bytes into a region of size 1000
when compiling for raspberry pi but works fine with x86_64
it's line 209 of fileutils.c that throws the error
It showed up after 588dea40055a537188dc12b37fb623f62ba3a02d

iceman1001, 2023-07-26:
duplicate https://github.com/RfidResearchGroup/proxmark3/issues/2027

However,  I can't replicate it on my RPi 4  with CC 11.3.0,  on latest source v4.16717-233

I pushed some checks and I need you to test.  If you can find the root cause then it would be helpful

iceman1001, 2023-07-28:
pull latest and see if it still gives this warning

lsellens, 2023-07-31:
That seems to have fixed it for me. Thanks!

closed 2023-07-31 by lsellens

iceman1001, 2023-08-02:
Excellent!