RfidResearchGroup/proxmark3 #2056: LEGIC Sim broken?

0none, 2023-07-27:
**Describe the bug**

Somewhere between the latest release ("seven") and master, tie ability to simulate a legic prime 1024 was lost.
Recompiling and repeating on current master (1a4df13aaec982f66777b06e1231807eaba54979) the B and C lights flicker on, but the reader does not react.

**To Reproduce**

Steps to reproduce the behavior:
- clone current master 
- flash firmware
- dump MIM1024 card
- eload dump to memory
- simulate
- try to authenticate against reader using simulated card

**Expected behavior**

Loading the same dump file to memory, at "seven" the pm3 (generic) successfully transmits to the reader, in this case a door control system, authenticating the user and opening the door.

**Desktop (please complete the following information):**

 OS: Ubuntu on WSL2 / Windows 11

<details><summary>hw version</summary>

    ```
    [ Proxmark3 RFID instrument ]

    [ CLIENT ]
    Iceman/master/v4.16717-236-g7f5590d94 2023-07-27 12:06:44 f67bd6537
    compiled with............. GCC 12.2.0
    platform.................. Linux / x86_64
    Readline support.......... present
    QT GUI support............ present
    native BT support......... present
    Python script support..... present
    Lua SWIG support.......... present
    Python SWIG support....... present

    [ PROXMARK3 ]
    firmware.................. PM3 GENERIC

    [ ARM ]
    bootrom: Iceman/master/v4.16717-236-g7f5590d94 2023-07-27 12:06:30 f67bd6537
        os: Iceman/master/v4.16717-236-g7f5590d94 2023-07-27 12:06:58 f67bd6537
    compiled with GCC 12.2.1 20221205

    [ FPGA ]
    fpga_pm3_lf.ncd image 2s30vq100 2023-07-12 16:12:04
    fpga_pm3_hf.ncd image 2s30vq100 2023-07-12 16:12:14
    fpga_pm3_felica.ncd image 2s30vq100 2023-07-12 16:12:34
    fpga_pm3_hf_15.ncd image 2s30vq100 2023-07-12 16:12:24

    [ Hardware ]
    --= uC: AT91SAM7S512 Rev B
    --= Embedded Processor: ARM7TDMI
    --= Internal SRAM size: 64K bytes
    --= Architecture identifier: AT91SAM7Sxx Series
    --= Embedded flash memory 512K bytes ( 60% used )
    ```
</details>
<details><summary>hw status</summary>
    ```
    [#] Memory
    [#]   BigBuf_size............. 42400
    [#]   Available memory........ 42400
    [#] Tracing
    [#]   tracing ................ 1
    [#]   traceLen ............... 0
    [#] Current FPGA image
[... 138 lines cut ...]
        6594 |       6720 | Tag |06  1d  00                                                               |     |
        6773 |       6891 | Rdr |06  39  00  00                                                           |     | ACK 256/1024
        6949 |       7125 | Rdr |09  01  00  00                                                           |     | READ Byte(0)
        7195 |       7447 | Tag |0c  57  06                                                               |     |
        7503 |       7679 | Rdr |09  5d  01  00                                                           |     | READ Byte(174)
        8057 |       8201 | Rdr |09  42  00  00                                                           |     | WRITE Byte(33)
        12722 |      12854 | Rdr |07  47  00  00                                                           |     | IV 0x47
        12924 |      13050 | Tag |06  1d  00                                                               |     |
        13103 |      13205 | Rdr |06  39  00  00                                                           |     | ACK 256/1024
        13263 |      13439 | Rdr |09  01  00  00                                                           |     | READ Byte(0)
        13509 |      13761 | Tag |0c  57  06                                                               |     |
        13817 |      13969 | Rdr |09  19  00  00                                                           |     | READ Byte(12)
        14347 |      14531 | Rdr |09  af  01  00                                                           |     | READ Byte(215)
        19100 |      19224 | Rdr |07  0b  00  00                                                           |     | IV 0x0B
        19294 |      19420 | Tag |06  1d  00                                                               |     |
        19473 |      19583 | Rdr |06  39  00  00                                                           |     | ACK 256/1024
        19641 |      19809 | Rdr |09  01  00  00                                                           |     | READ Byte(0)
        19879 |      20131 | Tag |0c  57  06                                                               |     |
        20187 |      20371 | Rdr |09  df  00  00                                                           |     | READ Byte(111)
        20749 |      20917 | Rdr |09  93  01  00                                                           |     | READ Byte(201)
        25432 |      25564 | Rdr |07  27  00  00                                                           |     | IV 0x27
        25634 |      25760 | Tag |06  1d  00                                                               |     |

    ```
</details>

iceman1001, 2023-07-27:
hmm..
I see,   pull latest and see if it works better again?

0none, 2023-07-27:
Unfortunately the situation seems unchanged

<details>
  <summary>log attached</summary>

    [=] downloading tracelog data from device
    [+] Recorded activity (trace len = 1145 bytes)
    [=] start = start of start frame end = end of frame. src = source of transfer
    [=] LEGIC - Reader Mode: Timings are in ticks (1us == 1.5ticks)
    [=]         Tag Mode: Timings are in sub carrier periods (1/212 kHz == 4.7us)

        Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
    ------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
            0 |        116 | Rdr |07  03  00  00                                                           |     | IV 0x03
            186 |        312 | Tag |06  1d  00                                                               |     |
            533 |        709 | Rdr |09  e7  00  00                                                           |     | READ Byte(115)
        1087 |       1279 | Rdr |09  fe  01  00                                                           |     | WRITE Byte(255)
        6371 |       6487 | Rdr |07  05  00  00                                                           |     | IV 0x05
        6557 |       6683 | Tag |06  1d  00                                                               |     |
        6736 |       6854 | Rdr |06  2b  00  00                                                           |     |
        6912 |       7048 | Rdr |09  08  00  00                                                           |     | WRITE Byte(4)
        7426 |       7578 | Rdr |09  30  01  00                                                           |     | WRITE Byte(152)
        12721 |      12861 | Rdr |07  5d  00  00                                                           |     | IV 0x5D
        12931 |      13057 | Tag |06  1d  00                                                               |     |
        13110 |      13212 | Rdr |06  39  00  00                                                           |     | ACK 256/1024
        13270 |      13454 | Rdr |09  01  00  00                                                           |     | READ Byte(0)
        13524 |      13776 | Tag |0c  57  06                                                               |     |
        13832 |      14000 | Rdr |09  ec  00  00                                                           |     | WRITE Byte(118)
        14378 |      14538 | Rdr |09  64  01  00                                                           |     | WRITE Byte(178)
        19056 |      19164 | Rdr |07  01  00  00                                                           |     | IV 0x01
        19234 |      19360 | Tag |06  1d  00                                                               |     |
        19413 |      19507 | Rdr |06  39  00  00                                                           |     | ACK 256/1024
        19565 |      19725 | Rdr |09  01  00  00                                                           |     | READ Byte(0)
        19795 |      20047 | Tag |0c  57  06                                                               |     |
        20103 |      20271 | Rdr |09  a3  01  00                                                           |     | READ Byte(209)
        20649 |      20793 | Rdr |09  42  00  00                                                           |     | WRITE Byte(33)
        25394 |      25526 | Rdr |07  55  00  00                                                           |     | IV 0x55
        25596 |      25722 | Tag |06  1d  00                                                               |     |
        25775 |      25885 | Rdr |06  39  00  00                                                           |     | ACK 256/1024
        25943 |      26111 | Rdr |09  01  00  00                                                           |     | READ Byte(0)
        26181 |      26433 | Tag |0c  57  06                                                               |     |
        26489 |      26665 | Rdr |09  7c  01  00                                                           |     | WRITE Byte(190)
        27043 |      27219 | Rdr |09  fc  00  00                                                           |     | WRITE Byte(126)
        31738 |      31870 | Rdr |07  0f  00  00                                                           |     | IV 0x0F
        31940 |      32066 | Tag |06  1d  00                                                               |     |
        32119 |      32229 | Rdr |06  39  00  00                                                           |     | ACK 256/1024
        32287 |      32455 | Rdr |09  01  00  00                                                           |     | READ Byte(0)
        32525 |      32777 | Tag |0c  57  06                                                               |     |
        32833 |      33001 | Rdr |09  37  00  00                                                           |     | READ Byte(27)
        33379 |      33547 | Rdr |09  5a  01  00                                                           |     | WRITE Byte(173)
        38083 |      38231 | Rdr |07  7d  00  00                                                           |     | IV 0x7D
[... cut ...]
        83031 |      83191 | Rdr |09  01  00  00                                                           |     | READ Byte(0)
        83261 |      83513 | Tag |0c  57  06                                                               |     |
        83569 |      83721 | Rdr |09  0e  00  00                                                           |     | WRITE Byte(7)
        84099 |      84275 | Rdr |09  d5  01  00                                                           |     | READ Byte(234)
    [usb] pm3 -->

</details>

0none, 2023-07-27:
Reverting the changes made to armsrc/legicrfsim.c in commit c70e5beeac7229605690230985e029031ae37a59 seems to fix the issue for me, tough I am not exactly sure what exactly causes it. 
Sorry if this is vague, I'm not a programmer.

iceman1001, 2023-07-27:
hm..
pull latest and test again

0none, 2023-08-01:
Sorry for the delay.
Doesn't seem to have worked.

<details>
  <summary>log attached</summary>

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation                                                                                      ------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------                                                                                       0 |        132 | Rdr |07  47  00  00                                                           |     | IV 0x47                                                                                                 202 |        328 | Tag |06  1d  00                                                               |     |                                                                                                         381 |        483 | Rdr |06  39  00  00                                                           |     | ACK 256/1024                                                                                            541 |        717 | Rdr |09  01  00  00                                                           |     | READ Byte(0)                                                                                            787 |       1039 | Tag |0c  57  06                                                               |     |                                                                                                        1095 |       1247 | Rdr |09  19  00  00                                                           |     | READ Byte(12)
       1625 |       1809 | Rdr |09  af  01  00                                                           |     | READ Byte(215)
       6382 |       6522 | Rdr |07  75  00  00                                                           |     | IV 0x75
       6592 |       6718 | Tag |06  1d  00                                                               |     |
       6771 |       6897 | Rdr |06  39  00  00                                                           |     | ACK 256/1024
       6955 |       7115 | Rdr |09  01  00  00                                                           |     | READ Byte(0)
       7185 |       7437 | Tag |0c  57  06                                                               |     |
       7493 |       7645 | Rdr |09  88  01  00                                                           |     | WRITE Byte(196)
       8023 |       8159 | Rdr |09  40  00  00                                                           |     | WRITE Byte(32)
      12724 |      12856 | Rdr |07  33  00  00                                                           |     | IV 0x33
      12926 |      13052 | Tag |06  1d  00                                                               |     |
      13105 |      13199 | Rdr |06  39  00  00                                                           |     | ACK 256/1024
      13257 |      13417 | Rdr |09  01  00  00                                                           |     | READ Byte(0)
      13487 |      13739 | Tag |0c  57  06                                                               |     |
      13795 |      13963 | Rdr |09  76  00  00                                                           |     | WRITE Byte(59)
      14341 |      14525 | Rdr |09  bb  01  00                                                           |     | READ Byte(221)
      19069 |      19209 | Rdr |07  1f  00  00                                                           |     | IV 0x1F
      19279 |      19405 | Tag |06  1d  00                                                               |     |
      19458 |      19568 | Rdr |06  39  00  00                                                           |     | ACK 256/1024
      19626 |      19794 | Rdr |09  01  00  00                                                           |     | READ Byte(0)
      19864 |      20116 | Tag |0c  57  06                                                               |     |
      20186 |      20324 | Rdr |08  43  00  00                                                           |     |
      20702 |      20886 | Rdr |09  dd  01  00                                                           |     | READ Byte(238)
      25440 |      25580 | Rdr |07  6b  00  00                                                           |     | IV 0x6B
      25650 |      25776 | Tag |06  1d  00                                                               |     |
      25829 |      25947 | Rdr |06  39  00  00                                                           |     | ACK 256/1024
      26005 |      26157 | Rdr |09  01  00  00                                                           |     | READ Byte(0)
      26227 |      26479 | Tag |0c  57  06                                                               |     |
      26535 |      26687 | Rdr |09  23  00  00                                                           |     | READ Byte(17)
      27065 |      27233 | Rdr |09  e4  01  00                                                           |     | WRITE Byte(242)
      31768 |      31884 | Rdr |07  03  00  00                                                           |     | IV 0x03
      31954 |      32080 | Tag |06  1d  00                                                               |     |
[... cut ...]
      32855 |      33039 | Rdr |09  fc  01  00                                                           |     | WRITE Byte(254)
      33417 |      33569 | Rdr |09  41  01  00                                                           |     | READ Byte(160)
      38108 |      38240 | Rdr |07  33  00  00                                                           |     | IV 0x33
      38310 |      38436 | Tag |06  1d  00                                                               |     |

</details>

iceman1001, 2023-09-07:
Pull latest and it should work just fine again

closed 2023-09-07 by iceman1001