RfidResearchGroup/proxmark3 #2075: Proxmark3 client cannot communicate with the device over Bluetooth.

anderszon61, 2023-08-13:
**Description**

When attempting to connect the Proxmark3 via Bluetooth on Kali Linux, I encountered an issue. 
Despite the Proxmark3 device being visible in the list of Bluetooth devices and seemingly connected, 
the Proxmark3 client cannot communicate with the device over Bluetooth.

**Steps to Reproduce**
Pair Proxmark3 device with Kali Linux over Bluetooth.
Attempt to list the device using the command: pm3 --list. This correctly lists the device with its Bluetooth address.
Try to communicate with the device using the command: pm3 -p bt:20:19:08:01:43:31.
The error ERROR: cannot communicate with the Proxmark is displayed.

**Expected Behavior**
The Proxmark3 client should be able to communicate with the Proxmark3 device over Bluetooth without any issues.

**Actual Behavior**
The Proxmark3 client displays an error stating it cannot communicate with the Proxmark3 device over Bluetooth.

**Additional Information**
Please note that the issue was not observed with using rfcomm0 connection (Pm3 not working with rfcom). It was tested to ensure the problem is not unique to my specific computer setup.

**Environment**
Operating System: Kali GNU/Linux Rolling 2023.3

Proxmark3 Details:

[ Proxmark3 RFID instrument ]

MCU....... AT91SAM7S512 Rev B
Memory.... 512 KB ( 67% used )
Client.... Iceman/master/v4.16717-304-g81a073915-dirty 2023-08-11 19:16:20
Bootrom... Iceman/master/v4.16717-304-g81a073915-dirty-suspect 2023-08-11 19:17:08 
OS........ Iceman/master/v4.16717-304-g81a073915-dirty-suspect 2023-08-11 19:17:34 
Target.... RDV4

 [ Additional Hardware Details ]
 External Flash........... present
 Smartcard Reader......... present
 FPGA Images:
- fpga_pm3_lf.ncd       2s30vq100 2023-07-12 16:12:04
- fpga_pm3_hf.ncd       2s30vq100 2023-07-12 16:12:14
- fpga_pm3_felica.ncd   2s30vq100 2023-07-12 16:12:34
- fpga_pm3_hf_15.ncd    2s30vq100 2023-07-12 16:12:24

iceman1001, 2023-08-14:
```
hw status
```

anderszon61, 2023-08-15:
I found the error. In Makefile.platform, there was an incorrect parameter: PLATFORM_EXTRAS=FLASH enabled by mistake. When I removed it and updated Proxmark3, the Bluetooth worked.
So the problem is resolved.

closed 2023-08-15 by iceman1001