RfidResearchGroup/proxmark3 #2083: CryptoRF tools doesn't work!

makhco, 2023-08-17:
I tried "SMA" and "SMA_multi" code several time with various valid (Ci,Q,Ch and Ci+1). even I tried it with "simulate" command example data but it could not find key never! please see below log (4 arguments taked from simulate command) for more information also:

sma_multi 36e54f5dab68eeb0 db4bc988775bfefe 309abcfe1cba373f dd412cc86feabe9f
←[36m
Authentication info

←[0m  Gc: unknown
  Ci: 36 e5 4f 5d ab 68 ee b0
   Q: db 4b c9 88 77 5b fe fe
  Ch: 30 9a bc fe 1c ba 37 3f
Ci+1: dd 41 2c c8 6f ea be 9f

  Ks: dd 30 41 9a 2c bc c8 fe 6f 1c ea ba be 37 9f 3f

Multithreaded, will use ←[33m16←[0m threads
Initializing lookup tables for increasing cipher speed
Determing the right states that correspond to the keystream
................................
Top-bin for the right state contains ←[32m100←[0m correct bits
Total count of right bins: ←[33m79←[0m
Using the state from the top-right bin: ←[33m0x0eb0693←[0m
Found ←[33m31776←[0m right candidates using the meet-in-the-middle attack
Calculating left states using the (unknown bits) mask from the top-right state
0.0%...........12.5%................25.0%...................37.5%...............................50.0%.....................62.5%.....................75.0%...........87.5%..........100%
Found a total of ←[33m132←[0m left cipher states, recovering left candidates...
....................................................................................................................................
The meet-in-the-middle attack returned ←[33m1986←[0m left cipher candidates
Combining left and right states, disposing invalid combinations
Outer  ←[33m31776←[0m , inner ←[33m1986←[0m
Found a total of ←[33m63107136←[0m combinations, but only ←[32m883←[0m were valid!
Filtering the correct one using the middle part
←[31m
Could not find key using this right cipher state.

←[0mUsing the state from the top-right bin: ←[33m0x1596b77←[0m
Found ←[33m31776←[0m right candidates using the meet-in-the-middle attack
Calculating left states using the (unknown bits) mask from the top-right state
0.0%.........12.5%.......25.0%..37.5%.....50.0%....62.5%...75.0%........87.5%........100%
Found a total of ←[33m38←[0m left cipher states, recovering left candidates...
......................................
The meet-in-the-middle attack returned ←[33m993←[0m left cipher candidates
Combining left and right states, disposing invalid combinations
Outer  ←[33m31776←[0m , inner ←[33m993←[0m
Found a total of ←[33m31553568←[0m combinations, but only ←[32m276←[0m were valid!
Filtering the correct one using the middle part
←[31m
Could not find key using this right cipher state.
←[0mUsing the state from the top-right bin: ←[33m0x114f96c←[0m
Found ←[33m31776←[0m right candidates using the meet-in-the-middle attack
Calculating left states using the (unknown bits) mask from the top-right state
0.0%.12.5%.25.0%..37.5%..50.0%.62.5%...75.0%.87.5%..100%
Found a total of ←[33m5←[0m left cipher states, recovering left candidates...
.....
The meet-in-the-middle attack returned ←[33m0←[0m left cipher candidates
Using the state from the top-right bin: ←[33m0x00db152←[0m
Found ←[33m0←[0m right candidates using the meet-in-the-middle attack
Using the state from the top-right bin: ←[33m0x00b0020←[0m
Found ←[33m0←[0m right candidates using the meet-in-the-middle attack
Using the state from the top-right bin: ←[33m0x004ffdf←[0m
Found ←[33m0←[0m right candidates using the meet-in-the-middle attack
Using the state from the top-right bin: ←[33m0x004ffc0←[0m
Found ←[33m0←[0m right candidates using the meet-in-the-middle attack
Using the state from the top-right bin: ←[33m0x00483df←[0m
Found ←[33m0←[0m right candidates using the meet-in-the-middle attack

I want to emphasize that I tried more than 20 examples (traces) but this code never returns an suitable answer.
Has anyone experimentally run this code and recovered the key?؟?

makhco, 2023-08-18:
I ran SMA.cpp/SMA_multi.cpp codes in windows 10. Do I need proxmark3 while I have traces (Ci,Q,CH,Ci+1)?

iceman1001, 2023-08-21:
This cryptoRF solver is seperate software. Runs w/o any device.

We don't have regression testing for this code,   it ran before on my WSL,    added a change and to the inital state object used and it works for me. 

Next time feel free to contribute with fixes.

closed 2023-08-22 by iceman1001