RfidResearchGroup/proxmark3 #2097: Can't detect T5577 tag

Ultimator14, 2023-08-28:
**Describe the bug**

I recently bought a bunch of [T5577 tags](https://www.amazon.de/dp/B00N5HD3R2).  

I was trying to read the tags but they were not detected.

```
[usb] pm3 --> lf t55xx detect
[!] ⚠️  Could not detect modulation automatically. Try setting it manually with 'lf t55xx config'
[usb] pm3 --> lf t55xx wipe
[=] Target T55x7 tag
[=] Default configuration block 000880E0

[=] Begin wiping...
[=] Writing page 0  block: 00  data: 0x000880E0 
[=] Writing page 0  block: 01  data: 0x00000000 
[=] Writing page 0  block: 02  data: 0x00000000 
[=] Writing page 0  block: 03  data: 0x00000000 
[=] Writing page 0  block: 04  data: 0x00000000 
[=] Writing page 0  block: 05  data: 0x00000000 
[=] Writing page 0  block: 06  data: 0x00000000 
[=] Writing page 0  block: 07  data: 0x00000000 
[usb] pm3 --> [usb] pm3 --> lf t55xx detect
[!] ⚠️  Could not detect modulation automatically. Try setting it manually with 'lf t55xx config'
[usb] pm3 --> lf t55xx info
[usb] pm3 --> lf t55xx config
[=] --- current t55xx config --------------------------
[=]  Chip type......... T55x7
[=]  Modulation........ ASK
[=]  Bit rate.......... 0 - RF/8
[=]  Inverted.......... No
[=]  Offset............ 0
[=]  Seq. terminator... No
[=]  Block0............ 00000000 (n/a)
[=]  Downlink mode..... default/fixed bit length
[=]  Password set...... No
```

I cloned a em410x tag on the chip using flipper zero (ID 0x123456789). Afterwards the `t55xx` was still not detected but I was able to read the tag using `lf em 410x reader` and `lf search`.

```
[usb] pm3 --> lf t55xx detect
[!] ⚠️  Could not detect modulation automatically. Try setting it manually with 'lf t55xx config'
[usb] pm3 --> lf em 410x reader
[+] EM 410x ID 0123456789
[usb] pm3 --> lf search

[=] NOTE: some demods output possible binary
[=] if it finds something that looks like a tag
[=] False Positives ARE possible
[=] 
[=] Checking for known tags...
[=] 
[!] ⚠️  Specify one authentication mode
[+] EM 410x ID 0123456789
[+] EM410x ( RF/64 )
[=] -------- Possible de-scramble patterns ---------
[+] Unique TAG ID      : 80C4A2E691
[=] HoneyWell IdentKey
[+]     DEZ 8          : 04548489
[+]     DEZ 10         : 0591751049
[+]     DEZ 5.5        : 09029.26505
[+]     DEZ 3.5A       : 001.26505
[+]     DEZ 3.5B       : 035.26505
[+]     DEZ 3.5C       : 069.26505
[+]     DEZ 14/IK2     : 00004886718345
[+]     DEZ 15/IK3     : 000553054824081
[+]     DEZ 20/ZK      : 08001204100214060901
[=] 
[+] Other              : 26505_069_04548489
[... 261 lines cut ...]
ASK, 64: 59c1cde8	
ASK, 100: 0	
ASK, 128: 0	
NRZ, 8: 1fe01fe	
NRZ, 16: f0f30cf	
NRZ, 32: 334b52ab	
NRZ, 40: 55555555	
NRZ, 50: 55555555	
NRZ, 64: 55555555	
BI, 50: aded4e3	
BI, 64: aded4e3	
BI, 100: ffffffff	
BI, 128: ffffffff	
BIA, 50: f5212b1c	
BIA, 64: f5212b1c	
BIA, 100: 0	
BIA, 128: 0
```

Similar (non-matching) resuilts when using `--r2,3,4` or `--st`.
Some readings were not consistent when executing the command twice.

</details>

I think the writing didn't do anything because I was still able to read the em tag id using  `lf em 410x reader` afterwards.

iceman1001, 2023-08-28:
I don't believe there is anything wrong with the source code.  

I suggest you head over to the discord server and ask there.    In the end there is no guarantee you got real t5577 cards,  it can be simple EM4100 or others.

iceman1001, 2023-08-28:
Since you on a RDV4,   make sure the Q switch on your antenna is  set to 7  and  freq to 125.

Ultimator14, 2023-08-28:
> I don't believe there is anything wrong with the source code.
> 
> I suggest you head over to the discord server and ask there. In the end there is no guarantee you got real t5577 cards, it can be simple EM4100 or others.

> 

Thanks, I'll do that.

> Since you on a RDV4, make sure the Q switch on your antenna is set to 7 and freq to 125.

I'm on RDV4.0 with the old antenna that doesn't have the switches.

iceman1001, 2023-08-28:
Then you will have some issues writing to your t5577 tag.

iceman1001, 2023-08-29:
since I am sure this is not an issue with the source code I will close this issue. 

OP will take the discussion in the discord server.

closed 2023-08-29 by iceman1001

Ultimator14, 2023-09-04:
I just wanted to report back here in case someone else has the same problem.
It was indeed the antenna. I bought the [RDV4.01 replacement antenna](https://lab401.com/de/products/proxmark-3-rdv4-01-upgraded-replacement-antenna) and flipped the switches accordingly.

`hw tune` now reports
```
[+] Approx. Q factor (*): 5,3 by frequency bandwidth measurement
[+] Approx. Q factor (*): 6,1 by peak voltage measurement
```
Previously that was
```
[+] Approx. Q factor (*): 11,5 by frequency bandwidth measurement
[+] Approx. Q factor (*): 11,9 by peak voltage measurement
```

The detect command and also all other commands now work flawlessly.

iceman1001, 2023-09-04:
Glad you found a solution!