RfidResearchGroup/proxmark3 #2100: [idea] Autodetect current hardware firmware type, and autoselect the firmware to flash or warn the user

nekohayo, 2023-09-01:
**Is your feature request related to a problem? Please describe.**
This is related to https://bugzilla.redhat.com/show_bug.cgi?id=2236868

In my case, using a "PM3GENERIC" device, the distro-provided package only included "RDV4" firmware by default, as this is the default upstream setting. However, in some situations, this can lead users to flash the wrong firmware, which would only be noticeable by the user then running `pm3` and encountering the "device / fw mismatch" message. That message is a hint, but it would be better if the software could prevent it from happening in the first place.

**Describe the solution you'd like**

Ideally, the various flashing utilities like `pm3-flash-all`, `pm3-flash-fullimage`, etc., should look at the existing firmware on the device and try to either pick the correct matching firmware update to flash, or warn the user about it (if it can't auto-find it or the firmware is unavailable for some reason).

**Additional context**

This situation occurred because the device was originally flashed from a distro's package  that provides both firmware types (Arch's), and then an update was attempted on another distro's package (Fedora's) that only provided the default "RDV4" firmware. It took me a lot of searching around and troubleshooting to figure out the cause, and how to correct it.

iceman1001, 2023-11-17:
Not all distro packages is doing the same thing and distro maintainers is very busy people.   
Just to get a distro package is hard.   

If you have improvements,  I suggest looking into the different packages available and see which one does handle all cases the best and suggest some PR to the package (Arch?)  to be improved in order to help the distro maintainers out.

This repo doesn't control nor have the people needed for such a task.

nekohayo, 2024-02-10:
> If you have improvements, I suggest looking into the different packages available and see which one does handle all cases the best

There is only one package in Fedora, and to quote its maintainer:

>  there are no modifications to the upstream defaults for compiling, which is why the warning is thrown. RDV4 firmware is default as it's the latest/current hardware release.  
> I will look at adding support for both PM3GENERIC and RDV4 in the same package, as it would be easier for users. 

...however, they did suggest I contact the upstream project (i.e.: here) to make the suggestion that the software should detect the hardware revision prior to flashing, and avoid flashing the wrong firmware in any case.

iceman1001, 2024-02-10:
> This repo doesn't control nor have the people needed for such a task.

closed 2025-06-08 by iceman1001