RfidResearchGroup/proxmark3 #2101: hf 15 dump pnly contains 1/2 of data

ThorstenEngel, 2023-09-03:
**Describe the bug**
My "hf 15 dump" only contains the first 4 bytes of each block. In the trace I can see the full 8 bytes of the block.

**To Reproduce**
Start pm3 and have an ISO15693 tag ready to scan. I did this on windows with the latest master commit of proxmark 3 (commit 32f892e512d97be402c0abaf6f00675686ed0893).

```
[usb] pm3 --> hf 15 dump

[+]  UID: E0 07 A4 00 0B 3F 72 65
[+] TYPE: Texas Instrument France
[+] Using UID... E0 07 A4 00 0B 3F 72 65
[+] Reading memory from tag UID E0 07 A4 00 0B 3F 72 65
[-]blk  43
[-] Tag returned Error 16: The specified block is not available (doesn't exist).

[=] block#   | data         |lck| ascii
[=] ---------+--------------+---+----------
[=]   0/0x00 | CB 18 09 D9  | 0 | ....
[=]   1/0x01 | 5E EB 3E 17  | 0 | ^.>.
[=]   2/0x02 | B4 A3 2F 4B  | 0 | ../K
```
Unfortunately this is not only a display problem, writing the dump to a file will as well only contain the first 4 bytes per block.

The trace shows all data:

```
trace list -t 15
[...]
 4254542528 | 4254564544 | Rdr |26  01  00  f6  0a                                                       |  ok | INVENTORY
 4254568832 | 4254622080 | Tag |00  00  65  72  3f  0b  00  a4  07  e0  ea  7a                           |  ok |
 4254542848 | 4254597632 | Rdr |22  20  65  72  3f  0b  00  a4  07  e0  00  4b  5d                       |  ok | READBLOCK(0)
 4254601920 | 4254651072 | Tag |00  cb  18  09  d9  90  78  bc  31  8c  43                               |  ok |
 4254542848 | 4254597632 | Rdr |22  20  65  72  3f  0b  00  a4  07  e0  01  c2  4c                       |  ok | READBLOCK(1)
 4254601920 | 4254651072 | Tag |00  5e  eb  3e  17  e2  1e  91  61  ef  fb                               |  ok |
 4254542848 | 4254597632 | Rdr |22  20  65  72  3f  0b  00  a4  07  e0  02  59  7e                       |  ok | READBLOCK(2)
 4254601920 | 4254651072 | Tag |00  b4  a3  2f  4b  60  95  12  04  a1  3b                               |  ok |
 4254542848 | 4254597632 | Rdr |22  20  65  72  3f  0b  00  a4  07  e0  03  d0  6f                       |  ok | READBLOCK(3)
 4254601920 | 4254651072 | Tag |00  b8  95  97  9f  ee  cc  1f  9e  33  a8                               |  ok |
```

**Expected behavior**
Having all data in the dump.

**Desktop (please complete the following information):**
 - OS: Windows 11
 - inside proxmark3 client run the following commands and paste the output here.

```
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ Client ]
  Iceman/master/v4.16717-402-g32f892e51-suspect 2023-09-01 10:29:21 3c525426b
  compiled with............. MinGW-w64 10.3.0
  platform.................. Windows (64b) / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... absent
  Python script support..... present
  Lua SWIG support.......... present
  Python SWIG support....... present

 [ Proxmark3 ]
  device.................... RDV4
  firmware.................. RDV4
  external flash............ present
  smartcard reader.......... present
  FPC USART for BT add-on... absent

 [ ARM ]
  bootrom: Iceman/master/v4.16717-402-g32f892e51-suspect 2023-09-01 10:27:06 3c525426b
       os: Iceman/master/v4.16717-402-g32f892e51-suspect 2023-09-01 10:28:08 3c525426b
  compiled with GCC 10.1.0

 [ FPGA ]
  fpga_pm3_lf.ncd image 2s30vq100 2023-08-29 16:44:07
  fpga_pm3_hf.ncd image 2s30vq100 2023-08-29 16:44:19
  fpga_pm3_felica.ncd image 2s30vq100 2023-08-29 16:44:43
  fpga_pm3_hf_15.ncd image 2s30vq100 2023-08-29 16:44:31

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 67% used )

[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 40620
[#]   Available memory........ 40620
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 6398
[#] Current FPGA image
[#]   mode.................... fpga_pm3_hf_15.ncd image 2s30vq100 2023-08-29 16:44:31
[#] Flash memory
[#]   Baudrate................ 24 MHz
[#]   Init.................... OK
[#]   Memory size............. 2 mbits / 256 kb
[#]   Unique ID (be).......... 0x25AC8C97307C69D5
[#] Smart card module (ISO 7816)
[#]   version................. v3.10
[#]     Outdated firmware. Please upgrade to v4.x or above.
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0
[#]
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A |
[#]     long leading reference |N/A | N/A | N/A | N/A | N/A | N/A | N/A |
[#]               leading zero |N/A | N/A | N/A | N/A | N/A | N/A | N/A |
[#]    1 of 4 coding reference |N/A | N/A | N/A | N/A | N/A | N/A | N/A |
[#]
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 146432
[#]   Transfer Speed PM3 -> Client... 292864 bytes/s
[#] Various
[#]   Max stack usage......... 4224 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... 54
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 30713 Hz
[#] Installed StandAlone Mode
[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[#] Flash memory dictionary loaded
[#]
[usb] pm3 --> data tune
[=] ---------- Reminder ------------------------
[=] `hw tune` doesn't actively tune your antennas,
[=] it's only informative.
[=] Measuring antenna characteristics, please wait...
[/] 10
[=] ---------- LF Antenna ----------
[+] LF antenna: 74.93 V - 125.00 kHz
[+] LF antenna: 32.67 V - 134.83 kHz
[+] LF optimal: 74.93 V - 125.00 kHz
[+] Approx. Q factor (*): 12.5 by frequency bandwidth measurement
[+] Approx. Q factor (*): 13.1 by peak voltage measurement
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[+] HF antenna: 46.16 V - 13.56 MHz
[+] Approx. Q factor (*): 8.1 by peak voltage measurement
[+] HF antenna is OK

(*) Q factor must be measured without tag on the antenna

[+] Displaying LF tuning graph. Divisor 88 (blue) is 134.83 kHz, 95 (red) is 125.00 kHz.
```

iceman1001, 2023-09-03:
Interesting,   I never seen a ISO15693 readblock return 8 bytes.   Which card is this?

```
hf 15 info
```

ThorstenEngel, 2023-09-03:
Dear Iceman,

thank you for replying that quickly! It is a glucose sensor.

```
[usb] pm3 --> hf 15 info

[+]  UID: E0 07 A4 00 0B 3F 72 65
[+] TYPE: Texas Instrument France
[+] Using UID... E0 07 A4 00 0B 3F 72 65

[=] --- Tag Information ---------------------------
[+]       TYPE: Texas Instrument France
[+]        UID: E0 07 A4 00 0B 3F 72 65
[+]    SYSINFO: 00 04 65 72 3F 0B 00 A4 07 E0 FF 07
[+]      - DSFID not supported
[+]      - AFI   not supported
[+]      - IC reference not supported
[+]      - Tag provides info on memory layout (vendor dependent)
[+]            19 (or 18) bytes/blocks x 90 blocks
```

iceman1001, 2023-09-03:
hrm..
what is the output from the rdbl command?
```
hf 15 rdbl
```

ThorstenEngel, 2023-09-03:
```
[usb] pm3 -->
[usb] pm3 --> hf 15 rdbl -b 0

[+]  UID: E0 07 A4 00 0B 3F 72 65
[+] TYPE: Texas Instrument France
[+] Using UID... E0 07 A4 00 0B 3F 72 65

[=]       #  0  |lck| ascii
[=] ------------+---+------
[=] 5C 8E 2A E7 01 07 A1 | 122 | \.*....

[usb] pm3 -->
[usb] pm3 --> hf 15 dump

[+]  UID: E0 07 A4 00 0B 3F 72 65
[+] TYPE: Texas Instrument France
[+] Using UID... E0 07 A4 00 0B 3F 72 65
[+] Reading memory from tag UID E0 07 A4 00 0B 3F 72 65
[\]blk  43
[-] Tag returned Error 16: The specified block is not available (doesn't exist).

[=] block#   | data         |lck| ascii
[=] ---------+--------------+---+----------
[=]   0/0x00 | FF A1 AB A7  | 0 | ....
[=]   1/0x01 | 6A 52 9C 69  | 0 | jR.i
[=]   2/0x02 | 5A A4 FB 57  | 0 | Z..W
[=]   3/0x03 | F8 7D 4C 94  | 0 | .}L.
```

iceman1001, 2023-09-03:
hm... only seven bytes...  odd,    what is the trace list?
```
hf 15 rdbl -b 0
hf 15 list
```

ThorstenEngel, 2023-09-03:
Here you are (it's another sensor of same type, same odd results):

```
[usb] pm3 -->
[usb] pm3 --> hf 15 rdbl -b 0

[+]  UID: E0 07 A4 00 0B 3F 75 47
[+] TYPE: Texas Instrument France
[+] Using UID... E0 07 A4 00 0B 3F 75 47

[=]       #  0  |lck| ascii
[=] ------------+---+------
[=] CD CF 99 B2 8F 95 C3 | 173 | .......

[usb] pm3 -->
[usb] pm3 --> trace list -t 15
[=] downloading tracelog data from device
[+] Recorded activity (trace len = 80 bytes)
[=] start = start of start frame end = end of frame. src = source of transfer
[=] ISO15693 / iCLASS - all times are in carrier periods (1/13.56MHz)

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |      22016 | Rdr |26  01  00  f6  0a                                                       |  ok | INVENTORY
      26304 |      79552 | Tag |00  00  47  75  3f  0b  00  a4  07  e0  6d  00                           |  ok |
        320 |      55104 | Rdr |62  20  47  75  3f  0b  00  a4  07  e0  00  83  60                       |  ok | READBLOCK(0)
      59392 |     108544 | Tag |00  ad  cd  cf  99  b2  8f  95  c3  44  c1                               |  ok |
[usb] pm3 -->
```

iceman1001, 2023-09-03:
so your tag has one byte flags,   no lock info byte and 8 bytes of data.
```
 0   1   2   3   4   5   6   7   8   9  10
00  ad  cd  cf  99  b2  8f  95  c3  44  c1   
                                    -- --
                                     crc
```

I never seen such.   Easier to chat on discord.

ThorstenEngel, 2023-09-03:
Sure, please allow me (borste_52888) to talk to you.

iceman1001, 2023-09-04:
@DidierA   great find!

iceman1001, 2023-09-04:
What is the output of: 
`hf 15 rdbl --blk 255`

ThorstenEngel, 2023-09-04:
Hi, sorry for the delay. Here is the output:

```
[usb] pm3 --> hf 15 rdbl --blk 255

[+]  UID: E0 07 A4 00 0B 3F 72 65
[+] TYPE: Texas Instrument France
[+] Using UID... E0 07 A4 00 0B 3F 72 65
[!!] iso15693 card returned error 1: The command is not supported
```

Block 42 is the last block I can read without this error.

By the way: is it normal that when sniffing the communication (there exists a reader hw for this tag) almost always shows 00 00 for the crc?

```
      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |      17920 | Rdr |26  01  00  00                                                           |  !! | INVENTORY
      24288 |      77536 | Tag |00  00  65  72  3f  0b  00  a4  07  e0  ea  7a                           |  ok |
      85600 |     103520 | Rdr |02  a1  70  00                                                           |  !! | FAST_INVENTORY_READ
     108320 |     149280 | Tag |00  c5  09  30  01  f8  01  8f  7b                                       |  ok |
     154816 |     176832 | Rdr |22  23  94  00  00                                                       |  !! | READ_MULTI_BLOCK(0-14)
    1050848 |    1165536 | Tag |00  f2  8b  3e  8b  18  67  7e  b7  8c  e4  9f  ba  ec  10  18  40  1e   |     |
            |            |     |54  7f  0e  b4  6b  f3  bc  eb  45                                       |  ok |
    1172032 |    1198144 | Rdr |22  23  65  20  00  d0                                                   |  !! | READ_MULTI_BLOCK(101-133)
    1340000 |    1454688 | Tag |00  48  01  ba  b9  95  e6  47  88  8a  60  d7  4a  52  3b  ec  fc  b8   |     |
            |            |     |d3  3c  e3  0b  a4  9a  03  b4  fd                                       |  ok |
```

Warm regards and really thank you for going that deep into this. This is really appreciated!

iceman1001, 2023-10-03:
I pushed some changes to handle / detect 4 / 8 byte ..

its more of a work in progress right now,   but since I don't have access to this kind of tag it is a bit hard to test/debug.

iceman1001, 2023-10-13:
ping

iceman1001, 2024-01-14:
try https://github.com/RfidResearchGroup/proxmark3/commit/8d0b41a911ca308eac990cc6bdb21d3fd073ef1c  and see if this works better.

closed 2024-01-14 by iceman1001