RfidResearchGroup/proxmark3 #2104: [RELEASE 4.17130] Checklist - release name: Raccon

iceman1001, 2023-09-07:
# Checklist

- [ ] CHANGELOG.md: add title: `## [Raccon]2023-09-07]`
- [ ] `make style`
- [ ] `make miscchecks`
- [ ] `make clean; make client CC=clang CXX=clang++ LD=clang++` on recent Debian or Ubuntu
- [ ] `mymanualchecks.sh`
- [ ] `mycppcheck.sh` no alarming warning?
- [ ] `tools/build_all_firmwares.sh` check that the script contains all standalone modes then compile all standalone modes (linux only)
- [ ] `experimental_lib` compilation & tests
- [ ] `experimental_client_with_swig` compilation & tests
- [ ] GitHub Actions - green across the board ( MacOS, Ubuntu, Windows)

# OS compilation and tests

Run `tools/release_tests.sh` on:

- [ ] RPI Zero
- [ ] Jetson Nano
- [ ] WSL
- [ ] PSv3.10
- [ ] Archlinux
- [ ] Kali
- [ ] Debian Stable
- [ ] Debian Testing
- [ ] Ubuntu 22
- [ ] ParrotOS
- [ ] Fedora 37
- [ ] OpenSuse Leap
- [ ] OpenSuse Tumbleweed
- [ ] OSX (MacPorts)
- [ ] OSX (Homebrew)
- [ ] Android
- [ ] Termux

# creating release

- [ ] `make release RELEASE_NAME="Raccon"`
  - last line of output,  gives you next command to run.
  - Sample:  `git push && git push origin v4.17130`
- [ ] CHANGELOG.md: edit title to add version info: `## [Raccon.4.17130]2023-09-07]`

## Step Github releases

- [ ] Go to Github releases,  create release based on the new created tag and publish

## Step Homebrew updates

- [ ] update homebrew repo, file `proxmark3.rb`
  - with a SHA256 sum of the file `v4.17130.tar.gz`
  - with updated list of standalone modes

## Step package maintains

- [ ] make a list of new standalone modes, so when we alert package maintainers they have a sporting chance of adding them

doegox, 2023-09-08:
FTR SWIG & Python seems broken since the attempt to migrate the API when I run it with Python 3.11
I suspect the problem appeared with a new version of Python, I'm pretty sure we tested it at that time and it was fine (https://github.com/RfidResearchGroup/proxmark3/issues/1870)

iceman1001, 2023-09-08:
Could be,  I rarely run the SWIG compilation,    running python 3.10.12 on WSL1 / Ubuntu ...
Haven't gone over to 3.11 yet.   GCC11 though.

doegox, 2023-09-08:
ok it works fine with Python 3.9 and fails with 3.11 (no 3.10 around here), but the code of `cmdscript.c` is very different for Python >= 10. No idea how to fix that.

iceman1001, 2023-09-08:
Nice!  Maybe for another release.

doegox, 2023-09-08:
I'm finishing Tumbleweed tests (and I need to push fix on its docker scripts) then I'm done.

doegox, 2023-09-08:
:green_apple: :green_book: :green_heart: :green_salad:

iceman1001, 2023-09-09:
Only one new standalone mode:  

```
HF_CARDHOPPER |	Relay 14a protocols over long distances (w/ IP backbone) | Sam Haskins | RDV4
```

closed 2023-09-09 by iceman1001