RfidResearchGroup/proxmark3 #2113: sma.multi.cpp: undefined reference

JohnDoeIsKIng, 2023-09-14:
***Things to try before submitting bug report***
read the [troubleshooting guide](https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Troubleshooting.md)

***Compilation problems***
Try compiling with verbose.  `make VERBOSE=1` with main makefile or `make V=1` with cmake.

***flashing problems***
Have you followed the instructions properly?  ie,  flashed bootrom seperately first if you are going from Offical repo to Iceman repo.

-

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
 - hw status
 - data tune

**Additional context**
Add any other context about the problem here.

Dear community,
greets iceman,

with [c26d33d][(url](https://github.com/RfidResearchGroup/proxmark3/commit/c26d33dfd3cebb803a384e838c07908ccda4e240)) I'm getting a compiler error:

`[=] CXX sma_multi
obj/sma_multi.o: In Funktion »std::thread::thread<void (&)(), , void>(void (&)())«:
sma_multi.cpp:(.text._ZNSt6threadC2IRFvvEJEvEEOT_DpOT0_[_ZNSt6threadC5IRFvvEJEvEEOT_DpOT0_]+0x40): Warnung: undefinierter Verweis auf »pthread_create«
obj/sma_multi.o: In Funktion »main«:
sma_multi.cpp:(.text.startup+0xb1f): Warnung: undefinierter Verweis auf »pthread_create«
sma_multi.cpp:(.text.startup+0x20f1): Warnung: undefinierter Verweis auf »pthread_create«
sma_multi.cpp:(.text.startup+0x282c): Warnung: undefinierter Verweis auf »pthread_create«
collect2: error: ld returned 1 exit status
../../Makefile.host:89: recipe for target 'sma_multi' failed
make[1]: *** [sma_multi] Error 1
Makefile:152: recipe for target 'cryptorf/all' failed
make: *** [cryptorf/all] Error 2
`

Any hints to avoid that?
Regards

JD.

iceman1001, 2023-09-14:
hm... 
Looks like -lpthread isn't linked?

JohnDoeIsKIng, 2023-09-14:
Hey iceman,

thanks for replying quickly.
The issue arised sometime within the last two days. I've tested on two different build systems (Debian and Ubuntu) - they all developed exactly the same error.
Do I have to edit my Makefile with something like -pthread ?
Regards

JD.

ShutdownRepo, 2023-09-14:
I can confirm I encounter the same issue
```
2023-09-14T00:15:43.4443817Z #21 101.1 /usr/bin/ld: obj/sma_multi.o: in function `std::thread::thread<void (&)(), , void>(void (&)())':
2023-09-14T00:15:43.4444391Z #21 101.1 sma_multi.cpp:(.text._ZNSt6threadC2IRFvvEJEvEEOT_DpOT0_[_ZNSt6threadC5IRFvvEJEvEEOT_DpOT0_]+0x32): undefined reference to `pthread_create'
2023-09-14T00:15:43.4445298Z #21 101.1 /usr/bin/ld: obj/sma_multi.o: in function `main':
2023-09-14T00:15:43.4445500Z #21 101.1 sma_multi.cpp:(.text.startup+0xae5): undefined reference to `pthread_create'
2023-09-14T00:15:43.4445739Z #21 101.1 /usr/bin/ld: sma_multi.cpp:(.text.startup+0xfc0): undefined reference to `pthread_create'
2023-09-14T00:15:43.4445986Z #21 101.1 /usr/bin/ld: sma_multi.cpp:(.text.startup+0x137d): undefined reference to `pthread_create'
2023-09-14T00:15:43.4446191Z #21 101.1 collect2: error: ld returned 1 exit status
2023-09-14T00:15:43.4446370Z #21 101.1 make[1]: *** [../../Makefile.host:90: sma_multi] Error 1
2023-09-14T00:15:43.4446547Z #21 101.1 make: *** [Makefile:153: cryptorf/all] Error 2
```

iceman1001, 2023-09-14:
on which host OS / compiler are you using?

ShutdownRepo, 2023-09-14:
Checking out 774f1c33efaaccf633ede6e704800345eb313878 and building works, so it must've been brought with c26d33dfd3cebb803a384e838c07908ccda4e240

iceman1001, 2023-09-15:
hm..   Can you test this?

Line 14
`MYLDLIBS += -lpthread -Wl,--as-needed -latomic -Wl,--no-as-needed`

ShutdownRepo, 2023-09-15:
It seems to work with:
`tools/cryptorf/Makefile`
```
[...]
12 # iOS 'fun'
13 else ifneq ($(findstring iP, $(cpu_arch)), )
14     MYLDLIBS += -lpthread -Wl,--as-needed -latomic -Wl,--no-as-needed
15     MYCFLAGS += -mcpu=native
16 else
17     MYCFLAGS += -march=native
18 endif
[...]
```
I don't know if you wanted to add MYLDLIBS on top of MYCFLAGS or as a replacement 🤷

JohnDoeIsKIng, 2023-09-18:
Solved by #22d7adf .

closed 2023-09-18 by JohnDoeIsKIng