RfidResearchGroup/proxmark3 #2114: Getting unknown command:: 0x61334d50, please help?

michalss, 2023-09-21:
-

**Describe the bug**
I'm scare to run pm3-flash-all, since im afraid i brick the device... I have been following the tutorial and all went smoot but i cannot connect device with client without flash and not sure if it should work without flashing device first?

[image]

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.x.x LTS

**Additional context**
[image]

michalss, 2023-09-21:
Now i was able to flash it ok.. but im getting 

[image]

michalss, 2023-09-21:
how to fix this pls ?

iceman1001, 2023-09-21:
We use the discord server for user related questions. 

In your case here,   are you sure you complied for the target hw  and flashed the device ,  then using the same client from the build?

I.e,     In the proxmark world the Firmware goes hand in hand with the Software.    Don't mix 'em up.

iceman1001, 2023-09-21:
That would be the installation guide.   
You should look at the compilation guide linked to in the bottom of the installation guide.
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Use_of_Proxmark/0_Compilation-Instructions.md
And take notice of which device you have.

michalss, 2023-09-21:
Well i have been through this already. Problem is im not exactly sure what version i do have. I got it from here ..

https://www.aliexpress.com/item/1005005209158934.html?spm=a2g0o.order_detail.order_detail_item.3.3ca0f19cp0sXWq

michalss, 2023-09-21:
All i did found is this AT91SAM7S512  and 512Kb..

michalss, 2023-09-21:
this could lead to an issue :D
[!] ⚠️  Contradicting measures seem to indicate you're running a PM3_RDV4 firmware on a generic device

michalss, 2023-09-21:
ok i will recompile it with generic platform tag.. thx ill try this

michalss, 2023-09-21:
ok closing it worked 

[image]

closed 2023-09-21 by michalss

iceman1001, 2023-09-21:
That device is not a RDV4 as you guessed.  You built the repo with that setting,  which lead to your issues.
Once understanding you need to change target for compilation,   a flash and voilá  you are on your feet again.

I do recommend the RFID Hacking discord server.